gphoto/libgphoto2 #784: Canon 100D gp_camera_exit regression between 2.5.27 and 2.5.28

david-polak, 2022-03-29:
After upgrading from `2.5.27` to `2.5.28`, the Canon 100D does not close shutter or mirror after a successful exit.

Mirror gets flipped open and shutter as well during the `gp_camera_capture_preview`. After calling either `gp_camera_exit` or `gp_camera_unref` `2.5.27` closes the shutter and pulls down the mirror. `2.5.28` and `2.5.29` leave the mirror up and shutter open.

## Camera model

```
❯ gphoto2 --auto-detect
Model                          Port
----------------------------------------------------------
Canon EOS 100D                 usb:001,020
```

## Versions used

### Broken
```
❯ gphoto2 --version
//snip
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.28         standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without lockin
```

### Working
```
❯ gphoto2 --version
//snip
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.27         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without lockin
```

## Minimal reproducible example

* Create `main.c` with the following contents

```
#include <stdio.h>
#include <gphoto2/gphoto2-camera.h>
int main(int argc, char **argv) {
    GPContext *context;
    CameraFile *file;
    Camera *camera;
    FILE *devnull = fopen("/dev/null", "w");
    context = gp_context_new();
    gp_camera_new(&camera);
    gp_camera_init (camera, context);
    gp_file_new_from_fd(&file, (int) devnull);
    gp_camera_capture_preview(camera, file, context);
    gp_camera_exit(camera, context);
    gp_camera_unref(camera);
}
```
* Downgrade libgphoto to `2.5.27`
* Run the following command `gcc main.c -l gphoto2 && ./a.out`
* Observe: **Mirror flips open and then closes.**

* Upgrade ligphoto to `2.5.28`
* Run the following command `gcc main.c -l gphoto2 && ./a.out`
* Observe: **Mirror flips open and program exits.**

## Additional information
### OS
```
Operating System: Arch Linux
          Kernel: Linux 5.16.16-arch1-1
    Architecture: x86-64
```
### Downgrade commands
```
sudo pacman -U https://archive.archlinux.org/packages/l/libgphoto2-2.5.27-1-x86_64.pkg.tar.zst
```
```
sudo pacman -U https://archive.archlinux.org/packages/l/libgphoto2-2.5.28-1-x86_64.pkg.tar.zst
```

david-polak, 2022-04-06:
I have done a `git bisect` on the source and found the breaking commit.

https://github.com/gphoto/libgphoto2/commit/07a2176e11f5601db730fe4c3bf5206c47fc1ee5

Is it possible to revert it?

```
07a2176e11f5601db730fe4c3bf5206c47fc1ee5 is the first bad commit
commit 07a2176e11f5601db730fe4c3bf5206c47fc1ee5
Author: Marcus Meissner <marcus@jet.franken.de>
Date:   Mon Jan 3 10:30:12 2022 +0100

    keep canon eos remotemode=1 on exit, to keep the display on

    https://github.com/gphoto/gphoto2/issues/465

 camlibs/ptp2/config.c  | 3 ++-
 camlibs/ptp2/library.c | 2 ++
 2 files changed, 4 insertions(+), 1 deletion(-)
```

## Bisect log
```
❯ git bisect log
git bisect start
# bad: [19c7d65757cda8fa0a4a7c0ba167434b83a0b98b] added nikon z9
git bisect bad 19c7d65757cda8fa0a4a7c0ba167434b83a0b98b
# good: [7bb5934c95ed420d5dbf4e7baf2796cdccf8062f] 2.5.27.1 dev release
git bisect good 7bb5934c95ed420d5dbf4e7baf2796cdccf8062f
# good: [94d9912e1ee0451a6001347885b27ac2df659e76] Fix check output for DEFAULT_CAMLIBS and DEFAULT_IOLIBS
git bisect good 94d9912e1ee0451a6001347885b27ac2df659e76
# good: [204d41b26ec940d62429f20d46421d5d7feeb04e] Factored out more things into add_object_to_fs_and_path helper (tested: nikon z, and eos m6 m2)
git bisect good 204d41b26ec940d62429f20d46421d5d7feeb04e
# good: [204d41b26ec940d62429f20d46421d5d7feeb04e] Factored out more things into add_object_to_fs_and_path helper (tested: nikon z, and eos m6 m2)
git bisect good 204d41b26ec940d62429f20d46421d5d7feeb04e
# bad: [5af863a47ef8e95073ba166b68ebd6ca0b109f5e] gitignore: add gp-gettext-setup.mk files
git bisect bad 5af863a47ef8e95073ba166b68ebd6ca0b109f5e
# good: [dc78aec97b96b66e7dc4bf276bc50d0712802e18] Add TODO comment about possibly using gettext's gettext.h
git bisect good dc78aec97b96b66e7dc4bf276bc50d0712802e18
# good: [dc78aec97b96b66e7dc4bf276bc50d0712802e18] Add TODO comment about possibly using gettext's gettext.h
git bisect good dc78aec97b96b66e7dc4bf276bc50d0712802e18
# good: [8c2ee45f8d6bd40e658e073c3bbcf3b1ba78d529] sync from libmtp
git bisect good 8c2ee45f8d6bd40e658e073c3bbcf3b1ba78d529
# bad: [b9b32e101d6d825d68920c8fff80976f24cc5e2f] updated ukramniasn
git bisect bad b9b32e101d6d825d68920c8fff80976f24cc5e2f
# bad: [7ac02fa09a92fce2eeb43f8e821ab171644835ac] 2.5.28 release
git bisect bad 7ac02fa09a92fce2eeb43f8e821ab171644835ac
# bad: [07a2176e11f5601db730fe4c3bf5206c47fc1ee5] keep canon eos remotemode=1 on exit, to keep the display on
git bisect bad 07a2176e11f5601db730fe4c3bf5206c47fc1ee5
# good: [a829e5088d0369b3db7d902d794103597f522616] added more NEWS
git bisect good a829e5088d0369b3db7d902d794103597f522616
# first bad commit: [07a2176e11f5601db730fe4c3bf5206c47fc1ee5] keep canon eos remotemode=1 on exit, to keep the display on
```

david-polak, 2022-04-07:
I am willing to run any branch of libgphoto2 against my camera if you want to debug it.

david-polak, 2022-04-07:
Oh, I tested master against my camera. And on current master the camera is not usable after my application exits. Screen is off and none of the buttons work.

david-polak, 2022-04-07:
Ah, wait. I actually got it working correctly, PR soon.

david-polak, 2022-04-07:
@msmeissn https://github.com/gphoto/libgphoto2/pull/787 this is working as expected on my camera

https://user-images.githubusercontent.com/5686338/162192637-91f7d53d-488f-4265-9e00-3343f748932e.mp4

closed 2022-04-07 by msmeissn

msmeissn, 2022-04-07:
i will test it, but already merged it into git. thanks for doing this! :)

PR #787 (2022-04-07): Configure Canon cameras on shutdown
Fixes #784 

https://user-images.githubusercontent.com/5686338/162192469-50d0b1c8-071f-4978-b2c2-2850e795036f.mp4