gphoto/libgphoto2 #785: Trigger capture while a file is downloading.

bkapadia02, 2022-04-03:
I am writing an application where it is critical that I trigger the capture at the exact moment when a condition is met. It seems that there is no way to trigger a capture while a file is being sent from the camera to the PC. Is there a workaround for this other than having my main process to somehow kill the thread that is downloading the files?

msmeissn, 2022-04-03:
The communication is pretty much over one pipe, the USB bus... interleaving commands / data like this is not really possible.

if the main process is downloading in smaller blocks as we do with getpartialobject these days, you can inject commands inbetween.

msmeissn, 2022-07-16:
no further comms received, please reopen if needed

closed 2022-07-16 by msmeissn