gphoto/libgphoto2 #790: How to manually install libgphoto2 update

Sdee0619, 2022-04-12:
I auto updated libgphoto2 but it gave me version 2.5.12 
 I am trying to download the latest version, I manually downloaded the newest version what command can I use to update my libgphoto2. Also any links to a list of commands or a tutorial cant seem to find any. thanks in advance

Sdee0619, 2022-04-12:
forgot to add I am trying to setup my Panasonic Dslr and it requires a newer version for gphoto2 to detect

Sdee0619, 2022-04-13:
I got Panasonic Gh5 to capture image the fix was to delete old gphoto2.
I am now stuck on Step 5 create a snapshot script when I input command
 cd /home/pi/scripts
I get  No such file or directory

Sdee0619, 2022-04-13:
pi@abcd-ch-canvas-hub:~ $ ls
canvas-hub          led.py       OctoPrint     P1000006.JPG  P1000008.JPG  P1000012.JPG  P1000445.JPG  Python-3.7.9.tgz  webcam
gphoto2-updater.sh  netconnectd  P1000003.JPG  P1000007.JPG  P1000010.JPG  P1000013.JPG  Python-3.7.9  take-snapshot.sh

msmeissn, 2022-04-13:
if you are using gphoto2-updater.sh already, rty to remove the old version from the system as in the README.md:

sudo apt-get remove gphoto2 libgphoto2-6 libgphoto2-dev libgphoto2-l10n libgphoto2-port12

snaksun, 2022-04-18:
do you have any instruction how I can install it on my raspberry pi ?

ndim, 2022-06-02:
We do not have instructions specifically for Raspberry Pi. You can install `libgphoto2` and `gphoto2` on Raspberry Pi OS the same way you install them on any BSD or Linux system.

Hmm. Maybe I should update the `INSTALL` file for `$HOME/.local` and `PKG_CONFIG_PATH`.

msmeissn, 2022-06-25:
also the the gphoto2-updater.sh should work on rasperry pi

closed 2022-06-25 by msmeissn