gphoto/libgphoto2 #791: Ask a question about the connection failure on the SONY RX100 M7 model.

liuhengmao, 2022-04-21:
Describe the bug
I used the framework(https://github.com/touchbyte/GPhoto2Framework) to test it successfully on Canon and Fuji cameras.
But on the sony camera, the connection fails.
log content
cmd: connection refused

Name the camera
SONY RX100 M7 model

libgphoto2 and gphoto2 version
libgphoto2 2.5.26

To Reproduce
cameraModel is set to "PTP/IP Camera"
The ip address is manually specified after scanning by the tool
Others are the same as the canon connection in the demo

jabdoa2, 2022-05-05:
Sony RX100 M7 works for me. You have to set the camera to PC mode. Thats a bit tricky (I googled it). You have to disable mobile phone remote control and probably a few more things.

liuhengmao, 2022-05-06:
> 
Thank you very much for your reply, do you mean Sony RX100 M7 can't connect wirelessly and access control?

jabdoa2, 2022-05-06:
In my experience you have to decide between USB (PC Control is that called in my camera) and wireless (remote) control.