gphoto/libgphoto2 #792: FujiFilm GFX100S

vepicurean, 2022-04-25:
FujiFilm GFX 100S supported? I don't see it specifically supported in the Wiki. But the Wiki list of cameras has not been updated in a long time. 

I don't have one of our Fujis with me, so can't list the USB IDs. 

Just want to make sure the current versions of ligphoto2 and gphoto2 support the GFX 100S before we buy more Fuji cameras. 

I see the older GFX 100 is supported. 

I'm assuming, but I'd like someone to confirm, that if the GFX 100 is supported, we should be OK on libgphoto2 and gphoto2 for support on the GFX100S

msmeissn, 2022-04-26:
While we had so far no report for the GFX100s, it might work. but we cannot reliably say yes or no.

The GFX100 was reported working FWIW, at least for capture: https://github.com/gphoto/gphoto2/issues/256

smaun92, 2022-04-27:
Hi,

smaun92, 2022-04-27:
Hi Marcus,

I have tested Fujifilm GFX100s with gphoto2. I have the following version of gphoto2, libgphoto2 and libghoto2_port
gphoto2                  2.5.27
libgphoto2              2.5.27
lingphoto2_port     0.12.0

Its not stating the model of the camera when I use gphoto2 --auto-detect, just getting this
Model                          Port                                           
----------------------------------------------------------
USB PTP Class Camera           usb:001,011

I get the following when I run gphoto2 --Summary

`Camera summary:                                                                
Manufacturer: FUJIFILM
Model: GFX100S
  Version: 1.10
  Serial Number: 
Vendor Extension ID: 0xe (1.0)
Vendor Extension Description: [fujifilm.co.jp](http://fujifilm.co.jp/): 1.0; 

Capture Formats: Defined Type JPEG Unknown(b103)
Display Formats: Defined Type, JPEG, Unknown(b103), TIFF

Device Capabilities:
	File Download, File Deletion, No File Upload
	Generic Image Capture, Open Capture, No vendor specific capture

Storage Devices Summary:
store_10000001:
	StorageDescription: Still
	VolumeLabel: 
	Storage Type: Builtin RAM
	Filesystemtype: Generic Hierarchical
	Access Capability: Read Only with Object deletion
	Maximum Capability: 18446744073709551615 (4294967295 MB)
	Free Space (Bytes): 18446744073709551615 (4294967295 MB)
	Free Space (Images): -1
store_10000002:
	StorageDescription: Live
	VolumeLabel: 
	Storage Type: Builtin RAM
	Filesystemtype: Generic Hierarchical
	Access Capability: Read Only with Object deletion
	Maximum Capability: 18446744073709551615 (4294967295 MB)
	Free Space (Bytes): 18446744073709551615 (4294967295 MB)
	Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001): error 200a on query.
Image Size(0x5003):(readwrite) (type=0xffff) Enumeration [
	'4000x2664',
	'4000x2248',
	'2992x2992',
	'8256x5504',
	'8256x4640',
	'6192x6192',
	'11648x7768',
	'11648x6552',
	'8736x8736',
	'4992x3328',
	'4992x2808',
	'3328x3328',
	'4000x3000',
	'4000x1480',
	'3744x3000',
	'3504x3000',
	'11648x8736',
	'11648x4304',
[... 211 lines cut ...]
FileNamePrefix2(0xd366):(readwrite) (type=0xffff) 'DSF'
FocusArea3(0xd374): error 201b on query.
TotalShotCount(0xd310):(read only) (type=0x6) 166002688
ExposurePreview(0xd359):(readwrite) (type=0x4) Enumeration [1,2,3] value: 1
FrameGuideGridInfo1(0xd375):(readwrite) (type=0xffff) '0,0,0,0,850,0,0,0,0,850,0,0,0,0,'
FrameGuideGridInfo2(0xd376):(readwrite) (type=0xffff) '0,0,0,0,850,0,0,0,0,850,0,0,0,0,'
CustomDispInfo(0xd36e): error 200a on query.
ViewMode1(0xd33f):(readwrite) (type=0x4) Enumeration [1,2,3,4,5,6,7,8,9] value: 3
CustomAutoPowerOff(0xd364):(readwrite) (type=0x4) Range [0 - 255, step 1] value: 0
LockButtonMode(0xd34e):(readwrite) (type=0x4) Enumeration [1,2] value: 1
WideDynamicRange(0xd02e):(readwrite) (type=0x4) Enumeration [0,1,2,3,32768] value: 0
LensNameAndSerial(0xd36d):(read only) (type=0xffff) 'LR107A,GF250mmF4 R LM OIS WR,05C00022'
LensUnknownData(0xd38a):(read only) (type=0x4) Enumeration [1,2,3,4,5] value: 2
BatteryInfo1(0xd36a):(read only) (type=0x6) Range [0 - 16777215, step 1] value: 10
BatteryInfo2(0xd36b):(read only) (type=0xffff) '53,0,0,0'
FunctionLockCategory1(0xd36f):(readwrite) (type=0x6) 65536
FunctionLockCategory2(0xd370):(readwrite) (type=0x6) 65536
SensitivityFineTune1(0xd222):(readwrite) (type=0x4) Range [0 - 65535, step 1] value: 8192
SensitivityFineTune2(0xd223):(readwrite) (type=0x4) Range [0 - 65535, step 1] value: 8192
LensZoomPosCaps(0xd38c):(readwrite) (type=0x4) Range [1 - 1, step 1] value: 1
LensFNumberList(0xd38d):(read only) (type=0xffff) '400,450,500,560,640,710,800,900,1000,1100,1300,1400,1600,1800,2000,2200,2600,2900,3200'
LensFocalLengthList(0xd38e):(read only) (type=0xffff) '25000,19800'
Property 0xd17b:(readwrite) (type=0x4) Enumeration [1] value: 1`

**Now I ll state the issue I am facing or more like a query I have which lead me to this. I am trying to configure AEB setting but I can't seem to find the option where I can. Is it something we can expect in the coming versions or is there any way to harness this feature**

smaun92, 2022-04-29:
@vepicurean Really appreciate the work you guys are doing. I am also having this (https://askubuntu.com/questions/993876/gphoto2-could-not-claim-the-usb-device) issue where I have to kill these processes to get it to work. I am not sure if it is Camera Model related or something else but is there anyway to avoid it at every startup?

msmeissn, 2022-04-29:
the camera blocking is not related to cameras, it is related to gvfs slaves.   

gio mount -s gphoto2

is supposed to stop it, or just delete the gphoto relatredbinaries of gvfs.-

msmeissn, 2022-07-16:
can you also run 
lsusb 
and report the FUJI GFX100s USB ids?

msmeissn, 2022-07-16:
as for AEB ... as we map only known names, can you do:

gphoto2 --summary > before
... change AEB on the camera one stepping ... 
gphoto2 --summary > after

diff -u before after

to see what property changed and might be AEB

vepicurean, 2025-04-11:
> Finally got around to testing the GFX 100 ( version 1 - the GFX 100S is next). I realized this camera is meant to run in tethered mode with a computer ( MacOS). I put the camera in tethered - fixed mode to download the images straight to the compute ( no SD card). This is for rapid capture.
> 
> gphoto2 --capture-tethered --filename="somefilename"
> 
> That tether seems to start easy enough. The process starts even though there is no messaging to indicated it's started.
> 
> The issue is triggering the tether ( capture and then download) in a stable manner.
> 
> kill SIGUSR1 PID ( the process ID of the capture-tether) may work once, it may even work twice. It won't work three times in a row. You get an PTP general error "-1 unspecified error". Strange that it will work one or twice, but that's it.

Some more background FYI information for this....

I installed the Fuji GFX tether app on the latest version of MacOS and the GFX 100 works perfectly with that app. Same settings on the GFX ( I didn't change anything, tethered - fixed setting on the connections menu).