gphoto/libgphoto2 #793: Focus drive in manual focus on Sony RX100 M7

jabdoa2, 2022-05-05:
**Is your feature request related to a problem? Please describe.**
I use the RX100 to capture images in a stationary application. With Sony software I can change the focus of the RX100 M7. This does not seem to work with gphoto.

**Describe the solution you'd like**
I would like an option/action to drive focus of the RX100. Ideally focus can be set to an absolute value. If not relative movements would work as well.

**Describe alternatives you've considered**
Currently I set focus manually. Unfortunately, I have to repeat this every time the camera is power cycled because it always restores to some default which I cannot change (or did not find it).

**Additional context**
Thanks for this awesome library! Can I provide any help here? Logs/dumps config listings?

msmeissn, 2022-05-08:
can you specify how you tried manual focusing? with what gphoto2 commands / api calls?

there is "manualfocus" option for sony, where you can specify stepping -7 to 7  for relative focusing.

absolute focusing usually is not possible, as the focus monitor does not know the position it is in with the modern non-mechanic drives.

jabdoa2, 2022-05-08:
> can you specify how you tried manual focusing? with what gphoto2 commands / api calls?
>
> there is "manualfocus" option for sony, where you can specify stepping -7 to 7 for relative focusing. [...]

I tried: `gphoto2 --set-config=/main/actions/manualfocus=7` but it said that this is unknown config. It also did not show up in `--list-all-config`. Do I need a specific version for that action?

msmeissn, 2022-05-09:
can you attach gphoto2 --summary 
output? and look if 0xd2d1 is listed?

jabdoa2, 2022-05-13:
Looks like that exists:
```
$ gphoto2 --summary 

Loading camera drivers from '/usr/l... |                     \\   0,0%        Loading camera drivers from '/usr/l... |-                    |   5,6%        Loading camera drivers from '/usr/l... |--                   /  11,1%        Loading camera drivers from '/usr/l... |----                 -  16,7%        Loading camera drivers from '/usr/l... |-----                \\  22,2%        Loading camera drivers from '/usr/l... |------               |  27,8%        Loading camera drivers from '/usr/l... |-------              /  33,3%        Loading camera drivers from '/usr/l... |--------             -  38,9%        Loading camera drivers from '/usr/l... |---------            \\  44,4%        Loading camera drivers from '/usr/l... |-----------          |  50,0%        Loading camera drivers from '/usr/l... |------------         /  55,6%        Loading camera drivers from '/usr/l... |-------------        -  61,1%        Loading camera drivers from '/usr/l... |--------------       \\  66,7%        Loading camera drivers from '/usr/l... |---------------      |  72,2%        Loading camera drivers from '/usr/l... |----------------     /  77,8%        Loading camera drivers from '/usr/l... |-----------------    -  83,3%        Loading camera drivers from '/usr/l... |-------------------  \\  88,9%        Loading camera drivers from '/usr/l... |-------------------- |  94,4%                                                                                     Kamerazusammenfassung:
Manufacturer: Sony Corporation
Model: DSC-RX100M7
  Version: 1.00
  Serial Number: 00000000000000008081885002911489
Vendor Extension ID: 0x11 (1.0)
Vendor Extension Description: Sony PTP Extensions
Capture Formats: 
Display Formats: JPEG, Unknown(b301), ARW
Device Capabilities:
        File Download, No File Deletion, File Upload
        No Image Capture, No Open Capture, Sony Capture
Storage Devices Summary:
Device Property Summary:
Compression Setting(0x5004):(readwrite) (type=0x2) Enumeration [16,20,19,18,4,3,2] value: RAW (4)
White Balance(0x5005):(readwrite) (type=0x4) Enumeration [2,4,32785,32784,6,32769,32770,32771,32772,7,32816,32786,32800,32801,32802] value: 32770
F-Number(0x5007):(readwrite) (type=0x4) Range [0 - 65535, step 1] value: f/8 (800)
Focus Mode(0x500a):(readwrite) (type=0x4) Enumeration [2,32773,32772,32774,1] value: Manual Focus (1)
Exposure Metering Mode(0x500b):(readwrite) (type=0x4) Enumeration [32769,32770,32772,32773,32771,32774] value: 32769
Flash Mode(0x500c):(readwrite) (type=0x4) Enumeration [] value: Flash off (2)
Exposure Program Mode(0x500e):(readwrite) (type=0x4) Enumeration [] value: M (1)
Exposure Bias Compensation(0x5010):(readwrite) (type=0x3) Enumeration [] value: 0,0 stops (0)
Still Capture Mode(0x5013):(readwrite) (type=0x4) Enumeration [1,32786,32789,2,32816,32817,32818,32772,32771,32773,32776,32777,32780,32781,32782,32783,33591,34103,35127,33655,34167,35191,33553,34065,35089,33569,34081,33585,34097,33590,34102,35126,33654,34166,35190,33552,34064,35088,33568,34080,33584,34096,32808,32792,32809,32793] value: Single Shot (1)
DOC Compensation(0xd200):(readwrite) (type=0x3) Enumeration [] value: 0
DRangeOptimize(0xd201):(readwrite) (type=0x2) Enumeration [1,31,17,18,19,20,21,32,33,34,35,36,37,38] value: 31
Image size(0xd203):(readwrite) (type=0x2) Enumeration [1,2,3] value: 1
Shutter speed(0xd20d):(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 262154
Property 0xd20e:(readwrite) (type=0x2) Enumeration [] value: 15
Color temperature(0xd20f):(readwrite) (type=0x4) Range [2500 - 9900, step 100] value: 5500
CC Filter(0xd210):(readwrite) (type=0x2) Range [164 - 220, step 1] value: 192
Aspect Ratio(0xd211):(readwrite) (type=0x2) Enumeration [1,3,2,4] value: 1
Focus status(0xd213):(readwrite) (type=0x2) Enumeration [] value: 1
Property 0xd214:(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 4294967295
Objects in memory(0xd215):(readwrite) (type=0x4) Range [0 - 65535, step 1] value: 0
Property 0xd217:(readwrite) (type=0x2) Enumeration [] value: 1
Battery Level(0xd218):(readwrite) (type=0x1) Range [-1 - 100, step 1] value: 70
Picture Effect(0xd21b):(readwrite) (type=0x4) Enumeration [32768,32769,32770,32771,32772,32773,32784,32800,32801,32816,32832,32848,32849,32850,32851,32864,32912,32880,32881,32882,32896,32897,32898,32928,32929,32930,32931,32932,32933,32934,32944,32960,32961,32962] value: 32768
AB Filter(0xd21c):(readwrite) (type=0x2) Range [164 - 220, step 2] value: 192
Property 0xd21d:(readwrite) (type=0x2) Range [0 - 2, step 1] value: 0
ISO(0xd21e):(readwrite) (type=0x6) Enumeration [] value: 100
Property 0xd220:(readwrite) (type=0x2) Enumeration [] value: 1
Property 0xd221:(readwrite) (type=0x2) Enumeration [] value: 0
Property 0xd222:(readwrite) (type=0x4) Enumeration [1,17,16] value: 1
Property 0xd223: error 2002 on query.
Property 0xd224:(readwrite) (type=0x3) Enumeration [] value: 0
Property 0xd226:(readwrite) (type=0x6) Enumeration [16777215,64,80,100,125,160,200,250,320,400,500,640,800,1000,1250,1600,2000,2500,3200,4000,5000,6400,8000,10000,12800,33554431,16777316,16777416,16777616,16778016,16778816,16780416,16783616,16790016,16802816,50331647,33554532,33554632,33554832,33555232,33556032,33557632,33560832,33567232,33580032] value: 100
Property 0xd227:(readwrite) (type=0x2) Enumeration [] value: 0
Property 0xd228:(readwrite) (type=0x2) Enumeration [4,5] value: 5
[... cut ...]
Property 0xd2d5:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2d6:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2d7:(readwrite) (type=0x4) Enumeration [1,2] value: 2
Property 0xd2d8:(readwrite) (type=0x4) Enumeration [1,2] value: 2
Property 0xd2d9:(readwrite) (type=0x4) Enumeration [1,2] value: 2
Property 0xd2dd:(readwrite) (type=0x1) Range [-1 - 1, step 1] value: 2
Property 0xd2de:(readwrite) (type=0x4) Enumeration [1,2] value: 2
```

Any idea what I should look into?

jabdoa2, 2022-11-27:
To follow up on this: It seems "manualfocus" is missing here as well: https://github.com/gphoto/libgphoto2/blob/master/camlibs/ptp2/cameras/sony-rx100m7.txt. Is there some workaround to enable it anyway? Where would I have to patch libgphoto2 to get this working? 

I tried:
```
gphoto2 --set-config=/main/other/d2d1=7
```

It did not complain but also seemed to have not done anything.

jabdoa2, 2022-11-27:
Another thing I noticed (not sure if this is related):
```
WARNING: gphoto2: (_get_config [config.c:11188]) Type of property 'Manual-Focus' expected: 0x0004 got: 0x0003
WARNING: gphoto2: (_get_config [config.c:11188]) Type of property 'Bulb Mode' expected: 0x0000 got: 0x0004
WARNING: gphoto2: (_get_config [config.c:11188]) Type of property 'Capture Target' expected: 0x0000 got: 0x0004
```

msmeissn, 2022-11-28:
Ah, yes. Manual Focus hat UINT16 as type, while it is INT16. This likely has hidden it from view.

I adjusted this in current GIT of libgphoto2 now.

Usually the camera reports it to us and it seems to have reported d2d1.
It might only work in preview mode.

jabdoa2, 2022-11-28:
Thanks! I will give that a try.