gphoto/libgphoto2 #794: Sony A7M4 camera connect error

SiuyuenChow, 2022-05-06:
we want to connect Sony a7m4 
protocol is "ptpip"
model is "PTP/IP Camera"

 gp_camera_new === 0
 gp_abilities_list_new === 0
 gp_abilities_list_load === 0
 gp_port_info_list_new === 0
 gp_port_info_list_load === 0
 gp_port_info_list_count ===== 2
 gp_port_info_list_free ===0
 gp_abilities_list_free === 0
 gp_setting_set ptpip === 0
 gp_setting_setp ptp2 ===0
gp_camera_init ====-7

i dont know why ,error == -7 (Generic I/O error),

Did I set it wrong?

Please help,thanks.

SiuyuenChow, 2022-05-06:
> 

I solved the problem,

but the new one come in

-----
The new question is

call  gp_camera_wait_for_event

get type == GP_EVENT_FILE_ADDED

name = "capt0000.jpg"
folder = "/"

*** Contexterror ***              
You need to specify a folder starting with /store_xxxxxxxxx/

SiuyuenChow, 2022-05-07:
here is the log

logdumper ========== ptp_sony_getalldevicepropdesc: Objects in memory(d215): value 0 -> 32769
logdumper ========== DEBUG== 0xd215 after capture = 32769
logdumper ========== SONY ObjectInMemory count change seen, retrieving file
logdumper ========== Sending PTP_OC 0x1008 (Get object info) (0xffffc001) request...
logdumper ========== hdr type 8, length 18
logdumper ========== Reading PTP_OC 0x1008 (Get object info) data...
logdumper ========== hdr type 8, length 18
logdumper ========== hdr type 8, length 18
logdumper ========== Reading PTP_OC 0x1008 (Get object info) response...
logdumper ========== hdr type 8, length 18
logdumper ========== PTPIP_END_DATA_PACKET
logdumper ========== hdr type 8, length 18
logdumper ========== PTPIP_CMD_RESPONSE
logdumper ========== ObjectInfo for 'DSC00048.JPG':
logdumper ==========   Object ID: 0xffffc001
logdumper ==========   StorageID: 0x00010000
logdumper ==========   ObjectFormat: 0x3801
logdumper ==========   ProtectionStatus: 0x0000
logdumper ==========   ObjectCompressedSize: 231470
logdumper ==========   ThumbFormat: 0x0000
logdumper ==========   ThumbCompressedSize: 0
logdumper ==========   ThumbPixWidth: 0
logdumper ==========   ThumbPixHeight: 0
logdumper ==========   ImagePixWidth: 0
logdumper ==========   ImagePixHeight: 0
logdumper ==========   ImageBitDepth: 0
logdumper ==========   ParentObject: 0x00000000
logdumper ==========   AssociationType: 0x0000
logdumper ==========   AssociationDesc: 0x00000000
logdumper ==========   SequenceNumber: 0x00000000
logdumper ==========   ModificationDate: 0x00000000
logdumper ==========   CaptureDate: 0x00000000
logdumper ========== trying to get object size=0x3882e
logdumper ========== Sending PTP_OC 0x1009 (Get object) (0xffffc001) request...
logdumper ========== hdr type 8, length 18
logdumper ========== Reading PTP_OC 0x1009 (Get object) data...
logdumper ========== Reading PTP_OC 0x1009 (Get object) response...
logdumper ========== hdr type 8, length 18
logdumper ========== PTPIP_END_DATA_PACKET
logdumper ========== hdr type 8, length 18
logdumper ========== PTPIP_CMD_RESPONSE
logdumper ========== Append //capt0001.jpg to filesystem
logdumper ========== Lookup folder '/'...
logdumper ========== Found! / is 0x28112d020
logdumper ========== Internal append capt0001.jpg to folder /
logdumper ========== Adding file 'capt0001.jpg' to folder '/' (type 1)...
logdumper ========== Lookup folder / file capt0001.jpg
logdumper ========== Lookup folder '/'...
logdumper ========== Found! / is 0x28112d020
logdumper ========== Adding file 'capt0001.jpg' from folder '/' to the fscache LRU list (type 1)...
logdumper ========== Lookup folder / file capt0001.jpg
logdumper ========== Lookup folder '/'...
logdumper ========== Found! / is 0x28112d020
logdumper ========== File 'capt0001.jpg' from folder '/' added in fscache LRU list.
logdumper ========== Checking fscache LRU list integrity...
logdumper ========== fscache LRU list ok with 2 items (463089 bytes)

msmeissn, 2022-05-08:
you will need to download this file right afterward.

if you could share the code snippet from wait_event file_added to download?

SiuyuenChow, 2022-05-08:
> you will need to download this file right afterward.
> 
> if you could share the code snippet from wait_event file_added to download?

the code is 
if (evttype == GP_EVENT_FILE_ADDED) {
 path = (CameraFilePath*)evtdata;
 fd = open (path->name, O_CREAT | O_WRONLY , 0644);
 retval = gp_file_new_from_fd(&file, fd);
}

It was always ok with Canon, now it's not work with SONY. Did I make a mistake here?

The other errors are just as printed in the previous log

msmeissn, 2022-05-08:
And then you run
 gp_camera_file_get(camera,path->folder,path->name,GP_FILE_TYPE_NORMAL,file,context); or similar?

which libgphoto2 version do you have?

SiuyuenChow, 2022-05-08:
> gp_camera_file_get

call gp_camera_file_get_info  ,canon 200D,5D4,6D,6R is work.

My logic goes something like this,
I listened on GP_EVENT_FILE_ADDED and then downloaded the photo via gp_camerA_file_get_info

msmeissn, 2022-05-08:
can you just try gp_camera_file_get instead?  but the info should also be available, weird.
(if that error appears it means the just downloaded image was not cached correctly.)

msmeissn, 2022-05-08:
Also capturetarget setting should not cause it, although /capt0000... would be SDRAM and not CARD captured.

SiuyuenChow, 2022-05-08:
> can you just try gp_camera_file_get instead? but the info should also be available, weird. (if that error appears it means the just downloaded image was not cached correctly.)

yup, I thought it was weird, too,

SiuyuenChow, 2022-05-08:
> Also capturetarget setting should not cause it, although /capt0000... would be SDRAM and not CARD captured.

Capturetarget SET SDRAM or CARD, same result

I'll try gp_camerA_file_get later

keep in touch

Since there is no special swift version now, I want to package Gphoto into swift version in the future

SiuyuenChow, 2022-05-09:
@msmeissn 
hello,
i try  this 

		case GP_EVENT_FILE_ADDED:
			path = (CameraFilePath*)evtdata;
			printf("File added on the camera: %s/%s\n", path->folder, path->name);

			retval = gp_camera_file_get_info (camera, path->folder, path->name, &info, context);
			printf ("	info reported flags: %d\n", info.file.fields);
			if (info.file.fields & GP_FILE_INFO_MTIME) printf ("	info reported mtime: %ld\n", info.file.mtime);
			if (info.file.fields & GP_FILE_INFO_SIZE) printf ("	info reported size: %ld\n", info.file.size);
			if (info.file.fields & GP_FILE_INFO_TYPE) printf ("	info reported type: %s\n", info.file.type);

			fd = open (path->name, O_CREAT | O_WRONLY | O_BINARY, 0644);
			retval = gp_file_new_from_fd(&file, fd);
			printf("  Downloading %s...\n", path->name);
			retval = gp_camera_file_get(camera, path->folder, path->name,
				     GP_FILE_TYPE_NORMAL, file, context);

			printf("  Deleting %s on camera...\n", path->name);
			retval = gp_camera_file_delete(camera, path->folder, path->name, context);
			gp_file_free(file);
			free(evtdata);
			break;

-----
Same result

-----

cannon :
logdumper --- entry type 0000
logdumper --- waiting for events timeout 300 ms
logdumper --- Sending PTP_OC 0x911d (EOS_KeepDeviceOn) request...
logdumper --- Reading PTP_OC 0x911d (EOS_KeepDeviceOn) response...
logdumper --- PTPIP_CMD_RESPONSE
logdumper --- Sending PTP_OC 0x9116 (EOS_GetEvent) request...
logdumper --- Reading PTP_OC 0x9116 (EOS_GetEvent) data...
logdumper --- Reading PTP_OC 0x9116 (EOS_GetEvent) response...
logdumper --- PTPIP_CMD_RESPONSE
logdumper --- entry type 0001
logdumper --- Found new objectinfo! OID 0x919259d2, name IMG_9629.JPG
logdumper --- (20001,91900000,/store_00020001/)
logdumper --- (20001,90000000,/store_00020001/)
logdumper --- (20001,0,/store_00020001/)
logdumper --- Append /store_00020001/DCIM/100CANON/IMG_9629.JPG to filesystem
logdumper --- Lookup folder '/store_00020001/DCIM/100CANON'...
logdumper --- Internal append IMG_9629.JPG to folder 100CANON
logdumper --- Getting file info for 'IMG_9629.JPG' in '/store_00020001/DCIM/100CANON'...
logdumper --- Getting information about 'IMG_9629.JPG' in '/store_00020001/DCIM/100CANON'...
logdumper --- Lookup folder /store_00020001/DCIM/100CANON file IMG_9629.JPG
logdumper --- Lookup folder '/store_00020001/DCIM/100CANON'...
logdumper --- (storage=0x00020001, handle=0x00000000)
logdumper --- (storage=0x00020001, handle=0x90000000)
logdumper --- (storage=0x00020001, handle=0x91900000)
logdumper --- Sending PTP_OC 0x1008 (Get object info) (0x919259d2) request...
logdumper --- Reading PTP_OC 0x1008 (Get object info) data...
logdumper --- Reading PTP_OC 0x1008 (Get object info) response...
logdumper --- PTPIP_CMD_RESPONSE
----------

sony:

logdumper --- waiting for events timeout 300 ms
logdumper --- _ptp_sony_getalldevicepropdesc: opcode 9209
logdumper --- Sending PTP_OC 0x9209 (GetAllDevicePropData) request...
logdumper --- hdr type 8, length 18
logdumper --- Reading PTP_OC 0x9209 (GetAllDevicePropData) data...
logdumper --- hdr type 8, length 18
[... 78 lines cut ...]
logdumper --- Querying folder /...
logdumper --- Delete all files in folder 0x283db4db0//
logdumper --- file_list_func(/)
logdumper --- Internal append capt0000.jpg to folder /
logdumper --- Adding file 'capt0000.jpg' to folder '/' (type 1)...
logdumper --- Lookup folder / file capt0000.jpg
logdumper --- Lookup folder '/'...
logdumper --- Found! / is 0x283db4db0
logdumper --- Adding file 'capt0000.jpg' from folder '/' to the fscache LRU list (type 1)...
logdumper --- Lookup folder / file capt0000.jpg
logdumper --- Lookup folder '/'...
logdumper --- Found! / is 0x283db4db0
logdumper --- File 'capt0000.jpg' from folder '/' added in fscache LRU list.
logdumper --- Checking fscache LRU list integrity...
logdumper --- fscache LRU list ok with 1 items (282749 bytes)
logdumper --- Getting file info for 'capt0000.jpg' in '/'...
logdumper --- Getting information about 'capt0000.jpg' in '/'...
logdumper --- Lookup folder / file capt0000.jpg
logdumper --- Lookup folder '/'...
logdumper --- Found! / is 0x283db4db0
errordumper=========You need to specify a folder starting with /store_xxxxxxxxx/
logdumper --- You need to specify a folder starting with /store_xxxxxxxxx/

*** Contexterror ***              
You need to specify a folder starting with /store_xxxxxxxxx/

SiuyuenChow, 2022-05-09:
New message :

sony sdcard had two folder  -> 100MSDCF、101MSD11

call ----> gp_camera_folder_list_folders (camera, “/”, list, context);
call ---> gp_list_count (list) ----> folders count = 1 
call --> gp_list_get_name (list, i, &newfolder); ---> folder = store_00010001
call --> gp_camera_folder_list_files (cam, folder, list, ctx);
call --> gp_list_count (list) ---> file count = 0

From above, 100MSDCF and 101MSD11 are not found

msmeissn, 2022-05-09:
we did not set the info for the sdram captured files, i have added that now to current git. Please test :)

workaround in your code if you cannot update would be to avoid the get_info call

SiuyuenChow, 2022-05-09:
> we did not set the info for the sdram captured files, i have added that now to current git. Please test :)
> 
> workaround in your code if you cannot update would be to avoid the get_info call

----
If I don't update the code for gphoto2 soon,
I need to get file information. How do I get it?

SiuyuenChow, 2022-05-10:
> we did not set the info for the sdram captured files, i have added that now to current git. Please test :)
> 
> workaround in your code if you cannot update would be to avoid the get_info call

-------------
                /* we also get the fs info for free, so just set it */
                info.file.fields = GP_FILE_INFO_TYPE |
                        GP_FILE_INFO_WIDTH | GP_FILE_INFO_HEIGHT |
                        GP_FILE_INFO_SIZE | GP_FILE_INFO_MTIME;
                strcpy_mime (info.file.type, params->deviceinfo.VendorExtensionID, oi.ObjectFormat);
                info.file.width        = oi.ImagePixWidth;
                info.file.height    = oi.ImagePixHeight;
                info.file.size        = oi.ObjectCompressedSize;
                info.file.mtime        = time(NULL);

                info.preview.fields = GP_FILE_INFO_TYPE |
                        GP_FILE_INFO_WIDTH | GP_FILE_INFO_HEIGHT |
                        GP_FILE_INFO_SIZE;
                strcpy_mime (info.preview.type, params->deviceinfo.VendorExtensionID, oi.ThumbFormat);
                info.preview.width    = oi.ThumbPixWidth;
                info.preview.height    = oi.ThumbPixHeight;
                info.preview.size    = oi.ThumbCompressedSize;
                GP_LOG_D ("setting fileinfo in fs");
                gp_filesystem_set_info_noop(camera->fs, path->folder, path->name, info, context);
-----

I added the code snippet above, and the result is the same.
---------

here is the log :
ptp_sony_getalldevicepropdesc: Objects in memory(d215): value 1 -> 32769
DEBUG== 0xd215 after capture = 32769
SONY ObjectInMemory count change seen, retrieving file
Sending PTP_OC 0x1008 (Get object info) (0xffffc001) request...
Reading PTP_OC 0x1008 (Get object info) data...
Reading PTP_OC 0x1008 (Get object info) response...
PTPIP_END_DATA_PACKET
PTPIP_CMD_RESPONSE
ObjectInfo for 'DSC00090.JPG':
  Object ID: 0xffffc001
  StorageID: 0x00010000
  ObjectFormat: 0x3801
  ProtectionStatus: 0x0000
  ObjectCompressedSize: 273332
  ThumbFormat: 0x0000
  ThumbCompressedSize: 0
  ThumbPixWidth: 0
  ThumbPixHeight: 0
  ImagePixWidth: 0
  ImagePixHeight: 0
  ImageBitDepth: 0
  ParentObject: 0x00000000
  AssociationType: 0x0000
  AssociationDesc: 0x00000000
  SequenceNumber: 0x00000000
  ModificationDate: 0x00000000
  CaptureDate: 0x00000000
trying to get object size=0x42bb4
Sending PTP_OC 0x1009 (Get object) (0xffffc001) request...
hdr type 8, length 18
Reading PTP_OC 0x1009 (Get object) data...
Reading PTP_OC 0x1009 (Get object) response...
hdr type 8, length 18
PTPIP_END_DATA_PACKET
hdr type 8, length 18
PTPIP_CMD_RESPONSE
Append //capt0000.jpg to filesystem
Lookup folder '/'...
Found! / is 0x283ea4960
Listing files in /
Lookup folder '/'...
Found! / is 0x283ea4960
Querying folder /...
Delete all files in folder 0x283ea4960//
file_list_func(/)
Internal append capt0000.jpg to folder /
Adding file 'capt0000.jpg' to folder '/' (type 1)...
Lookup folder / file capt0000.jpg
Lookup folder '/'...
Found! / is 0x283ea4960
Adding file 'capt0000.jpg' from folder '/' to the fscache LRU list (type 1)...
Lookup folder / file capt0000.jpg
Lookup folder '/'...
Found! / is 0x283ea4960
File 'capt0000.jpg' from folder '/' added in fscache LRU list.
Checking fscache LRU list integrity...
fscache LRU list ok with 1 items (273332 bytes)
Failed to find mime type for 0000
setting fileinfo in fs
Lookup folder / file capt0000.jpg
Lookup folder '/'...
Found! / is 0x283ea4960
2022-05-10 15:25:32.399756+0800 GPhoto2Example[80052:4244753] gp_camera_file_get_info==========/===capt0000.jpg
Getting file info for 'capt0000.jpg' in '/'...
Getting information about 'capt0000.jpg' in '/'...
Lookup folder / file capt0000.jpg
Lookup folder '/'...
Found! / is 0x283ea4960
2022-05-10 15:25:32.402189+0800 GPhoto2Example[80052:4244753]  Downloading ...
Getting file 'capt0000.jpg' in folder '/'...
Getting file 'capt0000.jpg' from folder '/' (type 1)...
Lookup folder / file capt0000.jpg
Lookup folder '/'...
Found! / is 0x283ea4960
Copying '' onto 'capt0000.jpg'...
Downloading 'capt0000.jpg' from folder '/'...
You need to specify a folder starting with /store_xxxxxxxxx/
You need to specify a folder starting with /store_xxxxxxxxx/

*** Contexterror ***              
You need to specify a folder starting with /store_xxxxxxxxx/
Download of 'capt0000.jpg' from '/' (type 1) failed. Reason: 'Unspecified error'
'gp_filesystem_get_file (camera->fs, folder, file, type, camera_file, context)' failed: -1
'gp_filesystem_get_file (camera->fs, folder, file, type, camera_file, context)' failed: -1
---------------

SiuyuenChow, 2022-05-12:
update message
I found some information
   ParentObject: 0x00000000
----

ObjectInfo for 'DSC00150.JPG':
  Object ID: 0xffffc001
  StorageID: 0x00010000
  ObjectFormat: 0x3801
  ProtectionStatus: 0x0000
  ObjectCompressedSize: 327733
  ThumbFormat: 0x0000
  ThumbCompressedSize: 0
  ThumbPixWidth: 0
  ThumbPixHeight: 0
  ImagePixWidth: 0
  ImagePixHeight: 0
  ImageBitDepth: 0
  ParentObject: 0x00000000
  AssociationType: 0x0000
  AssociationDesc: 0x00000000
  SequenceNumber: 0x00000000
  ModificationDate: 0x00000000
  CaptureDate: 0x00000000

SiuyuenChow, 2022-05-12:
cannon PTPObjectInfo :
(PTPCanon_changes_entry) entry = {
  type = PTP_CANON_EOS_CHANGES_TYPE_OBJECTINFO
  u = {
    object = {
      oid = 2442287666
      oi = {
        StorageID = 131073
        ObjectFormat = 14337
        ProtectionStatus = 0
        ObjectCompressedSize = 208487
        ThumbFormat = 0
        ThumbCompressedSize = 0
        ThumbPixWidth = 0
        ThumbPixHeight = 0
        ImagePixWidth = 0
        ImagePixHeight = 0
        ImageBitDepth = 0
        ParentObject = 2442133504
        AssociationType = 0
        AssociationDesc = 0
        SequenceNumber = 0
        Filename = 0x00000002827502b0 "IMG_9635.JPG"
        CaptureDate = 0
        ModificationDate = 0
        Keywords = 0x0000000000000000
      }
    }
    info = 0x0000000091925a32 ""
    propid = 23090
    status = -1852679630
  }
}

------
sony PTPObjectInfo : 
(PTPObjectInfo) oi = {
  StorageID = 65536
  ObjectFormat = 14337
  ProtectionStatus = 0
  ObjectCompressedSize = 316688
  ThumbFormat = 0
  ThumbCompressedSize = 0
  ThumbPixWidth = 0
  ThumbPixHeight = 0
  ImagePixWidth = 0
  ImagePixHeight = 0
  ImageBitDepth = 0
  ParentObject = 0
  AssociationType = 0
  AssociationDesc = 0
  SequenceNumber = 0
  Filename = 0x0000000280ca8170 "DSC00152.JPG"
  CaptureDate = 0
  ModificationDate = 0
  Keywords = 0x0000000000000000
}

----
It just looks like the ParentObject parameter is different

SiuyuenChow, 2022-05-14:
@msmeissn hello,
I have no idea to solve this problem, can you provide me with the logic of thinking?
thanks
Best wishes

msmeissn, 2022-10-21:
in virtual sdram capture mode, the images are in /capt0001.jpg and similar named, so the comment is not quite correct.

msmeissn, 2022-10-21:
FWIW The capture API returns the path/folder of the captured image, which is then directly downloaded (no need for listing).

Entropy512, 2022-10-21:
Yeah, that was working on A7M3 for me, I'm having some issues with M4 but I think it's a completely different issue to what @SiuyuenChow is having, I've been looking through issues to see if it's a known one, but I think mine is different, will need to take more diagnostics before filing an issue.

I plan on filing a separate issue with more logs next week, but have you ever seen a "stale" ObjectAdded event from a capture come in immediately after a second capture?

First capture and download works fine - I see the capture triggered, followed by some waiting for ObjectAdded, which returns a file handle that can be successfully downloaded.

Whether or not I delete the first file, the second capture and download fails - an ObjectAdded event happens immediately with no delay, with the same file handle as the previous capture.  GetObject fails with "Invalid object" leading to a segfault (the segfault may be a flaw in the Python bindings and not gphoto2 itself).

Have you ever seen anything like that?  I'll provide more data next week once my family leaves (they are visiting all weekend)

msmeissn, 2022-10-22:
please open a new issue in this case.

the sony have the issue that there is also the "objects in SDRAM" property which needs to be honored beside the objectadded event.

i dont remember stale events, but it could happen.