gphoto/libgphoto2 #798: call [gp_camera_folder_list_folders ] Unable to find folder

SiuyuenChow, 2022-05-17:
model  ---->  SONY camera A7M4

There are three REC folders inside the camera, which are 100MSDCF, 101MSD11 and 102MSDCF

When I call gp_camerA_FOLder_list_folders, only one is returned

here is the log：

Listing folders in '/'...
Listing folders in /
Lookup folder '/'...
Found! / is 0x282698c60
... is dirty, getting from camera
folder_list_func(/)
Internally deleting all folders from '/'...
Lookup folder '/'...
Found! / is 0x282698c60
Recurse delete folder 0x282698c60//
Append one folder store_00010001
Folder / contains 1 subfolders.
Sending PTP_OC 0x1016 (Set device property value) (0xd228) request...
Sending PTP_OC 0x1016 (Set device property value) data...
Reading PTP_OC 0x1016 (Set device property value) response...

msmeissn, 2022-05-18:
For Sony cameras, in pc control mode there is no folder access. It is only accessible in MTP/PTP mode.

SiuyuenChow, 2022-05-18:
> For Sony cameras, in pc control mode there is no folder access. It is only accessible in MTP/PTP mode.

Do I need to set gp_abilities_list_lookup_model to MTP/PTP?

SiuyuenChow, 2022-05-18:
> For Sony cameras, in pc control mode there is no folder access. It is only accessible in MTP/PTP mode.

@msmeissn 
Is it convenient to communicate by email? This is my email address :im_2y@126.com

msmeissn, 2022-05-18:
afaik the mode can only be switched on the camera menu, not over USB. (or perhaps it is possible, but we do not know the protocol codes to emit)

SiuyuenChow, 2022-05-18:
> afaik the mode can only be switched on the camera menu, not over USB. (or perhaps it is possible, but we do not know the protocol codes to emit)

At present, I set it through PTP/IP Camera, which can be connected. Instead of using USB, you get data directly through WiFi