gphoto/libgphoto2 #799: Error setting manual focus drive to Canon Powershot

gabrielesartor, 2022-05-19:
**Describe the bug**
I'm trying to set the manual focus drive for my camera (Canon Powershot 740HS), but it seems not working.
I've tried to set it using gphoto from terminal and this library with Python, but nothing changed. Could it be the definition of this functionality in gphoto for my specific camera?

**Name the camera**
```Canon PowerShot SX740 HS       usb:001,004```

**libgphoto2 and gphoto2 version**
```
gphoto2         2.5.27.1       gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
```

**To Reproduce**
I've tried in different ways to set it:
1. using terminal
```gphoto2 --set-config-value=/main/actions/autofocusdrive=1 # or other higher numbers```
I've tried also to use string, so ```gphoto2 --set-config-value=/main/actions/autofocusdrive="1"``` but nothing changed
```gphoto2 --set-config-value=/main/actions/autofocusdrive=0``` seems "working" with no errors but it doesn't change the settings.

The result is:
```*** Error ***              
Failed to set new configuration value 1 for configuration entry /main/actions/autofocusdrive.
*** Error (-1: 'Unspecified error') ***       
```

2. using Python, here the snippet:
```
    camera = gp.Camera()
    camera.init()
    sleep(2)

    camera_config = camera.get_config()
    print(camera_config)
    OK, config_widget = gp.gp_widget_get_child_by_name(camera_config,"manualfocusdrive")
    print(config_widget)

    # set value
    config_widget.set_value("1")

    # set config
    #gp.gp_camera_set_config(camera, camera_config)
    camera.set_config(camera_config)
    print("setting focus...")
```

The result is:
```WARNING: gphoto2: (gp_camera_set_config [gphoto2-camera.c:1115]) 'camera->functions->set_config (camera, window, context)' failed: -1
Traceback (most recent call last):
  File "src/camera/scripts/tests/set_focus.py", line 47, in <module>
    sys.exit(main())
  File "src/camera/scripts/tests/set_focus.py", line 30, in main
    camera.set_config(camera_config)
gphoto2.GPhoto2Error: [-1] Unspecified error
```
I report also the part of ```gphoto2 --list-all-config``` which can help to find the cause:
```
/main/actions/autofocusdrive
Label: Drive Canon DSLR Autofocus
Readonly: 0
Type: TOGGLE
Current: 0
END
/main/actions/manualfocusdrive
Label: Drive Canon DSLR Manual focus
Readonly: 0
Type: RADIO
Current: None
Choice: 0 Near 1
Choice: 1 Near 2
Choice: 2 Near 3
Choice: 3 None
Choice: 4 Far 1
Choice: 5 Far 2
Choice: 6 Far 3
END
/main/actions/cancelautofocus
Label: Cancel Canon DSLR Autofocus
Readonly: 0
Type: TOGGLE
Current: 0
END
```
Could it be the type RADIO? 
Thank you in advance!

msmeissn, 2022-05-19:
note: autofocusdrive is an action ... it is executed once when set to 1.

manualfocusdrive only works in liveview mode. "1" is not a valid value, you need to use one of the choices ... "Near 1" ... "Far 3"

hfiguiere, 2022-05-19:
Your libgphoto2 is old. Probably too old.

gabrielesartor, 2022-05-19:
Thank you for the feedback @msmeissn 
Sorry, but I didn't get the point, what do you suggest to do? :)
Do you need additional information by my side?