gphoto/libgphoto2 #800: Undefined symbol error with libgphoto2 built from source

Stephan-XUE, 2022-05-27:
I have gitted the latest version of libgphoto2 and gphoto2 and build them from source.
Running the `gphoto2 --version` command shows follows
```
$ gphoto2 --version
gphoto2 2.5.28.1

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.29.1       standard camlibs (SKIPPING lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

But when I try to call the libgphoto2 library with command like `gphoto2 --auto-detect`, it shows undefined symbol error:
``` 
$ gphoto2 --auto-detect 
gphoto2: symbol lookup error: /usr/local/lib/libgphoto2.so.6: undefined symbol: gp_port_init_localedir, version LIBGPHOTO2_5_0
```

Can anyone tells me this problem is because of something wrong with the build process  or do I miss some dependence?
Thanks a lot!

Sija, 2022-05-28:
It seems the `libgphoto2_port` library needs to be updated, as the missing function was added in the latest release.

ndim, 2022-05-28:
I have just rebuilt and reinstalled the current git HEAD into $HOME/.local for a test (https://github.com/gphoto/libgphoto2/tree/85f2b9fdf57d41ccc210b6c0b0209c03c9de41a9 and https://github.com/gphoto/gphoto2/tree/959b9018b25d50347f8faadf6408f2f2022bf1e6, respectively):

```
cd ~/src/gphoto-stuff
mkdir _b-libgphoto2
cd _b-libgphoto2
autoreconf -vis ../libgphoto2
../libgphoto2/configure --with-camlibs=everything --prefix=$HOME/.local
make -j$(nproc)
make -j$(nproc) check
make -j$(nproc) install
make -j$(nproc) installcheck

cd ~/src/gphoto-stuff
mkdir _b-gphoto2
cd _b-gphoto2
autoreconf -vis ../gphoto2
../gphoto2/configure PKG_CONFIG_PATH=$HOME/.local/lib/pkgconfig --prefix=$HOME/.local
make -j$(nproc)
make -j$(nproc) check
make -j$(nproc) install
make -j$(nproc) installcheck
```

```
$ ~/.local/bin/gphoto2 --version
gphoto2 2.5.28.1
[...]
$ ~/.local/bin/gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
$ _
```

So my guess is you have built with some weird option and are therefore running gphoto2 linked against an up to date libgphoto2 and an out of date (system provided maybe?) libgphoto2_port.

@Stephan-XUE What does ldd say?

```
$ ldd ~/.local/bin/gphoto2 | grep gphoto2
	libgphoto2.so.6 => /home/USER/.local/lib/libgphoto2.so.6 (0x00007f5757eaf000)
	libgphoto2_port.so.12 => /home/USER/.local/lib/libgphoto2_port.so.12 (0x00007f5757ea1000)
$ _
```

Stephan-XUE, 2022-05-30:
@[ndim](https://github.com/ndim)
Thanks for your help, and sorry for the delayed reply.
I didn't build `gphoto2` to `$HOME/.local` but to `/usr/local`, together with `libgphoto2.so.6`
Here is the ldd says for my `gphoto2`:
```
$ ldd /usr/local/bin/gphoto2 | grep gphoto2
	libgphoto2.so.6 => /usr/local/lib/libgphoto2.so.6 (0xb6edb000)
	libgphoto2_port.so.12 => /lib/arm-linux-gnueabihf/libgphoto2_port.so.12 (0xb6ec1000)
```
----------------------------------------------------
Problem is solved by removing outdated `libgphoto2_port` previously installed with `apt`

closed 2022-05-30 by Stephan-XUE