gphoto/libgphoto2 #803: Nikon 1 J3 with manual lens: capture doesn't respect shutter speed

rainbow-app, 2022-06-04:
Hello!

Lens is manual (=fully mechanical, no electronics).

I can successfully set shutter speed, e.g. to 1/10s, with "--set-config-value shutterspeed2=1/10" (I can see it both in "--get-config shutterspeed2", and on camera screen when I unplug USB cable).

However, when I --capture-image, it gets captured with some *auto* exposure, like 6s if dark, or 1s if brighter.

Latest master.

If I press the physical button, the expected speed (1/10) is used (of course). I (still) don't have native Nikon lens, so cannot check if the problem is in the lack of communication with the lens. (also I don't have the old J1 anymore, so can't check that one, sorry).

I'll attach debug log in the following message.

Thanks.

```
Model                          Port
----------------------------------------------------------
Nikon J3                       usb:001,044

gphoto2 -v --get-config shutterspeed2
gphoto2 2.5.28.1

...

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28.1       gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.29.1       standard camlibs (SKIPPING docupen lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.1         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
Label: Shutterspeed
Readonly: 0
Type: RADIO
Current: 1/10
...

```

rainbow-app, 2022-06-04:
0.000138 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000192 main                        (2): gphoto2 2.5.28.1
0.000203 main                        (2): gphoto2 has been compiled with the following options:
0.000210 main                        (2):  + gcc (C compiler used)
0.000216 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000222 main                        (2):  + exif (for displaying EXIF information)
0.000228 main                        (2):  + no cdk (for accessing configuration options)
0.000234 main                        (2):  + no aa (for displaying live previews)
0.000239 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000245 main                        (2):  + readline (for easy navigation in the shell)
0.000256 main                        (2): libgphoto2 2.5.29.1
0.000265 main                        (2): libgphoto2 has been compiled with the following options:
0.000273 main                        (2):  + standard camlib set SKIPPING SOME (ax203 canon digigr8 dimagev directory jl2005a jl2005c kodak_dc240 mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801 SKIPPING docupen lumix)
0.000279 main                        (2):  + gcc (C compiler used)
0.000285 main                        (2):  + no ltdl (for portable loading of camlibs)
0.000291 main                        (2):  + EXIF (for special handling of EXIF files)
0.000301 main                        (2): libgphoto2_port 0.12.1
0.000309 main                        (2): libgphoto2_port has been compiled with the following options:
0.000316 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000321 main                        (2):  + gcc (C compiler used)
0.000326 main                        (2):  + no ltdl (for portable loading of iolibs)
0.000332 main                        (2):  + EXIF (for vusb)
0.000338 main                        (2):  + USB (libusb1, for USB cameras)
0.000343 main                        (2):  + serial (for serial cameras)
0.000349 main                        (2):  + no resmgr (serial port access and locking)
0.000354 main                        (2):  + no ttylock (serial port locking)
0.000359 main                        (2):  + no lockdev (serial port locking)
0.000365 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000371 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000377 main                        (2): invoked with following arguments:
0.000385 main                        (2):   --debug-logfile=Capture.log
0.000391 main                        (2):   --debug
0.000396 main                        (2):   --capture-image
0.000439 load_settings               (2): Creating gphoto config directory ('/home/user/.gphoto')
0.000510 load_settings               (2): Loading settings from file '/home/user/.gphoto/settings'.
0.000580 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.000594 gp_port_init_localedir      (2): localedir has been set to /usr/local/share/locale (compile-time default)
0.000605 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/local/lib/libgphoto2_port/0.12.1'...
0.000721 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.1/disk'.
0.000970 gp_port_library_list        (2): found fstab fsname UUID=b9bf53d7-e0d1-4ada-9c26-edd4748f5446
0.001111 gp_port_library_list        (2): found mtab fsname sysfs
0.001123 gp_port_library_list        (2): found mtab fsname proc
0.001133 gp_port_library_list        (2): found mtab fsname udev
0.001157 gp_port_library_list        (2): found mtab fsname devpts
0.001168 gp_port_library_list        (2): found mtab fsname tmpfs
0.001191 gp_port_library_list        (2): found mtab fsname tmpfs
0.001213 gp_port_library_list        (2): found mtab fsname /dev/mapper/m5fvg-m5flv
0.001242 gp_port_library_list        (2): found mtab fsname overlay
0.001259 gp_port_library_list        (2): found mtab fsname tmpfs
0.001282 gp_port_library_list        (2): found mtab fsname securityfs
0.001292 gp_port_library_list        (2): found mtab fsname pstore
0.001316 gp_port_library_list        (2): found mtab fsname tmpfs
0.001340 gp_port_library_list        (2): found mtab fsname /dev/sda2
0.001359 gp_port_library_list        (2): found mtab fsname /dev/sdc1
0.001383 gp_port_library_list        (2): found mtab fsname /dev/sdb3
0.001440 foreach_func                (2): Loaded '' ('^disk:') from '/usr/local/lib/libgphoto2_port/0.12.1/disk'.
0.001447 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.1/ptpip'.
0.001656 foreach_func                (2): Loaded 'PTP/IP Connection' ('ptpip:') from '/usr/local/lib/libgphoto2_port/0.12.1/ptpip'.
0.001666 foreach_func                (2): Loaded '' ('^ptpip:') from '/usr/local/lib/libgphoto2_port/0.12.1/ptpip'.
0.001673 foreach_func                (2): Loaded 'IP Connection' ('ip:') from '/usr/local/lib/libgphoto2_port/0.12.1/ptpip'.
0.001680 foreach_func                (2): Loaded '' ('^ip:') from '/usr/local/lib/libgphoto2_port/0.12.1/ptpip'.
0.001686 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.1/serial'.
[... cut ...]
12.110887 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
12.110895 gp_filesystem_lru_clear     (2): fscache LRU list already empty
12.110904 delete_all_folders          (2): Internally deleting all folders from '/'...
12.110911 lookup_folder               (2): Lookup folder '/'...
12.110918 lookup_folder               (2): Found! / is 0x55b8d84fe820
12.110925 recurse_delete_folder       (2): Recurse delete folder 0x55b8d84fe820//
12.110932 delete_all_files            (2): Delete all files in folder 0x55b8d84fe820//

rainbow-app, 2022-07-15:
A minor update:
- Electronic (kit) lens doesn't matter, same problem.
- J1 doesn't have the shutter speed setting support in gphoto, it simply captures (with correct pre-set shutter speed).

msmeissn, 2022-07-15:
shutterspeed is not really a lens property, but a camera property.. so should be lens independend