gphoto/libgphoto2 #807: Sudden problem with compiled gphoto applications not running. Error Loading a Library

richbai90, 2022-06-10:
**Describe the bug**

Not really a bug but I don't know where to get support specifically for `libgphoto2`. If I am in the wrong place, please direct me to the correct one.

I am faced with the sudden inability to run a program compiled with libgphoto2. The build succeeds without any errors, but reports `Error Loading a Library` This happens with code that was working previously, and even example code from the repo. It started after I attempted to compile the library from source with debug flags so I could step through and understand the internals of the library better. Even after removing the library and reinstalling the version from the package manager ( apt : libgphoto2-dev/stable ) I am faced with the same error. Telling is the fact that if I run `print-camera-list idlist` I get an empty response. I think for some reason the camera libraries aren't linked correctly to the main library, but that's just my hunch

**Name the camera**
Camera is not detected by gphoto2 either, which only just started happening with this same issue. `lsusb` reports that the camera is connected.

`Bus 001 Device 003: ID 04b0:041c Nikon Corp. NIKON DSC D3`

**libgphoto2 and gphoto2 version**

```
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.29.1       NON-STANDARD CAMLIB SET (adc65 agfa_cl20 aox ax203 barbie canon casio_qv clicksmart310 digigr8 digita dimagev dimera3500 directory enigma13 fuji gsmart300 hp215 iclick jamcam jd11 jl2005a jl2005c kodak_dc120 kodak_dc210 kodak_dc240 kodak_dc3200 kodak_ez200 konica konica_qm150 largan lg_gsm mars mustek panasonic_coolshot panasonic_dc1000 panasonic_dc1580 panasonic_l859 pccam300 pccam600 pentax polaroid_pdc320 polaroid_pdc640 polaroid_pdc700 ptp2 ricoh ricoh_g3 samsung sierra sipix_blink sipix_blink2 sipix_web2 smal sonix sony_dscf1 sony_dscf55 soundvision spca50x sq905 st2205 stv0674 stv0680 sx330z template topfield toshiba_pdrm11 tp6801 SKIPPING docupen lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.1         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
```
I tried compiling all supported camera libraries to see if this would correct the issue. It did not.

**To Reproduce**
Not sure, but try first by compiling from source and removing the installed libraries via the package manager, and then any remaining instances by hand (`find / | grep libgphoto | xargs -I {} sudo rm -rf {}`)

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

rainbow-app, 2022-06-10:
I'm not a gphoto dev, just some general ideas. See if libs are fine using `ldd /usr/local/bin/gphoto2`. Also, running `ldconfig` may help.

richbai90, 2022-06-11:
Thanks for the advice, it looks like the libraries are all properly loaded though. The output from `ldd` shows:

```
	linux-vdso.so.1 (0xbedc3000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0xb6fa2000)
	libgphoto2.so.6 => /lib/arm-linux-gnueabihf/libgphoto2.so.6 (0xb6efb000)
	libgphoto2_port.so.12 => /lib/arm-linux-gnueabihf/libgphoto2_port.so.12 (0xb6ee1000)
	libcdk.so.5 => /lib/arm-linux-gnueabihf/libcdk.so.5 (0xb6e9f000)
	libncurses.so.6 => /lib/arm-linux-gnueabihf/libncurses.so.6 (0xb6e70000)
	libaa.so.1 => /lib/arm-linux-gnueabihf/libaa.so.1 (0xb6e45000)
	libjpeg.so.62 => /lib/arm-linux-gnueabihf/libjpeg.so.62 (0xb6dfe000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb6dd2000)
	libexif.so.12 => /lib/arm-linux-gnueabihf/libexif.so.12 (0xb6d8d000)
	libreadline.so.8 => /lib/arm-linux-gnueabihf/libreadline.so.8 (0xb6d3a000)
	libpopt.so.0 => /lib/arm-linux-gnueabihf/libpopt.so.0 (0xb6d1f000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb6cb0000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6b5c000)
	/lib/ld-linux-armhf.so.3 (0xb6fb7000)
	libltdl.so.7 => /lib/arm-linux-gnueabihf/libltdl.so.7 (0xb6b44000)
	libtinfo.so.6 => /lib/arm-linux-gnueabihf/libtinfo.so.6 (0xb6b12000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb6afe000)
	libslang.so.2 => /lib/arm-linux-gnueabihf/libslang.so.2 (0xb68ec000)
	libX11.so.6 => /lib/arm-linux-gnueabihf/libX11.so.6 (0xb67c6000)
	libgpm.so.2 => /lib/arm-linux-gnueabihf/libgpm.so.2 (0xb67b0000)
	libxcb.so.1 => /lib/arm-linux-gnueabihf/libxcb.so.1 (0xb677f000)
	libXau.so.6 => /lib/arm-linux-gnueabihf/libXau.so.6 (0xb676c000)
	libXdmcp.so.6 => /lib/arm-linux-gnueabihf/libXdmcp.so.6 (0xb6757000)
	libbsd.so.0 => /lib/arm-linux-gnueabihf/libbsd.so.0 (0xb6735000)
	libmd.so.0 => /lib/arm-linux-gnueabihf/libmd.so.0 (0xb671a000)
```

Running ldconfig doesn't seem to change anything either.

richbai90, 2022-06-11:
Okay I got it fixed. Here's what I did

closed 2022-06-11 by richbai90

Sija, 2022-06-11:
> rm -rf / | grep libgphoto | xargs -I {} rm -rf {}

is this supposed to erase your main partition or what? 😅

richbai90, 2022-06-11:
Lol! Yeah I guess so 😬. I fixed it in case someone opts to copy/paste without checking first lol. Thanks for the catch. > > rm -rf / | grep libgphoto | xargs -I {} rm -rf {}
> 
> @richbai90 is this supposed to erase your main partition or what? 😅

ndim, 2022-06-12:
I have two questions:
  * What operating system ist that? It ships a separate `libgphoto2-port12` package, and I am interested about that.
  * Have you compiled a newer libgphoto2 source, and installed it by just overwriting the files the operating system package ships with?

richbai90, 2022-06-12:
> I have two questions:
>   * What operating system ist that? It ships a separate `libgphoto2-port12` package, and I am interested about that.

Rasbian 

>   * Have you compiled a newer libgphoto2 source, and installed it by just overwriting the files the operating system package ships with?

I haven't tried just straight overwriting, I used the configure prefix option with make install.

ndim, 2022-06-12:
> Rasbian
>
>
> I haven't tried just straight overwriting, I used the configure prefix option with make install.

Well, building and installing libgphoto2 with `configure --prefix=/usr && make && sudo make install` *is* straight overwriting, and would explain your above `gphoto2 --version` output showing 2.5.27 for gphoto2, 2.5.29.1 for libgphoto2, and 0.12.1 for libgphoto2_port.

Just trying to understand what (might have) happened.