gphoto/libgphoto2 #808: EOS 90d - gphoto2 –capture-image -> -1: »Unspecified error«

staffapply, 2022-06-29:
**Describe the bug**

When using: gphoto2 –capture-image
i get this error:

*** Fehler (-1: »Unspecified error«) ***      

**Name the camera**
Modell                         Port                                            
----------------------------------------------------------
Canon EOS 90D                  usb:020,010  

**libgphoto2 and gphoto2 version**
gphoto2 2.5.28

Copyright (C) 2000-2021 Marcus Meißner und andere

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

Diese Version von gphoto2 benutzt die folgenden Softwareversionen und Optionen:
gphoto2         2.5.28         clang, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.29         standard camlibs, clang, no ltdl, no EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1, clang, no ltdl, no EXIF, USB, serial without locking

**To Reproduce**
Steps to reproduce the behavior:

0.001154 main                        (2): BITTE IMMER FOLGENDE ZEILE EINFÜGEN WENN DEBUG MESSAGES AN DIE MAILINGLISTE GESCHICKT WERDEN:
0.001218 main                        (2): gphoto2 2.5.28
0.001235 main                        (2): gphoto2 wurde mit den folgenden Optionen übersetzt:
0.001245 main                        (2):  + clang (C compiler used)
0.001253 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.001261 main                        (2):  + exif (for displaying EXIF information)
0.001269 main                        (2):  + no cdk (for accessing configuration options)
0.001276 main                        (2):  + no aa (for displaying live previews)
0.001284 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.001292 main                        (2):  + readline (for easy navigation in the shell)
0.001305 main                        (2): libgphoto2 2.5.29
0.001314 main                        (2): libgphoto2 wurde mit den folgenden Optionen übersetzt:
0.001326 main                        (2):  + standard camlib set (ax203 canon digigr8 dimagev directory docupen jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801)
0.001334 main                        (2):  + clang (C compiler used)
0.001341 main                        (2):  + no ltdl (for portable loading of camlibs)
0.001348 main                        (2):  + no EXIF (for special handling of EXIF files)
0.001356 main                        (2): libgphoto2_port 0.12.0
0.001364 main                        (2): libgphoto2_port wurde mit den folgenden Optionen übersetzt:
0.001371 main                        (2):  + iolibs: disk ptpip serial usb1
0.001378 main                        (2):  + clang (C compiler used)
0.001386 main                        (2):  + no ltdl (for portable loading of iolibs)
0.001393 main                        (2):  + no EXIF (for vusb)
0.001400 main                        (2):  + USB (libusb1, for USB cameras)
0.001407 main                        (2):  + serial (for serial cameras)
0.001415 main                        (2):  + no resmgr (serial port access and locking)
0.001422 main                        (2):  + no ttylock (serial port locking)
0.001429 main                        (2):  + no lockdev (serial port locking)
0.001437 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.001444 main                        (2): IOLIBS env var not set, using compile-time default instead
0.001452 main                        (2): invoked with following arguments:
0.001459 main                        (2):   –capture-image
0.001466 main                        (2):   env
0.001474 main                        (2):   LANG=C
0.001481 main                        (2):   gphoto2
0.001488 main                        (2):   --debug
0.001496 main                        (2):   --debug-logfile=my-logfile.txt
0.001503 main                        (2):   –capture-image
0.001520 load_settings               (2): Creating gphoto config directory ('/Users/maikbecker/.gphoto')
0.002075 load_settings               (2): Loading settings from file '/Users/maikbecker/.gphoto/settings'.
0.002261 main                        (2): Uploading file...
0.002328 gp_camera_free              (2): Freeing camera...
0.002340 gp_port_free                (2): Freeing port...
0.002354 gp_filesystem_reset         (2): resetting filesystem
0.002365 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.002373 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.002381 delete_all_folders          (2): Internally deleting all folders from '/'...
0.002389 lookup_folder               (2): Lookup folder '/'...
0.002398 lookup_folder               (2): Found! / is 0x600002d24390
0.002408 recurse_delete_folder       (2): Recurse delete folder 0x600002d24390//
0.002417 delete_all_files            (2): Delete all files in folder 0x600002d24390//

msmeissn, 2022-06-30:
it is --capture-image (with a double - in front) ... you had only one "-" in front.

staffapply, 2022-06-30:
Hi,
sorry but this is not solved.

gphoto2 –capture-image
*** Fehler (-1: »Unspecified error«) ***      

return the same error

msmeissn, 2022-06-30:
you need to specify two - minus signs in front capture-image, in your example it is still just 1.

msmeissn, 2022-07-01:
can you provide another debuglogfile of the last?