gphoto/libgphoto2 #809: Olympus E-M5 MarkIII, PTP Device Prop Not Supported

brettowe, 2022-06-29:
**Describe the bug**
while trying to capture an image via gphoto2 command line I get a PTP Device Prop Not Supported error

**Name the camera**
Olympus OM-D E-M5 MarkIII
auto-detect shows
Olympus E-M5                   usb:001,008

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version
gphoto2         2.5.27         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.27         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

attached is output of --config and --debug from trying --capture-image

[em5m3-config.txt](https://github.com/gphoto/libgphoto2/files/9014721/em5m3-config.txt)
[em5m3-debuglog.txt](https://github.com/gphoto/libgphoto2/files/9014722/em5m3-debuglog.txt)

msmeissn, 2022-07-02:
when i compare the USB Modes of the M5 Mark 2 and Mark 3 ... there is a non-MTP computer connect in the mark 2, but not the mark 3... they might have removed the remote control abilities.

brettowe, 2022-07-02:
looking at the manual i only see mention of the app for remote control. 
does gphoto do network based remote control?
https://cs.olympus-imaging.jp/en/support/imsg/digicamera/download/manual/omd/man_em5m3_e.pdf

msmeissn, 2022-07-16:
We impelemented PTP/IP if the camera supports that ... 

For Lumix we support the Lumix specific http interface.

I am not sure the Olympus talks PTP/IP, its probably yet another HTTP based protocol.

I find with Google some file download over wifi things, 
https://www.dpreview.com/forums/thread/4272680
https://github.com/mauriciojost/olympus-photosync

here is a page with remote control on other olympus models:
https://rebs.biz/cc/en/index.html