gphoto/libgphoto2 #814: Cannot open session to Canon with v2.5.28 (Windows build)

manongjohn, 2022-07-23:
**Describe the bug**

I know Windows builds aren't really supported but I have built a Win64 version of libgphoto2 using MSVC 2019 and integrated into my application.  I can normally connect and control my camera with v2.5.27, but cannot connect to the camera when I use v2.5.28  or later. 

I've narrowed the issue to commit [b6fae4fcceb3d3352bf24728045293d694a3e835](https://github.com/gphoto/libgphoto2/commit/b6fae4fcceb3d3352bf24728045293d694a3e835).  I verified I could still connect/control my camera with the commit just before this one, but breaks when this commit is present.

Here are excerpts of my application logs with GP_LOG_ALL enabled as it tries to open sessions:
[v2.5.27-working.txt](https://github.com/gphoto/libgphoto2/files/9174663/v2.5.27-working.txt)
[v2.5.28-broken.txt](https://github.com/gphoto/libgphoto2/files/9174665/v2.5.28-broken.txt)

Doesn't appear to be writing the correct amount of bytes to the camera after this change? Was working when it was writing 16 bytes but with this commit it appears to only be writing 12 bytes to the camera.

I can provide more information if desired. Hopefully someone can help with this problem.

**Name the camera**
Camera recognized as `Canon EOS 1500D`. I had to change it's normal USB driver to `libusbK (v3.1.0.0)` in order to be able to control my camera with libgphoto2.

**libgphoto2 and gphoto2 version**
libgphoto2 v.2.5.28 and later

manongjohn, 2022-07-24:
Nevermind.  After some research, I found out I was a missing option when compiling in MSCV so it was not handling `__VA_ARGS__` correctly.  I had to make sure I had `/Zc:preprocessor` compiler switch enabled.

closed 2022-07-24 by manongjohn

ndim, 2022-07-25:
Hmm. I would prefer if the compile failed instead of producing buggy executable code. Let me keep this open to see what I have broken here.