gphoto/libgphoto2 #817: Bulb capture on Nikon D3

richbai90, 2022-08-08:
Bulb capture fails with `*** Error (-1: 'Unspecified error') *** ` when issuing this simple test script

```sh
#! /usr/bin/env bash
export LANG=C 
gphoto() {

gphoto2 --debug --debug-logfile="$@"
}

gphoto "set-bulb.txt" set-config bulb=1
gphoto "wait.txt" wait-event 100s
gphoto "unset-bulb.txt" set-config bulb=0
gphoto "wait-and-dl.txt" wait-event-and-download=5s
```

**Name the camera**
```
Model                          Port                                            
----------------------------------------------------------
Nikon D3 (PTP mode)            usb:001,010  
```

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version

```
Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.27         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

**To Reproduce**
Connect a Nikon D3 to a computer with gphoto2 2.5.27 installed, and attempt to run the script mentioned above.

Log files for each command in the script are attached
[wait-and-dl.txt](https://github.com/gphoto/libgphoto2/files/9285143/wait-and-dl.txt)
[wait.txt](https://github.com/gphoto/libgphoto2/files/9285144/wait.txt)
[unset-bulb.txt](https://github.com/gphoto/libgphoto2/files/9285145/unset-bulb.txt)
[set-bulb.txt](https://github.com/gphoto/libgphoto2/files/9285146/set-bulb.txt)

msmeissn, 2022-08-21:
did you try them on the same commandline?

exiting gphoto2 also exists some of the modes needed.

msmeissn, 2022-08-21:
gphoto2 --set-config bulb=1 --wait-event 100s --set-config bulb=0 --wait-event-and-download=5s

i think also the shutterspeed on camera needs to be in "bulb" setting already before doing this

richbai90, 2022-08-23:
I did try that, no error, but no response either. From what I could tell in the documentation, I don't think Nikon exposes the correct features to make blub work for their older cameras. The documentation says specifically bulb mode works for "newer nikon models". I'm not sure what qualifies as "newer", but I have a hunch the D3 isn't it.