gphoto/libgphoto2 #819: Nikon Z30

dean513, 2022-08-20:
Fill in fields below. If you don't know how, leave it free.

Nikon Z30 is the same camera as the nikon z50, just without the viewfinder. Darktable will not even open images from this camera. Could you PLEASE add support in Darktable for this camera ASAP.  I'm just sitting around waiting...

Use the Marketing name.

**USB IDs**
e.g. by running:   lsusb 

**camera summary output**
run

  gphoto2 --summary > summary.txt

attach summary.txt here

**camera configuration output**

gphoto2 --list-all-config > list-all-config.txt

and attach list-all-config.txt here

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
 
  gphoto2 --capture-preview

msmeissn, 2022-08-21:
1. can you fill in the usb id  by running "lsusb"
and perhaps also the other tests from the template?

2. what is gphoto2 --auto-detect showing
3. what is gphoto2 --capture-image-and-download   doing?

nah2015, 2023-01-04:
I don't want to make a new thread, so I am posting here to say I am also interested in Z30 support. I do not have the camera yet, but I am wondering if it works. I do not see it listed in the supported cameras list.

Thank you

msmeissn, 2023-01-08:
The Z30 will likely just work as-is, as camera support is largely class based and we support quite a numober of Nikon D and Z series already just fine.