gphoto/libgphoto2 #823: Olympus E-M5 Mark III questions

ontheway16, 2022-08-30:
Hello, this is not an issue, but more of a question.
I am planning to invest on M5 mark3, solely to use with gphoto2, from a RasPi device. I was hoping to be able to;
-remotely focus,
-take shot,
-access storage to copy/move files.
-if possible, have a live view (via raspi) during shooting.
-if possible, set some settings, like, aperture, shutter, focus mode etc.
The camera has MTP mode, according to the documentation. 

Can someone please confirm that libgphoto2 can handle above tasks problem free, with Olympus E-M5 Mark III ?
Since in my use case, the camera itself will be held in a box which prevent me to access to physical menu buttons, I will try to use it remotely.
I appreciate if someone clarifies above points, thanks in advance.

hfiguiere, 2022-08-30:
> I am planning to invest on Olympus M5 mark3, solely to use with libgphoto2, from a RasPi device.

Make sure there is a return policy that allow you to return the camera for a full refund. Nobody here is responsible whatsoever if whatever you want to do with the camera remote controlled doesn't work.

ontheway16, 2022-08-30:
> Make sure there is a return policy that allow you to return the camera for a full refund. Nobody here is responsible whatsoever if whatever you want to do with the camera remote controlled doesn't [...]

Of course, that was not my intention.

ontheway16, 2022-09-04:
Apparently, m5 mk3 has no USB tethering support, which answers my above questions.

msmeissn, 2022-09-05:
i think we saw an earlier report of this camera where we did also not see remote tether support.

ontheway16, 2022-09-05:
Very unfortunate for mark III, but it seems they pruned that functionality, by hardware.
I will switch my route either to Oly M1 m2, or Pana G9.
Currently i have carried out some tests with my old M1, all m1-listed abilities were there. Wondering G9 now. No USB-tethering in it also?

bezineb5, 2022-10-30:
@ontheway16 It's probably too late now, but if I remember correctly, you can use the G9 with gphoto2, it's just recognized as another camera (Panasonic DC-GH5).
It also **seems** that the new OM System OM-5 (successor to the E-M5 Mark III) supports tethering. But then it will have to be added to gphoto2.

ontheway16, 2022-10-30:
> It also **seems** that the new OM System OM-5 (successor to the E-M5 Mark III) supports tethering. But then it will have to be added to libgphoto2.

I hope it supports, it is certainly a big miss on E-M5 III. And thanks for clearing the G9 question.