gphoto/libgphoto2 #824: Reduce capture-image-and-download command time to less than 1 second (SONY A6000)

Jing-de, 2022-08-31:
**Describe the bug**
In my project I want to caputre and download an image in 1 second. But when I issue the command _--caputure-image-and-download_, this process always takes around 2.5 second.
By reading the log file, I found the process is stuck in a loop for almost 2 seconds and always returns the following WARNING message in each loop: _ptp_usb_event [usb.c:589]   (0): Reading PTP event failed: Timeout reading from or writing to the port (-10)_

The same delay also exists with command _-I=1 --capture-image-and-download_. And always return 1 seconds behind schedule.

> $ gphoto2 --debug --debug-logfile=logfile-cap-1sec-2 -I=1 --capture-image-and-download
> Time-lapse mode enabled (interval: 1s).                                        
> Capturing frame #1...
> New file is in location /capt0000.jpg on the camera                            
> Saving file as capt0000.jpg
> Deleting file /capt0000.jpg on the camera
> not sleeping (1 seconds behind schedule)
> Capturing frame #2...
> New file is in location /capt0001.jpg on the camera                            
> Saving file as capt0001.jpg
> Deleting file /capt0001.jpg on the camera
> not sleeping (1 seconds behind schedule)
> Capturing frame #3...
> New file is in location /capt0002.jpg on the camera                            
> Saving file as capt0002.jpg
> Deleting file /capt0002.jpg on the camera
> not sleeping (2 seconds behind schedule)

**Name the camera**
output of: gphoto2 --auto-detect
_Sony Alpha-A6000 (Control)_

**libgphoto2 and gphoto2 version**

output of: gphoto2 --version
_gphoto2         2.5.28         gcc, popt(m), exif, cdk, aa, jpeg, no readline
libgphoto2      2.5.30         standard camlibs (SKIPPING docupen), gcc, no ltdl, EXIF
libgphoto2_port 0.12.1         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking_

**To Reproduce**
Steps to reproduce the behavior:

_$ gphoto2 --debug --debug-logfile=logfile-cap --capture-image-and-download_
and
_$ gphoto2 --debug --debug-logfile=logfile-cap-1sec -I=1 --capture-image-and-download_

Log file:
[logfile-cap.log](https://github.com/gphoto/libgphoto2/files/9461606/logfile-cap.log)

msmeissn, 2022-09-04:
Well, the image becomes available for download after 2.4 seconds in your trace, this is a camera side thing. Not sure if the camera could be made faster here.

You could queue triggering multiple images with multiple --trigger-capture after each other and then --wait-event-and-download=5s (or whatever time)

this would queue them in camera SDRAM, which is limited to a small amount of pictures though.

Jing-de, 2022-09-06:
Thank you very much for your reply, I can shoot faster with the method you provided :+1: 

I also have a problem, sometimes when I issue the command _--trigger-capture_, I can hear the beeping sound when focusing is done, but there is no shutter sound, which means the capture doesn't has not been carried out. By reading the debug log, I can see that the shutter button has been held down and released. The same problem also exists with command _--capture-image-and-download_
Do you know how I can solve this problem? Because I need to take a lot of photos in a short period of time, I want each shooting command to be executed normally

Steps to reproduce the behavior:
_$ gphoto2 --debug --debug-logfile=logfile-trigger-capture --trigger-capture_

Log file:
[logfile-cap.log](https://github.com/gphoto/libgphoto2/files/9497566/logfile-cap.log)