gphoto/libgphoto2 #828: gphoto2 not able to see vusb

boriscosic, 2022-09-09:
**Describe the bug**
Compiled libgphoto2-2.5.30 with --enable-vusb, removed usb1.* and added files and folders to /usr/local/share/libgphoto2_port/0.12.1 and /usr/share/libgphoto2_port/<version>/ as per README but I can't seem to get the virtual camera to show up.

gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------

gphoto2 --list-ports shows 
usb:001,001                      Universal Serial Bus            

**Name the camera**
vusb

**libgphoto2 and gphoto2 version**
[gphoto2         2.5.9          gcc, popt(m), exif, no cdk, aa, jpeg, readline
libgphoto2      2.5.30         NON-STANDARD CAMLIB SET (ptp2), gcc, no ltdl, EXIF
libgphoto2_port 0.12.1         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi vusb, gcc, no ltdl, EXIF, USB, serial without locking
](usb:001,001                      Universal Serial Bus)

**To Reproduce**
[gphoto2-debug.log](https://github.com/gphoto/libgphoto2/files/9533240/gphoto2-debug.log)

msmeissn, 2022-09-09:
ok, i broke non-fuzz mode at some point ... I now applied a fix to current libgphoto2 GIT. Now gphoto2 --auto-detect should work and a virtual Nikon D750 be found.

closed 2022-09-09 by msmeissn

boriscosic, 2022-09-09:
@msmeissn thanks so much for quick turn around. With new build the camera shows up but list and capture have issues: 

```
gphoto2 -l | more
001,001: No such file or directory
There is 1 folder in folder '/'.
 - store_00010001
There is 1 folder in folder '/store_00010001'.
 - 
There is 1 folder in folder '/store_00010001/'.
...
There is 1 folder in folder '/store_00010001/'.
 - 
Segmentation fault (core dumped)
```

```
gphoto2 --capture-image
001,001: No such file or directory                                             
Segmentation fault (core dumped)
```

Adding 001,001 to /usr/local/share/libgphoto2_port/0.12.1/vcamera,
```
gphoto2 --capture-image
                                                                               
*** Error ***              
PTP I/O Error

*** Error ***              
An error occurred in the io-library ('I/O problem'): No error description available
ERROR: Could not capture image.
ERROR: Could not capture.

*** Error ***              
PTP I/O Error

*** Error ***              
An error occurred in the io-library ('I/O problem'): No error description available
```

msmeissn, 2022-09-12:
I fixed the endless directory list loop in GIT and it should not crash currently.

For virtual capture to work you need to put 1 JPG file into /usr/share/libgphoto2_port/0.12.1/vcamera/   (beside the README.txt file), this will be the file that --capture-image returns.

boriscosic, 2022-09-12:
Thanks @msmeissn works great now!