gphoto/libgphoto2 #829: Sony NEX7 wont take photos but is detected

Airboomin, 2022-09-15:
Hello, I have a sony nex7 and i cant get it to take pictures. Please someone help. I have no idea how to fix this or troubleshoot this. 

pi@octopi2:~ $ gphoto2 --auto-detect
Model                          Port
----------------------------------------------------------
Sony NEX-7 (PTP mode)          usb:001,021
pi@octopi2:~ $ gphoto2 --capture-image

*** Error ***
An error occurred in the io-library ('I/O problem'): No error description available
ERROR: Could not capture image.
ERROR: Could not capture.

*** Error ***
PTP General Error

*** Error ***
An error occurred in the io-library ('Unspecified error'): No error description available
pi@octopi2:~ $ gphoto2 --trigger-capture

*** Error ***
An error occurred in the io-library ('I/O problem'): No error description available
ERROR: Could not trigger capture.
*** Error (-7: 'I/O problem') ***

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --trigger-capture

Please make sure there is sufficient quoting around the arguments.

msmeissn, 2022-09-15:
can you attach the logfile as the message below says how to create?

Is there a PC Cntrol setting in the cameras USB settings Menu?

We currently have no success reports of remote control of this camera FWIW.

msmeissn, 2024-10-22:
so it chokes on the 0x9280 command we send.

I have already disabled this in current libgphoto2 git for the next release.

if you can try that and see if it helps?

theamnesic, 2024-10-22:
Sure!

I'm using the last Ubuntu, with: 

```
gphoto2         2.5.28         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.31         standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
```
Can you tell me what is the best way to update with the current libgphoto2 git please?

theamnesic, 2024-10-23:
@msmeissn I'm using [gphoto2-updater](https://github.com/gonzalo/gphoto2-updater), and now: 

```
gphoto2         2.5.28.1       gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.31.1       standard camlibs (SKIPPING docupen lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
```

Have his error: 

```
*** Error ***              
Sorry, your camera does not support generic capture
ERROR: Could not trigger capture.
*** Error (-6: 'Unsupported operation') *** 
```

And this logfile:

```
0.000017 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000063 main                        (2): gphoto2 2.5.28.1
0.000074 main                        (2): gphoto2 has been compiled with the following options:
0.000083 main                        (2):  + gcc (C compiler used)
0.000092 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000100 main                        (2):  + exif (for displaying EXIF information)
0.000107 main                        (2):  + no cdk (for accessing configuration options)
0.000114 main                        (2):  + no aa (for displaying live previews)
0.000121 main                        (2):  + no jpeg (for displaying live previews in JPEG format)
0.000127 main                        (2):  + no readline (for easy navigation in the shell)
0.000139 main                        (2): libgphoto2 2.5.31.1
0.000147 main                        (2): libgphoto2 has been compiled with the following options:
0.000155 main                        (2):  + standard camlib set SKIPPING SOME (ax203 canon digigr8 dimagev directory jl2005a jl2005c kodak_dc240 mars pentax ptp2 quicktake1x0 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801 SKIPPING docupen lumix)
0.000170 main                        (2):  + gcc (C compiler used)
0.000178 main                        (2):  + no ltdl (for portable loading of camlibs)
0.000185 main                        (2):  + EXIF (for special handling of EXIF files)
0.000199 main                        (2): libgphoto2_port 0.12.2
0.000210 main                        (2): libgphoto2_port has been compiled with the following options:
0.000219 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000226 main                        (2):  + gcc (C compiler used)
0.000235 main                        (2):  + no ltdl (for portable loading of iolibs)
0.000243 main                        (2):  + EXIF (for vusb)
0.000252 main                        (2):  + USB (libusb1, for USB cameras)
0.000260 main                        (2):  + serial (for serial cameras)
0.000269 main                        (2):  + no resmgr (serial port access and locking)
0.000286 main                        (2):  + no ttylock (serial port locking)
0.000301 main                        (2):  + no lockdev (serial port locking)
0.000310 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000319 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000328 main                        (2): invoked with following arguments:
0.000336 main                        (2):   --debug
0.000344 main                        (2):   --debug-logfile=my-logfile.txt
0.000352 main                        (2):   --trigger-capture
0.000385 load_settings               (2): Creating gphoto config directory ('/home/guillaume/.gphoto')
0.000440 load_settings               (2): Loading settings from file '/home/guillaume/.gphoto/settings'.
0.000568 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.000580 gp_port_init_localedir      (2): localedir has been set to /usr/local/share/locale (compile-time default)
0.000589 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/local/lib/libgphoto2_port/0.12.2'...
0.000657 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.2/disk'.
0.000802 gp_port_library_list        (2): found fstab fstype ext4
0.000821 gp_port_library_list        (2): found fstab fstype vfat
0.000831 gp_port_library_list        (2): found fstab fstype swap
0.000913 gp_port_library_list        (2): found mtab fstype sysfs
0.000924 gp_port_library_list        (2): found mtab fstype proc
0.000933 gp_port_library_list        (2): found mtab fstype devtmpfs
0.000955 gp_port_library_list        (2): found mtab fstype devpts
0.000965 gp_port_library_list        (2): found mtab fstype tmpfs
0.000985 gp_port_library_list        (2): found mtab fstype ext4
0.001008 gp_port_library_list        (2): found mtab fstype securityfs
0.001021 gp_port_library_list        (2): found mtab fstype tmpfs
[... 1514 lines cut ...]
0.074999 _cb_irq                     (2): Transfer 0x5eb76f4014f0 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.075011 _cb_irq                     (2): 0x5eb76f3f8ff0 with status 3
0.075021 _cb_irq                     (2): Transfer 0x5eb76f3f8ff0 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.075033 _cb_irq                     (2): 0x5eb76f3fef00 with status 3
0.075042 _cb_irq                     (2): Transfer 0x5eb76f3fef00 should be in LIBUSB_TRANSFER_COMPLETED, but is 3!
0.075185 gp_filesystem_reset         (2): resetting filesystem
[... cut ...]
0.075467 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.075473 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.075479 delete_all_folders          (2): Internally deleting all folders from '/'...
0.075485 lookup_folder               (2): Lookup folder '/'...
0.075493 lookup_folder               (2): Found! / is 0x5eb76f3d16d0
0.075501 recurse_delete_folder       (2): Recurse delete folder 0x5eb76f3d16d0//
0.075509 delete_all_files            (2): Delete all files in folder 0x5eb76f3d16d0//

```

theamnesic, 2024-12-03:
I just read your comment here @msmeissn : https://github.com/gphoto/libgphoto2/issues/439#issuecomment-593076149

> We currently still assume Sony NEX as not capture capable.

Is it still relevant?

msmeissn, 2024-12-18:
still same status. current Sony PTP documentation also does not list the NEX series

closed 2024-12-18 by msmeissn