gphoto/libgphoto2 #831: Gphoto and USB-C/USB-3.0

nsmith1024, 2022-09-19:
Hello,

Im just trying to find out if GPHOTO  can work with a camera that uses USB-C or USB.3.0, 3.1.....

Is there any camera that GPHOTO wont work with USB-C or USB 3.1..?

Im asking because i tried it with a USB-C camera and it would'nt detect the camera at all, so i was wondering if GPHOTO's usb system was upgraded to use USB-C or USB 3.0... as yet.

Thanks

msmeissn, 2022-09-19:
Both my Nikon Z6 and Canon EOS M6 Mark 2 both have USB-C and work fine.

Usually the type of USB is not relevant.

ndim, 2022-10-16:
@nsmith1024 Do any other USB 3 devices (e.g. storage sticks) work with that system?

I remember I had a number of problems with USB 3 devices on my AMD Ryzen system until I turned on a few items in the firmware setup which appear to be unrelated to USB (iirc thunderbolt was amongst them).

hfiguiere, 2022-10-16:
Thunderbolt might actually be related as TB3 and 4 goes through USB-C connectors.

miccoli, 2022-12-18:
Just a quick comment: USB-C is a connector standard, while 3.x, 2.x are USB specs. Quoting from the wiki [USB-C article](https://en.wikipedia.org/wiki/USB-C):

> The designation C refers only to the connector's physical configuration or form factor and should not be confused with the connector's specific capabilities, which are designated by its transfer specifications (such as USB 3.2).

E.g. I was able to connect a Canon RP which has a USB-C connector but is a USB 2.0 device.

msmeissn, 2022-12-18:
original reporter had no further replies... 

Please reopen if issue is not solved yet.

closed 2022-12-18 by msmeissn