gphoto/libgphoto2 #832: Support Canon EOS R7

dvreed, 2022-09-20:
**Name of the camera**

Canon EOS R7

**USB IDs**
04a9:32f7

**camera summary output**
run

  gphoto2 --summary > summary.txt

[summary.txt](https://github.com/gphoto/libgphoto2/files/9602976/summary.txt)

**camera configuration output**

gphoto2 --list-all-config > list-all-config.txt

[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/9602981/list-all-config.txt)

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download

*** Error ***              
An error occurred in the io-library ('Unspecified error'): The supplied vendor or product id (0x0,0x0) is not valid.
ERROR: Could not capture image.
ERROR: Could not capture.

 
  gphoto2 --capture-preview
 ** works! **

msmeissn, 2022-09-21:
Thanks!

can you attach a debug trace of the --capture-image-and-download ?

dvreed, 2022-09-22:
Debug trace of --capture-image-and-download
[capture-image-and-download-debug.txt](https://github.com/gphoto/libgphoto2/files/9629691/capture-image-and-download-debug.txt)

givoly, 2023-06-12:
@msmeissn so you're saying R7 support already should exist? That would be outstanding (however, it isn't on any of the lists I've seen). So how can I verify this - I have an R7 and a Mac (or PC if required). What's the fastest/easiest way to confirm support works?

givoly, 2023-06-13:
Thanks. A naive attempt doesn't work. Here are 3 command lines:

(base) TG-2:2.5.30_1 admin2$ **gphoto2 -v**
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         clang, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.30         standard camlibs, clang, no ltdl, EXIF
libgphoto2_port 0.12.1         iolibs: disk ptpip serial usb1, clang, no ltdl, EXIF, USB, serial without locking

(base) TG-2:2.5.30_1 admin2$ **gphoto2 --capture-image-and-download**
                                                                              
*** Error ***              
Canon EOS Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image-and-download

Please make sure there is sufficient quoting around the arguments.

(base) TG-2:2.5.30_1 admin2$  env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image-and-download 
                                                                               
*** Error ***              
Canon EOS Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***       

Attached is the my-logfile.txt
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/11737390/my-logfile.txt)

I hope this helps figure out what's going on.

givoly, 2023-06-13:
@msmeissn thanks for this platform and the assistance!

I don't know if this helps, but more:

(base) TG-2:2.5.30_1 admin2$ gphoto2 --summary
Camera summary:                                                                
Manufacturer: Canon.Inc
Model: Canon EOS R7
  Version: 3-1.0.1
  Serial Number: **removed**
Vendor Extension ID: 0xb (1.0)
Vendor Extension Description: 

Capture Formats: JPEG Unknown(b108) Unknown(b10b) Unknown(b982)
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b105), Unknown(b982), Unknown(b10a), Unknown(b10b), Unknown(b109)

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Capture 2
	Canon Wifi support

Storage Devices Summary:
store_00010001:
	StorageDescription: SD1
	VolumeLabel: 
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read-Write
	Maximum Capability: 128001769472 (122072 MB)
	Free Space (Bytes): 71273283584 (67971 MB)
	Free Space (Images): -1
store_00020001:
	StorageDescription: SD2
	VolumeLabel: 
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read-Write
	Maximum Capability: 128001769472 (122072 MB)
	Free Space (Bytes): 128000851968 (122071 MB)
	Free Space (Images): -1

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon EOS R7'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 86% (86)

givoly, 2023-06-15:
@msmeissn Anything I can check out on my end to help move this forward? Is there any specific mode that the camera should be in? "B(ulb)" or "M(anual)" for instance? Would this mode possibly affect ability to connect/control the camera? I know that APT and ASCOM driver connecting to this camera ARE sensitive to that mode and work in one or the other only.

msmeissn, 2023-06-15:
there seems to be some waiting missing before startup.

can you try to insert a --wait-event=1s   before the capture-image

givoly, 2023-06-15:
Yup. That seems to work - both in Bulb and Manual mode!
I also worked with darktable - and it works in tethering mode and controls camera (including focus, etc.).
So does this mean everything is working with the camera?

Here's the command line and terminal output from it (photo was captured and transferred):

(base) TG-2:2.5.30_1 admin2$ gphoto2 --wait-event=1s --capture-image-and-download
Waiting for 1 seconds for events from camera. Press Ctrl-C to abort.           
UNKNOWN PTP Property d116 changed, "model" to "2147484772"
UNKNOWN PTP Property d105 changed, "autoexposuremode" to "Manual"
UNKNOWN PTP Property d138 changed, "autoexposuremodedial" to "Manual"
UNKNOWN PTP Property d102 changed, "shutterspeed" to "30"
UNKNOWN PTP Property d101 changed, "aperture" to "3.5"
UNKNOWN PTP Property d103 changed, "iso" to "Unknown value 0001"
UNKNOWN PTP Property d106 changed, "drivemode" to "Super high speed continuous shooting"
UNKNOWN PTP Property d108 changed, "focusmode" to "Manual"
UNKNOWN PTP Property d107 changed, "meteringmode" to "Evaluative"
UNKNOWN PTP Property d109 changed, "whitebalance" to "Shadow"
UNKNOWN PTP Property d11b changed, "availableshots" to "30019"
UNKNOWN PTP Property d104 changed, "exposurecompensation" to "0"
UNKNOWN PTP Property d10c changed, "whitebalanceadjustb" to "0"
UNKNOWN PTP Property d10b changed, "whitebalanceadjusta" to "0"
UNKNOWN PTP Property d10e changed, "whitebalancexb" to "0"
UNKNOWN PTP Property d10d changed, "whitebalancexa" to "0"
UNKNOWN PTP Property d1d9 changed, "aeb" to "off"
UNKNOWN PTP Property d1d5 changed
UNKNOWN PTP Property d1c1 changed, "alomode" to "x3"
UNKNOWN PTP Property d10f changed, "colorspace" to "sRGB"
UNKNOWN PTP Property d114 changed, "autopoweroff" to "30"
UNKNOWN PTP Property d1b7 changed
UNKNOWN PTP Property d110 changed, "picturestyle" to "Auto"
UNKNOWN PTP Property d157 changed
UNKNOWN PTP Property d158 changed
UNKNOWN PTP Property d159 changed
UNKNOWN PTP Property d15a changed
UNKNOWN PTP Property d15b changed
UNKNOWN PTP Property d15c changed
UNKNOWN PTP Property d15d changed
UNKNOWN PTP Property d15e changed
UNKNOWN PTP Property d163 changed
UNKNOWN PTP Property d164 changed
UNKNOWN PTP Property d165 changed
UNKNOWN PTP Property d1b1 changed, "evfmode" to "1"
UNKNOWN PTP Property d1bc changed
UNKNOWN PTP Property d1b2 changed, "depthoffield" to "0"
UNKNOWN PTP Property d1b3 changed
UNKNOWN PTP Property d1b4 changed
UNKNOWN PTP Property d1b6 changed
UNKNOWN PTP Property d1ab changed
UNKNOWN PTP Property d238 changed
UNKNOWN PTP Property d1b0 changed, "output" to "TFT"
UNKNOWN PTP Property d10a changed, "colortemperature" to "5700"
UNKNOWN PTP Property d1a8 changed
UNKNOWN PTP Property d198 changed
UNKNOWN PTP Property d1dd changed
UNKNOWN PTP Property d1a1 changed
UNKNOWN PTP Property d115 changed, "ownername" to ""
UNKNOWN PTP Property d1a4 changed
UNKNOWN PTP Property d1b8 changed, "movierecordtarget" to "SDRAM"
UNKNOWN PTP Property d1ba changed
UNKNOWN PTP Property d1c9 changed, "continuousaf" to "On"
UNKNOWN PTP Property d17a changed
UNKNOWN PTP Property d20d changed
UNKNOWN PTP Property d1c5 changed
UNKNOWN PTP Property d1d4 changed
UNKNOWN PTP Property d1c2 changed, "eosmovieswitch" to "0"
UNKNOWN PTP Property d1bd changed
UNKNOWN PTP Property d194 changed, "aspectratio" to "3:2"
UNKNOWN PTP Property d195 changed
UNKNOWN PTP Property d196 changed
[... 283 lines cut ...]
UNKNOWN OLCInfo exposure indicator 0,0,0.0 (01000000)
UNKNOWN OLCInfo event 0x0080 content 00000000
UNKNOWN Focus Info 000100000000
UNKNOWN Focus Mask 00000000000000
UNKNOWN OLCInfo event 0x0400 content 00000000000000
UNKNOWN OLCInfo event 0x0800 content 0000000000000000
UNKNOWN OLCInfo event 0x1000 content 00
UNKNOWN OLCInfo event mask=1fff
UNKNOWN PTP Property d115 changed, "ownername" to ""
UNKNOWN PTP Property d1d0 changed, "artist" to "Tal Givoly"
UNKNOWN PTP Property d1d1 changed, "copyright" to "Tal Givoly"
UNKNOWN PTP Property d1af changed, "eosserialnumber" to "043031000258"
UNKNOWN PTP Property d11b changed, "availableshots" to "30019"
UNKNOWN PTP Property d11c changed
UNKNOWN PTP Property d11b changed, "availableshots" to "30019"
UNKNOWN PTP Property d11c changed
UNKNOWN PTP Property d11b changed, "availableshots" to "30019"
UNKNOWN PTP Property d11c changed
UNKNOWN PTP Property d11b changed, "availableshots" to "30019"
UNKNOWN StorageInfoChanged 0x00010001
UNKNOWN StorageInfoChanged 0x00020001
UNKNOWN PTP Property d1b7 changed
New file is in location /capt0000.cr3 on the camera
Saving file as capt0000.cr3
Deleting file /capt0000.cr3 on the camera

msmeissn, 2023-06-15:
that the initial wait needs to be specified is something I also need to hardcode in.

Can you capture logfiles for the case without wait and with ? so the failing and non-failing case?

Perhaps I can spot a pattern that I can catch programmatically.

givoly, 2023-06-15:
I went to do just that, but now it's not reconstructable - it seems to work also without the wait, if I turn on debug, it simply works in Manual mode as well... and also Fv mode... so maybe the issue is gone... 
Anyway, this seems to simply work now.  Thanks!
What's the best GUI on top of this for tethered shooting? darktable is "okay", but anything better?

axxel, 2024-09-04:
Looks like the R7 support is completed as well?

msmeissn, 2024-09-04:
should be. 
Likely the OLC version is already known too. If not, I need to look at:

gphoto2 --set-config testolc=1 --debug --debug-logfile=olc.log

closed 2024-09-04 by msmeissn