gphoto/libgphoto2 #834: Turn off/on camera?

gabrielesartor, 2022-09-21:
**Is it possible to turn off/on the camera using this library?**

Given that there are some electronic components which can do it (like https://www.seagulluav.com/product/seagull-rec/) using USB, maybe there is a way to do it also with a libgphoto command via USB? I didn't find this feature in this library, right?

(I'm using a Canon Powershot 740 HS)
Thank you in advance!

msmeissn, 2022-09-22:
I currently have no experience with that, sorry. 
As for PTP level, while there is a PTP PowerDown opcode, it is only supported by some Kodaks.

For Canon I have not seen such a command.

So my guess is "no" currently.

jabdoa2, 2022-11-28:
You can enable/disable the usb device using libusb1. It will only turn on/off cameras which are powered via usb though.