gphoto/libgphoto2 #835: vusb camera capture hangs forever

RReverser, 2022-09-21:
**Describe the bug**

When using `vusb` virtual camera, capturing hangs forever, although earlier docs (e.g. https://hackweek.opensuse.org/projects/simulate-ptp-cameras-in-libgphoto2) suggest that capturing used to work.

**Name the camera**

(virtual camera)

**libgphoto2 and gphoto2 version**

```
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.30.1       standard camlibs (SKIPPING docupen lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.1         iolibs: disk ptpip serial vusb, gcc, no ltdl, EXIF, no USB, serial without locking
```

**To Reproduce**

Build with `--enable-vusb` and run e.g.

```
gphoto2 --capture-image-and-download
```

Here's the debug log, looks like an internal error.

[out.log](https://github.com/gphoto/libgphoto2/files/9620731/out.log)

msmeissn, 2022-09-22:
did you put 1 .jpg file into the vusb cameras virtual directory?
i need to check if "no image" is handled well enough.

RReverser, 2022-09-22:
> did you put 1 .jpg file into the vusb cameras virtual directory?

Hmm, no, I didn't know it's required... Is that documented somewhere? I thought vusb just produces some hardcoded static image in that case. 

> i need to check if "no image" is handled well enough.

Yeah it would be useful to print a helpful error in that case.

RReverser, 2022-09-23:
Interesting, I see that the code does try to report a meaningful error message, I guess it's just not printed by libgphoto2 CLI?

msmeissn, 2022-09-24:
The problem is more that we returned GeneralError and the ptp2 driver ignored it. I am returning a different error now.

RReverser, 2022-09-25:
Linking my comments from the commit here so that they don't get lost: https://github.com/gphoto/libgphoto2/commit/cf154a3d81fcd4b6ce12c57f428c2e712607e109#r84900329

RReverser, 2022-09-26:
Actually, re:

> There is also one more similar code at line 1107 above that still reports PTP_RC_GeneralError.

Aren't those two copies of code doing exactly the same thing? Looks like a mistake:

https://github.com/gphoto/libgphoto2/blob/a3bd587689cf13a89c99f1fe7d0c1014a046835a/libgphoto2_port/vusb/vcamera.c#L1136-L1170

RReverser, 2022-09-26:
FWIW it doesn't look like changing PTP error type alone fixed the hang-up when a file is missing.

closed 2022-09-27 by msmeissn

RReverser, 2022-09-27:
I don't think https://github.com/gphoto/libgphoto2/commit/8b462f8c835b9d3ab0cee4117e0b0458dac00445 fixed it :/

Now seeing "Setting port timeout to 20000 milliseconds." in logs, but after that it still hangs trying "gp_port_vusb_check_int", resulting "Reading PTP event failed: Timeout reading from or writing to the port (-10)" and so on and on forever.

msmeissn, 2022-09-28:
i tried with gphoto2 --capture-image and it terminates for me.

can you get a debugtrace of gphoto2 --capture-image --debug --debug-logfile=xx.log