gphoto/libgphoto2 #842: vcamera prints error messages about opening invalid filename upon startup

RReverser, 2022-09-27:
When using virtual camera, I've noticed that I'm always getting error message like

```
001,001: No such file or directory
```

upon startup, even though everything works fine afterwards.

I've tracked it down to this piece of code: https://github.com/gphoto/libgphoto2/blob/8b462f8c835b9d3ab0cee4117e0b0458dac00445/libgphoto2_port/vusb/vcamera.c#L1802-L1816

Basically, `vcam_open` receives port like `usb:001,001` from `camera_init`, truncates it to second half and then for some reason tries to open what's left (`001,001`), fails and prints an error via `perror`.

I could either remove `perror` code or that code altogether, but I suppose it serves some purpose which is not documented in the file, so I'm not confident with making any fixes there. @msmeissn could you help out?

closed 2022-09-28 by msmeissn

RReverser, 2022-09-28:
Thanks for removing the warning. I wonder though, should it even try to open filenames like `001,001` in the first place? Maybe it could skip that branch altogether for `usb:` ports?

msmeissn, 2022-09-28:
Reason is that the vusb driver doubles as both a virtual ptp camera and a fuzzing driver. over the path statement it is fed the fuzzing data.

RReverser, 2022-09-28:
Could fuzzing data be fed under custom prefix only? E.g. when someone passes `vusb:some_path`, then `some_path` should be used for fuzzing, but when a regular `usb:...` prefix is used (which is the default when using `gp_camera_autodetect`), then assume it's just virtual ptp camera?

msmeissn, 2022-09-29:
i just put all the fuzzing code behind an ifdef FUZZING so it does not confuse the virtual ptp usage. ;)

jim-easterbrook, 2023-08-06:
I'm glad I found this bug report. I was having difficulty using a virtual camera with python-gphoto2 until I built it with libgphoto2 compiled from the current GitHub repos. Keenly awaiting the next release of libgphoto2 now.  (-: