gphoto/libgphoto2 #845: Nikon 1 S1 capture hangs

aluckycode, 2022-10-02:
I could not get --capture-image-and-download work on the Nikon 1 S1, but by looking at the J1 and V1 issues I was able to make it work by modifying some code in library,c

`@@ -514,6 +514,7 @@ fixup_cached_deviceinfo (Camera *camera, PTPDeviceInfo *di) {

                        /* V1 and J1 are a less reliable then the newer 1 versions, no changecamera mode, no getevent, no initiatecapturerecinsdram */
                        if (    !strcmp(params->deviceinfo.Model,"V1") ||
+                               !strcmp(params->deviceinfo.Model,"S1") ||
                                !strcmp(params->deviceinfo.Model,"J1")
                        ) {
                                /* on V1 and J1 even the 90c7 getevents does not work */
@@ -528,6 +529,13 @@ fixup_cached_deviceinfo (Camera *camera, PTPDeviceInfo *di) {
                                                GP_LOG_D("On Nikon V1: disable NIKON_InitiateCaptureRecInSdram as its unreliable");
                                                di->OperationsSupported[i] = PTP_OC_InitiateCapture; /* overwrite */
                                        }
+                                       if (!strcmp(params->deviceinfo.Model,"S1") &&
+                                            (di->OperationsSupported[i] == PTP_OC_NIKON_InitiateCaptureRecInMedia))
+                                       {
+
+                                               GP_LOG_D("On Nikon S1: disable NIKON_InitiateCaptureRecInMedia as its unreliable");
+                                               di->OperationsSupported[i] = PTP_OC_InitiateCapture; /* overwrite */
+                                       }
                                }`

I do not know if this is the best way to fix the problem but it seems to work for me for basic capture. I did see errors in the log for InitiatecaptureRecInMedia.  

Does Bulb mode work in any on of the Nikon 1 series through gphoto2/libgphoto2 ? I do see an action  for bulb
`/main/actions/bulb
Label: Bulb Mode
Readonly: 0
Type: TOGGLE
Current: 2
END`

msmeissn, 2022-10-03:
can you also attach the s1.log from

gphoto2 --summary --debug --debug-logfile=s1.log

msmeissn, 2022-10-03:
i put in your fixed code, also try to get my hands on a S1.
(The whiole area of code is really hacky there.)

aluckycode, 2022-10-03:
> can you also attach the s1.log from
> 
> gphoto2 --summary --debug --debug-logfile=s1.log

[s1.log](https://github.com/gphoto/libgphoto2/files/9699125/s1.log)

This is for firmware version 1.2

aluckycode, 2022-10-03:
Looking at the open issues it is possible that  J5 (#816) and J3 (#803) have the same problem. My guess based on what I saw on the S1 is that the J3 has older firmware and once updated will show the same problems as the S1 and J5.

msmeissn, 2022-10-08:
did you have an sd card inserted
if I insert an SD card the before code works as-is.

msmeissn, 2022-10-08:
I now own a:
Model: S1
  Version: Ver.1.1003

I see you have
Model: S1
device version: Ver.1.2004
e.g. newer firmware.

aluckycode, 2022-10-08:
> did you have an sd card inserted if I insert an SD card the before code works as-is.

Yes I have an SD card inserted.

aluckycode, 2022-10-08:
> I now own a: Model: S1 Version: Ver.1.1003
> 
> I see you have Model: S1 device version: Ver.1.2004 e.g. newer firmware.

The camera came with 1.0 firmware and it worked with the old gphoto2. The 
shutterspeed2 settings did not make it to the picture though. I could read and write them
with correct values, just that the camera did not use the value.

So I upgraded the firmware from the Nikon site which is 1.2. It was then the capture stopped working.
Not sure which firmware would have been easier to fix, but I wish I had some way to try out the older firmware.
So if you can extract the firmware from the camera before you upgrade and hold on to it, it would be good.

Can 1.1003 do bulb captures?