gphoto/libgphoto2 #847: Infinite loop while trying to find attached cameras

phako, 2022-10-08:
**Describe the bug**

Trying to list the available cameras results in an inifite loop. Note: I am not entirely sure this is a gphoto2 issue, the flatpak issue listed below suggests that it might be a combination of software that causes this.

I have two independent reports on Shotwell's initial start-up being blocked which both were pointing down to libgphoto2's gp_port_info_list_load

https://gitlab.gnome.org/GNOME/shotwell/-/issues/5002
https://github.com/flathub/org.gnome.Shotwell/issues/39

Both issues go away if either the usb1.so port is removed or the flatpak has its access to devices revoked.

**Name the camera**
No cameras are involved

**libgphoto2 and gphoto2 version**
libgphoto2 is 2.5.30

**To Reproduce**
Steps to reproduce the behavior:

if system is affected, can be reproduced by running gphoto2 -l (https://gitlab.gnome.org/GNOME/shotwell/-/issues/5002#note_1562811). I will add debug.log as soon as I get it from the user(s).

msmeissn, 2022-10-20:
(as long as its not reproducable i cannot do much).

What occasionaly could happen is that also the "disk" driver tries to look for mountpoints with /DCIM/ directory and that might hang on network drives, or buggy local storage.

msmeissn, 2022-10-20:
https://gist.githubusercontent.com/queeup/e79e17c516038a8e75dbe84cd4c3dfb2/raw/075e53fd138658bb2246023ded41c49ddbf615f6/shotwell_gdb3.html

Here we hang on opening a /dev/bus/usb/xxx/yyy device entry.

queeup, 2022-12-18:
Any progress about this? What do you need to investigate further?

msmeissn, 2022-12-18:
Still no idea what happens ... The open of the usb device node hangs (not loops).

Does this always happen?

Also can you try a regular gphoto2 app (not flatpak based) on the same system and does it happen there?

queeup, 2022-12-18:
Yes, It does always happens. Shotwell is getting unresponsive at the beginning. I can't use that app anymore.

This is regular gphoto2 cli printout on Silverblue 37

```bash
❯ rpm -aq gphoto2
gphoto2-2.5.27-4.fc37.x86_64

❯ gphoto2 --version
gphoto2 2.5.27

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27         gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.30         standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.1         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial lockdev locking

❯ gphoto2 -l  --debug --debug-logfile=debug.log
                                                                               
*** Error ***              
Could not detect any camera
*** Error (-105: 'Unknown model') ***
```

<details>
<summary>debug.log:</summary>
<pre><span style="color:#BE84FF">0.000297</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:</span>
<span style="color:#BE84FF">0.000399</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">): gphoto2 </span><span style="color:#BE84FF">2.5</span><span style="color:#F8F8F2">.</span><span style="color:#BE84FF">27</span>
<span style="color:#BE84FF">0.000438</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">): gphoto2 has been compiled with the following options:</span>
<span style="color:#BE84FF">0.000454</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">):  + gcc (C compiler used)</span>
<span style="color:#BE84FF">0.000468</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">):  + popt (mandatory, for handling command-line parameters)</span>
<span style="color:#BE84FF">0.000481</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">):  + exif (for displaying EXIF information)</span>
<span style="color:#BE84FF">0.000493</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">):  + no cdk (for accessing configuration options)</span>
<span style="color:#BE84FF">0.000505</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">):  + no aa (for displaying live previews)</span>
<span style="color:#BE84FF">0.000518</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">):  + jpeg (for displaying live previews in JPEG format)</span>
<span style="color:#BE84FF">0.000530</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">):  + readline (for easy navigation in the shell)</span>
<span style="color:#BE84FF">0.000544</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">): libgphoto2 </span><span style="color:#BE84FF">2.5</span><span style="color:#F8F8F2">.</span><span style="color:#BE84FF">30</span>
<span style="color:#BE84FF">0.000558</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">): libgphoto2 has been compiled with the following options:</span>
<span style="color:#BE84FF">0.000573</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">):  + standard camlib set (ax203 canon digigr8 dimagev directory docupen jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801)</span>
<span style="color:#BE84FF">0.000586</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">):  + gcc (C compiler used)</span>
<span style="color:#BE84FF">0.000598</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">):  + no ltdl (for portable loading of camlibs)</span>
<span style="color:#BE84FF">0.000610</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">):  + EXIF (for special handling of EXIF files)</span>
<span style="color:#BE84FF">0.000629</span><span style="color:#F8F8F2"> main                        (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">): libgphoto2_port </span><span style="color:#BE84FF">0.12</span><span style="color:#F8F8F2">.</span><span style="color:#BE84FF">1</span>
[... cut ...]
<span style="color:#BE84FF">0.191269</span><span style="color:#F8F8F2"> gp_port_info_list_get_info  (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">): Getting info of entry </span><span style="color:#BE84FF">39</span><span style="color:#F8F8F2"> (</span><span style="color:#BE84FF">47</span><span style="color:#F8F8F2"> available)...</span>
<span style="color:#BE84FF">0.195548</span><span style="color:#F8F8F2"> gp_port_set_timeout         (</span><span style="color:#BE84FF">2</span><span style="color:#F8F8F2">): Setting port timeout to </span><span style="color:#BE84FF">5000</span><span style="color:#F8F8F2"> milliseconds.</span>
<span style="color:#BE84