gphoto/libgphoto2 #852: OM-1, from the company formerly known as Olympus

400trix, 2022-10-21:
Fill in fields below. If you don't know how, leave it free.

OM Digital Solutions (OMDS) "Olympus" OM-1

lsusb output:

Bus 001 Device 001: ID 33a2:0136 33a2 OM-1  Serial: XXXXXXXXX

run

  gphoto2 --summary > summary.txt

Bus error on my 14" MBP running macOS 12.6

**camera configuration output**

gphoto2 --list-all-config > list-all-config.txt

and attach list-all-config.txt here

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
 
Error: 
*** Error ***
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***

  gphoto2 --capture-preview
*** Error (-6: 'Unsupported operation') ***

[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/9842321/list-all-config.txt)

msmeissn, 2022-10-22:
which libgphoto2 and gphoto2 version are in use?

can you also get a debug logfile?

gphoto2 --debug --debug-logfile=summary.log --summary

goldff, 2023-07-24:
Added summary and log for the two modes.  

[summary_MTP_OM1.log](https://github.com/gphoto/libgphoto2/files/12151934/summary_MTP_OM1.log)
[summary_MTP_OM1.txt](https://github.com/gphoto/libgphoto2/files/12151954/summary_MTP_OM1.txt)

[summary_RAW_OM1.log](https://github.com/gphoto/libgphoto2/files/12151958/summary_RAW_OM1.log)
[summary_RAW_OM1.txt](https://github.com/gphoto/libgphoto2/files/12151978/summary_RAW_OM1.txt)

msmeissn, 2023-08-01:
support now in git, commit b431a03440fda35c33ff92da869f25a8bc2e3170 and commit 1393c27683711d0b401d8eadc9a263208a9dbbed

closed 2023-08-01 by msmeissn

goldff, 2023-08-01:
Thx!  Confirm, working perfectly!