gphoto/libgphoto2 #854: ZV-E10 Setting shutter speed causes erratic behaviour

TheStaticTurtle, 2022-10-23:
**Camera:** ZV-Z10 (USB PTP Class Camera / usb:001,016)
**libgphoto2 version:** 2.5.27 (standard camlibs, gcc, ltdl, EXIF)
**libgphoto2_port version:**  0.12.0 (iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking)

I'm trying to set the shutter speed of my ZV-E10 with libgphoto2, but I'm facing a very weired issue. 

Regardless of what I set it to with `gp_widget_set_value`, once I apply it with `gp_camera_set_config` the shutter speed is almost completely random. 
For example, I may set it to 1/1000 and the camera will report 1/640 or 1/4000 or anything really. After querying the config & widget again, libghoto2 also reports the "wrong" one. 
The only time it seems not to crap the bed is when I set it to a value that it's already at (for example, setting it to 1/500 will not set it to something else if the camera is already at 1/500)

I also need to add that this does not affect the `f-number` widget (didn't try any others yet). And that setting the shutter speed seems to work correctly in the command line (However, if I spam it fast enough I can see some wired behaviour).
 
I've noticed with `sudo gphoto2 --list-config` and reading some code that there are two `/main/capturesettings/shutterspeed`.
Here is the output of `gphoto2 --list-all-config --debug`: https://termbin.com/90dep where there is two shutter speed (0xd229 and 0xd20d)

My program is much more complicated, but the issue boils down to the following code.
```c++
void set_shutter_speed (Camera *camera, GPContext *context, std::string val) {
    int retval;
    CameraWidget *root;
    CameraWidget *widget;
    
    retval = gp_camera_get_config(camera, &root, context);
    printf("gp_camera_get_config: %d\n", retval);
    
    retval = gp_widget_get_child_by_name (root, "shutterspeed", &widget);
    printf("gp_widget_get_child_by_id: %d\n", retval);
    
    const char* test = "1/1000"; //val.c_str();
    printf("%s\n",test);
    retval = gp_widget_set_value(widget, test);
    printf("gp_widget_set_value: %d\n", retval);
    
    retval = gp_camera_set_config(camera, root, context);
    printf("gp_camera_set_config: %d\n", retval);
    
    gp_widget_free (root);
}
```

Camera & context are initialized somewhere else like this:
```c++
Camera	*usb_camera;
GPContext *usb_camera_context;

usb_camera_context = gp_context_new();
gp_context_set_error_func(usb_camera_context, gphoto_ctx_error_func, NULL);
gp_context_set_status_func(usb_camera_context, gphoto_ctx_status_func, NULL);
gp_camera_new(&usb_camera);

int gpret = gp_camera_init(usb_camera, usb_camera_context);
if (gpret != GP_OK) {
    spdlog::get("gphoto2")->error("Failed to init: {}", gpret);
    return -4;
}
```

Anyone knows what the hell might be going on?

ioannismihailidis, 2022-12-26:
Have you found a solution?

Maybe this is related to #775 . There are also two /main/capturesettings/shutterspeed and a strange behavior when changing it.

msmeissn, 2022-12-27:
can you first update to 2.5.30, we did various fixes to Sony config handling in between.

do you have a debugtrace with the commandline tool?

jufa, 2025-01-05:
Work around as of 2.5.30: https://github.com/gphoto/libgphoto2/issues/1068#issuecomment-2571679122

watsonmw, 2025-01-06:
Have a fix for this in this pull request: https://github.com/gphoto/libgphoto2/pull/1067

To test this, you may need to add the camera model to list of sony_mode_300 cameras e.g.:

     if (!strcmp(params->deviceinfo.Model, "ZV-E10")) return 1;

https://github.com/gphoto/libgphoto2/blob/e6fa2e5a355f8589c042e595febb6a8a2d66f287/camlibs/ptp2/ptp.h#L4825