gphoto/libgphoto2 #867: Dakota Ritz camera gives error "*** Error (-7: 'I/O problem') ***"

puterboy, 2022-11-10:
I have been trying to download photos from an old Ritz Dakota camera which is supported (as TESTING) by libgphoto2.
I last tried to do this more than 15 years ago in Fedora FC12 when it worked.

Now I am trying to run it under Ubuntu 18.04 using libgphoto2 2.5.16-2 and gphoto2 2.5.15-2

gphoto2 identifies the camera correctly:
```
#gphoto2 --auto-detect
Model                          Port
----------------------------------------------------------
PureDigital Ritz Disposable    usb:001,026
```
But I am get an error when I try to list or retrieve files.
For example, when I run: `]gphoto2 --debug --get-all-files`

 I get the following:
```
0.035495 gp_abilities_list_detect_usb(2): Auto-detecting USB cameras...
0.035726 gp_libusb1_find_device_by_class_lib(2): Looking for USB device (class 0x29a, subclass, 0xffffffff, protocol 0xffffffff)...
0.035793 gp_libusb1_find_device_lib  (2): Looking for USB device (vendor 0x4fc, product 0xffff)... found.
0.035829 gp_libusb1_find_device_lib  (2): Detected defaults: config 1, interface 0, altsetting 0, inep ffffffff, outep ffffffff, intep ffffffff, class ff, subclass 00
0.035831 gp_abilities_list_detect_usb(2): Found 'PureDigital Ritz Disposable' (0x4fc,0xffff)
0.035834 gp_port_free                (2): Freeing port...
0.035835 gp_port_close               (2): Closing port...
0.036047 gp_camera_set_abilities     (2): Setting abilities ('PureDigital Ritz Disposable')...
0.036052 gp_setting_set              (2): Setting key 'model' to value 'PureDigital Ritz Disposable' (gphoto2)
0.036055 save_settings               (2): Saving 2 setting(s) to file "/home/myuser/.gphoto/settings"
0.036115 gp_port_info_list_lookup_path(2): Looking for path 'usb:001,026' (41 entries available)...
0.036117 gp_port_info_list_get_info  (2): Getting info of entry 36 (41 available)...
0.036119 gp_camera_set_port_info     (2): Setting port info for port 'Universal Serial Bus' at 'usb:001,026'...
0.039363 gp_port_set_timeout         (2): Setting port timeout to 5000 milliseconds.
0.039365 gp_port_set_settings        (2): Setting settings...
0.039368 gp_libusb1_update           (2): (old int=0, conf=-1, alt=-1) port usb:001,026, (new int=0, conf=-1, alt=-1) port usb:001,026
0.039369 gp_libusb1_update           (2): lowlevel libusb1 port not yet opened, no need for libusb changes
0.039371 gp_setting_set              (2): Setting key 'port' to value 'usb:001,026' (gphoto2)
0.039373 save_settings               (2): Saving 2 setting(s) to file "/home/myuser/.gphoto/settings"
0.039433 gp_camera_folder_list_files (2): Listing files in '/'...
0.039435 gp_camera_init              (2): Initializing camera...
0.039443 gp_libusb1_find_device_lib  (2): Looking for USB device (vendor 0x4fc, product 0xffff)... found.
0.039488 gp_libusb1_find_device_lib  (2): Detected defaults: config 1, interface 0, altsetting 0, inep ffffffff, outep ffffffff, intep ffffffff, class ff, subclass 00
0.039490 gp_camera_init              (2): Loading '/usr/lib/x86_64-linux-gnu/libgphoto2/2.5.16/spca50x'...
0.039550 gp_port_open                (2): Opening USB port...
0.039552 gp_libusb1_open             (2): ()
0.039566 gp_libusb1_open             (2): Device has a kernel driver attached (1), detaching it now.
0.039791 gp_libusb1_open             (2): claiming interface 0
0.039839 gp_libusb1_queue_interrupt_urbs [libusb1.c:713](0): 'libusb_submit_transfer (port->pl->transfers[i])' failed: Input/Output error (-1)
0.039870 gp_context_error            (0): An error occurred in the io-library ('I/O problem'): No error description available

*** Error ***
An error occurred in the io-library ('I/O problem'): No error description available
*** Error (-7: 'I/O problem') ***

0.040239 gp_camera_free              (2): Freeing camera...
0.040241 gp_port_free                (2): Freeing port...
0.040242 gp_port_close               (2): Closing port...
```

Any idea what might be causing this?

Note that I had the same problem on 2 different Ritz Dakota cameras so I am assuming this is not a problem with the camera itself.

Note that I am able to connect to and list files from an old Canon Digital camera attached to the same USB port so I assume this is not a computer hardware or more general libgphoto software problem.

msmeissn, 2022-11-15:
First communication attempt failed.

It likely is blocked by the kernel spca driver already, the kernel loads this automatically as a webcam perhaps.

puterboy, 2023-01-19:
When I plug in the Dakota, it then automatically loads:
`gspca_sunplus
gspca_main`

There is no other spca module listed under 'lsmod'
(I am running Ubuntu 18.04)

Manually removing the modules doesn't seem to change anything.
It still gets identified under auto-detect but fails with above error on other queries as above

What can I do to fix/troubleshoot further?

msmeissn, 2023-01-22:
the trace reports the kernel driver and that it detaches it already.
hmm.
not sure if you could move the gspca_main.ko to the side somewhere to avoid it interfering.

also gvfs will likely try to get the camera and use it now