gphoto/libgphoto2 #875: New Camera: GoPro HERO11 Black

maxkapalla, 2022-11-28:
Fill in fields below. If you don't know how, leave it free.

GoPro HERO11 Black

Use the Marketing name.

**USB IDs**
ID 2672:0059 GoPro HERO11 Black

**camera summary output**
run

  gphoto2 --summary > summary.txt

[summary.txt](https://github.com/gphoto/libgphoto2/files/10106295/summary.txt)

**camera configuration output**

gphoto2 --list-all-config > list-all-config.txt

[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/10106299/list-all-config.txt)

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
 
  gphoto2 --capture-preview

[logfile.txt](https://github.com/gphoto/libgphoto2/files/10106303/logfile.txt)

msmeissn, 2022-11-30:
added via commit 9fd14ca6f4bd8c440d6eb3918d94939111b7bae3

maxkapalla, 2022-11-30:
Thanks for the addition
I compiled and installed the new version and still get a PTP General Error like before.

I assume the PTP Capture mechanism doesnt work for the GoPro 11?

Attached you find the logfile

[logfile.log](https://github.com/gphoto/libgphoto2/files/10124858/logfile.log)

msmeissn, 2022-12-01:
the camera does not report any capture ability over PTP.

So far the GoPro have not supported this via PTP.

Capture usually works over WLAN only. 
Recent GoPro have additional software that makes a network connection appear over USB Ethernet emulation and can access the WLAN capture over that.

Neither is supported at this time by gphoto, sorry.

closed 2022-12-02 by maxkapalla