gphoto/libgphoto2 #876: Digikam crashes on import

pehy59, 2022-12-02:
**Describe the bug**
Digikam crashes on import from galaxy note 9 while connecting via network. Connection via USB in Digicam fails also, but without crash.

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs
Samsung Galaxy models (MTP)    usb:004,005

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version
gphoto2         2.5.28         gcc, popt(m), exif, cdk, no aa, jpeg, readline
libgphoto2      2.5.30         standard camlibs (SKIPPING lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.1         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial lockdev locking

**To Reproduce**
Steps to reproduce the behavior:

I'm trying to import pictures from my Galaxy Note 9 with digikam. Via USB cable there is an error message "Connection failed, try again" but without crash. The phone is visible as an MTP device, downloading photos manually via dolphin works most time. 
When connected via kde-connect, I can see the file systems on the phone and manually downloading images from /run/user/1000/82178b4676aa987a/primary/DCIM/ works outside Digicam!
In Digicam I manually configured the import from the same path as above. Digicam starts the import and crashes immediately.
Also the import via the IP address of the camera causes digicam to crash immediately.

The crash can be reproduced every time.

Logfile can be provided, it's to long for pasting here...

msmeissn, 2022-12-07:
can you get a backtrace using gdb / drkonqi of the crash?

what does over the network mean? what kind of protocol is used when importing?

pehy59, 2022-12-07:
Am Mittwoch, 7. Dezember 2022, 10:53:33 CET schrieb Marcus Meissner:

> can you get a backtrace using gdb / drkonqi of the crash?

I`ve attached two protocols here. 

> what does over the network mean? what kind of protocol is used when
> importing?

The camera was connected via wlan to the router. On my local Linux-PC the 
camera was visible and accessible via MTP: in Dolphin, using Kconnect. 
Kconnect is installed on both, the Linux-PC and the Camera.

When i copy the actual Path to the camera from Dolphin and use this path in 
Digicam for import, Digicam crashes immediately. 

I've tested the same procedure on my Laptop without any problems, so the cause 
for the reported bug is somewhere on the Linux-PC. 

Operating System: openSUSE Tumbleweed 20221206
KDE Plasma Version: 5.26.4
KDE Frameworks Version: 5.100.0
Qt Version: 5.15.7
Kernel Version: 6.0.10-1-default (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-4800MQ CPU @ 2.70GHz
Memory: 31.3 GiB of RAM
Graphics Processor: Mesa Intel® HD Graphics 4600
Manufacturer: Dell Inc.
Product Name: Precision M6800
System Version: 00

pehy59, 2022-12-07:
I've attached two logfiles to my reply-mail, but can't see them here?