gphoto/libgphoto2 #881: Canon EOS R6 Mark II support

ecidiego, 2022-12-19:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*

Canon EOS R6 Mark II

**USB IDs**
e.g. by running:   lsusb 

**camera summary output**
run

  gphoto2 --summary > summary.txt

attach summary.txt here

**camera configuration output**

gphoto2 --list-all-config > list-all-config.txt

and attach list-all-config.txt here

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
 
  gphoto2 --capture-preview

Will work getting the other information

msmeissn, 2022-12-20:
we need the above information to add it ... at least the USB ids. :)

ecidiego, 2022-12-20:
> we need the above information to add it ... at least the USB ids. :)

I'll have this today. Really hoping you guys can squeeze it in. Where can I donate? I see you will be adding EOS R7 and R10 - gotta get R6 Mark II in the next release! Thanks so much for this project.

ecidiego, 2022-12-20:
> we need the above information to add it ... at least the USB ids. :)

**USB ID:**

04a9:330B

**Marketing name:**

Canon EOS R6 Mark II

**summary.txt:**

[summary.txt](https://github.com/gphoto/libgphoto2/files/10271151/summary.txt)

**list-all-config.txt:**

[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/10271156/list-all-config.txt)

**capture DOES work, attaching files in case you need them:**

[image]

Thank you so much, again, for this project!

msmeissn, 2022-12-23:
if you want to donate money: http://gphoto.org/doc/donate.php

But already that you help testing cameras is a great contribution!

If you are able to, can you build libgphoto2 master, and try 

gphoto2 --set-config testolc=1 --debug --debug-logfile=olc-debug.log 
and attach olc-debug.log ?

ecidiego, 2022-12-24:
I got:

*** Error ***
testolc not found in configuration tree.
*** Error (-1: 'Unspecified error') ***

I did download and build off the Master

msmeissn, 2023-01-05:
(sorry for late reply)
what does gphoto2 --version   report? if it does not say 2.5.30.1 libgphoto2 it probably still using the system libs

ecidiego, 2023-05-04:
>
>
> Any guestimates if an R6 Mark II supporting build is on the way?

From the App dev I am trying to use it with...

[image: image.png]

givoly, 2023-06-12:
Will you be adding support for the R6M2 as well as R7/R10 soon?

msmeissn, 2023-06-18:
The image did not go through, sorry.

gphoto2 --version should show 
libgphoto2 2.5.30.1 

for other EOS cameras, please report them via separate tickets. As written here, they likely will work already, we just need some small adjustments, and the USB ids for correct name reporting.

ecidiego, 2023-07-20:
Any idea when gphoto2 will be updated? The vendor I want to use my R6 Mkii with won't budge... they say "when gphoto2 updates we will update. "

msmeissn, 2023-07-21:
i am planning to cut a new release in the next weeks, life is just super busy the last year for me.

axxel, 2024-09-04:
libgphoto2-2.5.31 is released and includes the R6 Mark II support, right?

msmeissn, 2024-09-06:
yes

closed 2024-09-06 by msmeissn