gphoto/libgphoto2 #882: Sigma FP L support

G-DP, 2022-12-20:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*

Sigma FP L

**USB IDs**
e.g. by running:   lsusb 

[USB-IDs.txt](https://github.com/gphoto/libgphoto2/files/10267913/USB-IDs.txt)

**camera summary output**
run

  gphoto2 --summary > summary.txt

attach summary.txt here

[summary.txt](https://github.com/gphoto/libgphoto2/files/10267915/summary.txt)

**camera configuration output**

gphoto2 --list-all-config > list-all-config.txt

and attach list-all-config.txt here

[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/10267917/list-all-config.txt)

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
 
  gphoto2 --capture-preview

Debug info attached

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/10267923/my-logfile.txt)

Good morning,

After reading the previous #537 issue, where Sigma FP support was asked by an user request on 2020, and celebrating the recent Sigma Fp basic beta support for capture, live view, etc, added on the libgphoto2 2.5.28 release, I've installed the latest the libgphoto2 2.5.30 release and the gphoto2 2.5.28 release to test gphoto2 with the more recent Sigma FP L camera.

I assumed, my fault, that as the Sigma FP was supported by gphoto2, the new Sigma FP L model would be too, but maybe that's not true as I'm not able to capture anything regardless of whether I use the capture-image or capture-image-and-download commands.

In any two cases I'm retrieving the: Sorry, your camera does not support generic capture, error message.

I've attached a debug file to help clarify if something is wrong in my setup or just to confirm that my initial assumption about the Sigma FP L support was too optimistic.

If the debugging messages show the setup it's fine and being aware that the camera is not a very commonly used camera, is there any chance you might consider including the Sigma FP L among the cameras supported by gphoto2?

Can I do something to help further development of this rebel amazing camera? 

Thank you very much

Kind regards

Gabriel

msmeissn, 2022-12-20:
It looks we might just need to add the USB ids to libgphoto2.

I have added them in libgphoto2 master, if you can try to build and test it it would be appreciated.

G-DP, 2022-12-20:
Of course, I'll download it, reinstall it and give a try.
Thank you very much

G-DP, 2022-12-20:
I have built a new libgphoto2 (libgphoto2 - 2.5.30.1) based on the master version that includes today updates. Now the camera is recognized!!!. I've been able to do some capture tests but the results are a bit puzzling. Tomorrow I will continue with the capture tests and with the debug messages to attach them here.

Thanks again for your support

G-DP, 2022-12-21:
Here the steps executed in the tests.

I've attached some of the command's debug files, except for the ones of the CaptureImageDownload commands, cause each of them are very big text files 124Mb.

I understand that the hexdump inside those CaptureImageDownload debug files cause this big size. Can I remove the major part of those lines to help reduce file size in order to attach, at least one of them, here or are they necessary for troubleshooting?

**STEP 1**
$ lsusb > lsusb-v2.txt
[lsusb-v2.txt](https://github.com/gphoto/libgphoto2/files/10278027/lsusb-v2.txt)

**STEP 2**
$ gphoto2 --summary > summary-v2.txt
[summary-v2.txt](https://github.com/gphoto/libgphoto2/files/10278033/summary-v2.txt)

**STEP 3**
$ gphoto2 --list-all-config > list-all-config-v2.txt
[list-all-config-v2.txt](https://github.com/gphoto/libgphoto2/files/10278038/list-all-config-v2.txt)

**STEP 4**
$ gphoto2 --capture-preview --debug --debug-logfile=SigmaFPL-CapturePreview-debug.txt
_Saving file as capture_preview.jpg_
[SigmaFPL-CapturePreview-debug.txt](https://github.com/gphoto/libgphoto2/files/10278054/SigmaFPL-CapturePreview-debug.txt)

**STEP 5**                                         
$ gphoto2 --capture-image-and-download --debug --debug-logfile=SigmaFPL-CaptureImageDownload-debug.txt
                                                                               
_*** Error ***              
PTP General Error
ERROR: Could not capture image.
ERROR: Could not capture._

**STEP 6** 
(Verify if image was captured even the error message: No, no new image in folder except for the previous capture_preview.jpg)

**STEP 7** 
(Repeat capture generating new debug file. Image is captured)
$ gphoto2 --capture-image-and-download --debug --debug-logfile=SigmaFPL-CaptureImageDownloadTry2-debug.txt
_New file is in location /SDIM0001.JPG on the camera                            
Saving file as SDIM0001.JPG
Deleting file /SDIM0001.JPG on the camera_

**STEP 8** 
(Continue with captures adding new filenames. Try3.jpg was not generated in folder, as requested. Not existant)

$ gphoto2 --capture-image-and-download --debug --debug-logfile=SigmaFPL-CaptureImageDownloadTry3-debug.txt --filename Try3.jpg
                                                                               
_*** Error ***              
PTP General Error
ERROR: Could not capture image.
ERROR: Could not capture._

**STEP 9** 
(Continue with captures adding new filenames. Try4.jpg was generated in folder, as requested)

$ gphoto2 --capture-image-and-download --debug --debug-logfile=SigmaFPL-CaptureImageDownloadTry4-debug.txt --filename Try4.jpg
_New file is in location /SDIM0002.JPG on the camera                            
Saving file as Try4.jpg
Deleting file /SDIM0002.JPG on the camera_

**STEP 10** 
(Continue with captures adding new filenames. Try5.jpg was generated in folder, as requested but image content its a clone of Try4.jpg and not what was in front of the camera. File size it's identical of Try4.jpg, also modified time)

$ gphoto2 --capture-image-and-download --debug --debug-logfile=SigmaFPL-CaptureImageDownloadTry5-debug.txt --filename Try5.jpg
_New file is in location /SDIM0002.JPG on the camera                            
Saving file as Try5.jpg
Deleting file /SDIM0002.JPG on the camera_

**STEP 11** 
(Continue with captures adding new filenames. Try6.jpg was generated in folder, as requested but image content its a clone of Try4.jpg and not what was in front of the camera. File size it's identical of Try4.jpg, also modified time)

$ gphoto2 --capture-image-and-download --debug --debug-logfile=SigmaFPL-CaptureImageDownloadTry6-debug.txt --filename Try6.jpg
_New file is in location /SDIM0002.JPG on the camera                            
Saving file as Try6.jpg
Deleting file /SDIM0002.JPG on the camera_

**STEP 12**
(Ending captures as results are the same)

andreamassetti, 2023-03-21:
Is there any development on the support for the Sigma fp-L ?

msmeissn, 2023-03-27:
at least something is working even if its a bit unstable. :/

G-DP, 2023-03-27:
Dear Marcus I have not been able to capture 3 photos correctly using gphoto2 together with the Sigma FP L. My conclusion after testing is that the Sigma FP L is not compatible with the current version of gphoto. 

I have the debug files of previously performed tests, on Dec 21, 2022, and can contribute with new ones in case anyone is interested in further development.

gabrielesartor, 2023-04-17:
Any news on this topic?

andreamassetti, 2023-04-17:
Dear Marcus, like everyone else I am having trouble getting the camera to capture multiple images and I also can't download them. 
I was able to use the camera through the CameraControlSDK for Windows provided by Sigma, I am attaching two different outputs from the sample program included in the SDK.
This is what gets printed after connecting the camera to the computer:
_**
[19:21:07]	image file save path is C:\Users\andrea\Pictures
[19:21:07]	[VID]vid_1003	[PID]pid_c442	[Serial]0000000091506172
[19:21:07]	Add Camera [Serial]0000000091506172
[19:21:08]	[Command] SetCamDataGrp1
[19:21:08]	    [OpeCode] 0x9016
[19:21:08]	        [Data] 0x00,0x00,0x00,0x00,0x04,0x00,0x00,0x00,0x05,0x13,0x00,0x58,0x08,0x20,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
[19:21:08]	        [CheckSum] 0x98
[19:21:08]	[Command] SetCamDataGrp2
[19:21:08]	    [OpeCode] 0x9017
[19:21:08]	        [Data] 0x00,0x00,0x00,0x00,0x04,0x00,0x00,0x00,0x03,0x04,0x00,0x04,
[19:21:08]	        [CheckSum] 0x0B
[19:21:08]	[Command] SetCamDataGrp3
[19:21:08]	    [OpeCode] 0x9018
[19:21:08]	        [Data] 0x00,0x00,0x00,0x00,0x04,0x00,0x00,0x00,0x03,0x00,0x80,0x03,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
[19:21:08]	        [CheckSum] 0x86
[19:21:08]	[Command] SetCamDataGrp4
[19:21:08]	    [OpeCode] 0x9024
[19:21:08]	        [Data] 0x00,0x00,0x00,0x00,0x04,0x00,0x00,0x00,0x03,0x01,0x00,0x02,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,
[19:21:08]	        [CheckSum] 0x06
[19:21:08]	[Command] GetCamCanSetInfo5
[19:21:08]	    [OpeCode] 0x9030
[19:21:08]	        [Data] 0x00,0x00,0x00,0x00,0x03,0x00,0x00,
[19:21:08]	        [CheckSum] 0x00
[19:21:08]	    [Response] 0x2001
[19:21:08]	        [Data] 0x1C,0x05,0x00,0x00,0x58,0x00,0x00,0x00,0x01,0x00,0x01,0x00,0x04,0x00,0x00,0x00,0x05,0x03,0x04,0x01,0x02,0x00,0x01,0x00,0x03,0x00,0x00,0x00,0x03,0x02,0x01,0x00,0x03,0x00,0x01,........
[19:21:08]	        [CheckSum] 0xE8
[19:21:08]	[Command] GetCamDataGrp1
[19:21:08]	    [OpeCode] 0x9012
[19:21:08]	        [Data] 0x00,0x00,0x00,0x00,0x03,0x00,0x00,
[19:21:08]	        [CheckSum] 0x00
[19:21:08]	    [Response] 0x2001
[19:21:08]	        [Data] 0x13,0xFF,0x7F,0x58,0x20,0x00,0x01,0x20,0x00,0x00,0x01,0x09,0x82,0x0A,0x01,0xA0,0x05,0x02,0x00,0x00,
[19:21:08]	        [CheckSum] 0x68
[19:21:08]	[Command] GetCamDataGrp2
[19:21:08]	    [OpeCode] 0x9013
[19:21:08]	        [Data] 0x00,0x00,0x00,0x00,0x03,0x00,0x00,
[19:21:08]	        [CheckSum] 0x00
[19:21:08]	    [Response] 0x2001
[19:21:08]	        [Data] 0x0F,0x3F,0xFD,0x01,0x02,0x04,0x01,0x00,0x01,0x00,0x00,0x00,0x00,0x01,0xFF,0x02,
[19:21:08]	        [CheckSum] 0x56
[19:21:08]	[Command] GetCamDataGrp3
[19:21:08]	    [OpeCode] 0x9014
[19:21:08]	        [Data] 0x00,0x00,0x00,0x00,0x03,0x00,0x00,
[19:21:08]	        [CheckSum] 0x00
[19:21:08]	    [Response] 0x2001
[19:21:08]	        [Data] 0x10,0xFF,0xA3,0x00,0x00,0x00,0x01,0x03,0x01,0xA0,0x05,0xA0,0x05,0x02,0x05,0x05,0x03,
[19:21:08]	        [CheckSum] 0x10
[19:21:08]	[Command] GetCamDataGrp4
[19:21:08]	    [OpeCode] 0x9023
[19:21:08]	        [Data] 0x00,0x00,0x00,0x00,0x03,0x00,0x00,
[19:21:08]	        [CheckSum] 0x00
[19:21:08]	    [Response] 0x2001
[19:21:08]	        [Data] 0x11,0xF0,0x3F,0x00,0x03,0x01,0x03,0xFF,0x0E,0x00,0x01,0x01,0x02,0x01,0xFF,0x02,0x02,0x05,
[19:21:08]	        [CheckSum] 0x61
[19:21:08]	[Command] GetCamDataGrp5
[19:21:08]	    [OpeCode] 0x9027
[19:21:08]	        [Data] 0x00,0x00,0x00,0x00,0x03,0x00,0x00,
[19:21:08]	        [CheckSum] 0x00
[19:21:08]	    [Response] 0x2001
[19:21:08]	        [Data] 0x0C,0x2B,0x00,0x1E,0x00,0x02,0x00,0x00,0x02,0x88,0x13,0x03,0x01,
[19:21:08]	        [CheckSum] 0xF8
[19:21:08]	[Command] GetCamDataGroupFocus
[19:21:08]	    [OpeCode] 0x9031
[19:21:08]	        [Data] 0x00,0x00,0x00,0x00,0x03,0x00,0x00,
[19:21:08]	        [CheckSum] 0x00
[19:21:08]	    [Response] 0x2001
[... 370 lines cut ...]
[19:30:37]	[Command] GetCamCaptStatus
[19:30:37]	    [OpeCode] 0x9015
[19:30:37]	        [Data] 0x01,0x00,0x00,0x00,0x03,0x00,0x00,
[19:30:37]	        [CheckSum] 0x00
[19:30:37]	    [Response] 0x2001
[19:30:37]	        [Data] 0x06,0x1A,0x01,0x01,0x00,0x00,0x00,
[19:30:37]	        [CheckSum] 0x22
[19:30:37]	[Command] GetCamCaptStatus
[19:30:37]	    [OpeCode] 0x9015
[19:30:37]	        [Data] 0x01,0x00,0x00,0x00,0x03,0x00,0x00,
[19:30:37]	        [CheckSum] 0x00
[19:30:37]	    [Response] 0x2001
[19:30:37]	        [Data] 0x06,0x1B,0x01,0x01,0x00,0x00,0x00,
[19:30:37]	        [CheckSum] 0x23
[19:30:38]	[Command] GetCamCaptStatus
[19:30:38]	    [OpeCode] 0x9015
[19:30:38]	        [Data] 0x01,0x00,0x00,0x00,0x03,0x00,0x00,
[19:30:38]	        [CheckSum] 0x00
[19:30:38]	    [Response] 0x2001
[19:30:38]	        [Data] 0x06,0x00,0x01,0x01,0x00,0x00,0x00,
[19:30:38]	        [CheckSum] 0x08
[19:30:38]	    [CaptStatus] Uninitialized / Cleared [ImageID = 0x00]
**_
Please let me know if there is a way to use this information to solve the problems in libgphoto2 for the Sigma fp L.
p.s.: Sigma provides many .dll files for Windows inside the CameraControlSDK, maybe they could be useful.

gabrielesartor, 2023-05-09:
> Dear Marcus I have not been able to capture 3 photos correctly using gphoto2 together with the Sigma FP L. My conclusion after testing is that the Sigma FP L is not compatible with the current [...]

Hi @G-DP !
Did you solve your problem?

G-DP, 2023-05-09:
> Hi @G-DP ! Did you solve your problem?

Hi Gabriele,

No, I couldn't get any further with Gphoto2 and the Sigma FP L. A pity since it is a very promising application.

gabrielesartor, 2023-10-16:
Hi @msmeissn,
I've seen that there's a new commit on this issue, any news?

Damani2, 2023-11-19:
Also hoping for an update on this. I’ve been trying to use the FPL tethered in Darktable to no avail.

Damani2, 2024-01-05:
Updating this here. The FPL works in Darktable which uses a Gphoto dependency. But none of the camera settings work. I can't control shutter speed, aperture, or iso. Any advice for this? I love Darktable and don't want to have to use capture one.

ascertrobw, 2025-01-31:
I noticed the guys over on CloudyNights have managed to create an ASCOM driver for the Sigma FP L, with control of shutter & aperture and pull back of images. Have responded on the thread to see if the source code has been shared anyway to see if it can help. I have an FP L, and I'm a coder by trade - although I've not worked in this specific area or with this SDK. Including thread link below for info. Will report back if I hear more:

https://www.cloudynights.com/topic/879238-ascom-driver-for-sigma-fp-l/