gphoto/libgphoto2 #887: Ability to specify on-camera capture filenames

spiderkeys, 2023-01-27:
Is there a limitation with the PTP capture interface for most cameras that requires libgphoto2 to ignore the path provided by the user to `gp_camera_capture()` and instead generate the `/captXXXX.Y` folder/filename pattern?
```c
	/* NOP: This gets overridden in the library to /capt0000.jpg */
	strcpy(camera_file_path.folder, "/");
	strcpy(camera_file_path.name, "foo.jpg");

	retval = gp_camera_capture(camera, GP_CAPTURE_IMAGE, &camera_file_path, context);
```
I would like to be able to explicitly set this filename (folder too, if possible), specifically for the purpose of being able to include a timestamp and other metadata in the filename, such that it can be later correlated to some sensor data in post-processing.

If there is no strict limitation or requirement for the files to match this format, that would be helpful to know. I can imagine scenarios where the camera itself might have some expectation/requirement for the directory and filenames to match these specific patterns, but I thought I'd check before making any modifications.

Thanks

msmeissn, 2023-01-27:
The path is returned by the libgphoto2 library from gp_camera_capture, it is not an input parameter.

This path is a path into the virtual libgphoto2 filesystem. After downloading it from the virtual filesystem you can select your own folder and filename where to save the image to disk.

(e.g. this is what gphoto2 --filename "..." option is doing)

spiderkeys, 2023-02-02:
Understood for `capturetarget==sdram`. When `capturetarget` is set to "card", is there any way to influence the filename written to the SD card (i.e. `Pathname on the camera: /store_00020001/DCIM/100FOCUS/021A6916.JPG`), or is that purely camera vendor controlled?

msmeissn, 2023-02-03:
the on camera path is entirely controlled by the cameras, have not seen any camera that allows configuration of it.

spiderkeys, 2023-02-03:
Thanks, will close this issue then!

closed 2023-02-03 by spiderkeys