gphoto/libgphoto2 #888: Add support for Fuji XT5

xiliugatech, 2023-01-27:
Can you add support for Fuji XT5? Thanks

Here's what I got with a xt5 connected to a desktop
gphoto2 --summary
                                                                               
*** Error ***              
Could not detect any camera
*** Error (-105: 'Unknown model') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --summary

Please make sure there is sufficient quoting around the arguments.

msmeissn, 2023-01-28:
this looks like the camera might not be detected by the system.

is it visible when you run

lsusb

xiliugatech, 2023-01-28:
yes, lsusb shows:

04cb:02fc Fuji Photo Film Co., Ltd USB PTP Camera

msmeissn, 2023-01-29:
it should detect at least USB PTP Camera ..

can you quote
gphoto2 --version

to see if it was built with USB support?

xiliugatech, 2023-01-29:
Thank you. It was a install problem. After installing with libusb-dev now it works.

xiliugatech, 2023-01-29:
I was able to find the camera but the --summary says this:

```
Camera summary:                                                                
Manufacturer: FUJIFILM
Model: X-T5
  Version: 1.00
  Serial Number: 593536383831221207F4A310211F05
Vendor Extension ID: 0xe (1.0)
Vendor Extension Description: fujifilm.co.jp: 1.0; 

Capture Formats: 
Display Formats: JPEG, JFIF, Unknown(3812)

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:
store_10000001:
	StorageDescription: External Memory
	VolumeLabel: 000000000000000000000000000000000000000000000000
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read Only with Object deletion
	Maximum Capability: 0 (0 MB)
	Free Space (Bytes): 0 (0 MB)
	Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Range [0 - 10, step 1] value: 10% (10)
Property 0xd303:(read only) (type=0x2) 1
Property 0xd406:(readwrite) (type=0xffff) ''
Property 0xd407:(read only) (type=0x6) 1
Exposure Time(0x500d): error 200a on query.
F-Number(0x5007): error 200a on query.
LensZoomPosCaps(0xd38c): error 200a on query.
FocusPosition(0xd171): error 200a on query.
Property 0xd21c: error 200a on query.
Focus Point(0xd347): error 200a on query.
LensZoomPos(0xd170): error 200a on query.
BatteryLevel(0xd242): error 200a on query.
LiveViewImageSize(0xd174): error 200a on query.
VideoOutOnOff(0xd168): error 200a on query.
LiveViewImageQuality(0xd173): error 200a on query.
ForceMode(0xd230): error 200a on query.
Property 0xd16e: error 200a on query.
FocusArea1(0xd372): error 200a on query.
FaceDetectionMode(0xd020): error 200a on query.
RawCompression(0xd022): error 200a on query.
GrainEffect(0xd023): error 200a on query.
SetEyeAFMode(0xd024): error 200a on query.
FocusPoints(0xd025): error 200a on query.
MFAssistMode(0xd026): error 200a on query.
InterlockAEAFArea(0xd027): error 200a on query.
Shadowing(0xd029): error 200a on query.
CropMode(0xd16f): error 200a on query.
TNumber(0xd02f): error 200a on query.
FocusArea4(0xd395): error 200a on query.
HighLightTone(0xd320): error 200a on query.
ShadowTone(0xd321): error 200a on query.
LongExposureNR(0xd322): error 200a on query.
FullTimeManualFocus(0xd323): error 200a on query.
LensISSwitch(0xd346): error 200a on query.
InstantAFMode(0xd34a): error 200a on query.
PreAFMode(0xd34b): error 200a on query.
LMOMode(0xd34d): error 200a on query.
ISMode(0xd351): error 200a on query.
FocusCheckMode(0xd35e): error 200a on query.
LiveViewImageQuality(0xd173): error 200a on query.
FileNamePrefix1(0xd365): error 200a on query.
FileNamePrefix2(0xd366): error 200a on query.
FocusArea3(0xd374): error 200a on query.
TotalShotCount(0xd310): error 200a on query.
ExposurePreview(0xd359): error 200a on query.
FrameGuideGridInfo1(0xd375): error 200a on query.
FrameGuideGridInfo2(0xd376): error 200a on query.
CustomDispInfo(0xd36e): error 200a on query.
ViewMode1(0xd33f): error 200a on query.
CustomAutoPowerOff(0xd364): error 200a on query.
LockButtonMode(0xd34e): error 200a on query.
WideDynamicRange(0xd02e): error 200a on query.
LensNameAndSerial(0xd36d): error 200a on query.
LensUnknownData(0xd38a): error 200a on query.
BatteryInfo1(0xd36a): error 200a on query.
BatteryInfo2(0xd36b): error 200a on query.
FunctionLockCategory1(0xd36f): error 200a on query.
FunctionLockCategory2(0xd370): error 200a on query.
SensitivityFineTune1(0xd222): error 200a on query.
SensitivityFineTune2(0xd223): error 200a on query.
LensZoomPosCaps(0xd38c): error 200a on query.
LensFNumberList(0xd38d): error 200a on query.
LensFocalLengthList(0xd38e): error 200a on query.
Property 0xd17b: error 200a on query.
```

Looks like we can't read the camera information using PTP?

msmeissn, 2023-01-31:
It does not report any property besides battery level. I see indication remote control is possible, but perhaps fuji has changed the code again on how to do that :(

xiliugatech, 2023-01-31:
after setting the capture mode to USB auto the --summary now works ok. The only problem we had is that seems like Fuji change the register values and looks like the wb fine tune registers are different now. For instance, when I try to set 0xD00C via PTP it returns 0x2002 error code.

msmeissn, 2023-02-02:
0x2002 is PTP_RC_GeneralError so a generic error.

does it also have the 0x2fc product id in this mode?
can you run: 

- gphoto2 --summary --debug --debug-logfile=xt5-debug.txt > summary.txt
- gphoto2  --list-all-config > list-all-config.txt 

and attach xt5-debug.txt summary.txt list-all-config.txt  here?

xiliugatech, 2023-02-03:
Thanks for the follow up. Turns out that we set the wrong white balance mode in the camera and was unable to overwrite register setting. Now we are able to use PTP to communicate to the camera and I think XT5 is supported. You might want to update gphoto2 document to add XT5 to the supported camera. Thanks.

closed 2023-02-03 by xiliugatech

msmeissn, 2023-02-04:
does it have the 0x2fc usb product id in the mode that works?

xiliugatech, 2023-02-05:
Yes, I can read that.