gphoto/libgphoto2 #889: Nikon Z30 Support

Nic2012, 2023-02-07:
Tested with Pi Zero 2W
```
Raspbian GNU/Linux Debian 11 (bullseye)
```

gphoto2 --version
```
gphoto2         2.5.27         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.27         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

Fill in fields below. If you don't know how, leave it free.
Use the Marketing name.
```
NIKON Z 30
```

**USB IDs**
```
Bus 001 Device 002: ID 04b0:0452 Nikon Corp. NIKON DSC Z 30
```

**camera summary output**
[summary@z30.txt](https://github.com/gphoto/libgphoto2/files/10673154/summary%40z30.txt)

gphoto2 --list-all-config > list-all-config.txt
[list-all-config@z30.txt](https://github.com/gphoto/libgphoto2/files/10673156/list-all-config%40z30.txt)

**test capture**
  gphoto2 --capture-image-and-download
```
New file is in location /capt0000.jpg on the camera
Saving file as capt0000.jpg
Deleting file /capt0000.jpg on the camera
```

Nic2012, 2023-02-07:
gphoto2 --auto-detect
```
USB PTP Class Camera           usb:001,002
```

msmeissn, 2023-02-07:
thanks for the report! 

id is in git for 2.5.31, but as you see it already works as we support the Z series as class based camera.

Nic2012, 2023-02-07:
yes, but it seems dependants like Darktable are not working without the id: https://github.com/gphoto/libgphoto2/issues/819

msmeissn, 2023-02-07:
then it will probably need a new release to pick those up. have not yet have not yet scheduled one, but i think i have enough fixes now to do one in the next weeks.

Nic2012, 2023-02-07:
I have installed the latest version on the Raspbian GNU/Linux Debian 11 and it works fine:
```
$ gphoto2 --auto-detect
Model                          Port
----------------------------------------------------------
Nikon Z30                      usb:001,003
```
Thank you for your efforts and keeping this valuable library still alive :)

closed 2023-02-07 by Nic2012

ngreve, 2023-04-08:
I am thinking about buying that camera.
Is it possible to update the [`Remote controling cameras`](http://gphoto.org/doc/remote/) documentation and include the Z30 with the supported features?