gphoto/libgphoto2 #892: Sony Alpha 1 | ILCE-1

karlkopf, 2023-02-14:
**Name of the camera**
Sony Alpha 1
ILCE-1

**USB IDs**
054c:0d1b

**camera summary output**
[summary.txt](https://github.com/gphoto/libgphoto2/files/10732319/summary.txt)

**camera configuration output**
[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/10732321/list-all-config.txt)

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
```*** Error ***
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***
```

  gphoto2 --capture-preview
```
*** Error (-7: 'I/O problem') ***
```

It's not possible to capture an image or to set any configuration option.

Do you think you can add the camera?

karlkopf, 2023-02-15:
I have tested the camera for a while and there are some things that didn't working as expected.

- It's not possible to set the shutterspeed (setting keep unchanged)
- It's not possible to download files from the sdcard (no files are listed)
- It's not possible to download images from the sdram (--wait-event-and-download=1)
- At the moment it's not possible anymore to do a --capture-image-and-download

I have uploaded all the debug logs to a repository, please have a look and try to find the issue

[debug_logs](https://github.com/karlkopf/ILCE-1_debug)

If you need anymore logs or tests I can do it and report the results

msmeissn, 2023-02-17:
(sorry, currently clear lack of time, also saw your email, still need to find time to reply and to dig into this)

One point

- in Sony PC Control mode there is no SD card access, this is a camera side limitation.
For the rest I will try to find time in the next days.

karlkopf, 2023-02-17:
Thanks you for the info.

I managed to get pictures from the camera using the tethered function of gphoto and can do a workaround for my problem.

If you need anything else like logs or some tests I'll help you out