gphoto/libgphoto2 #896: Sony FX3 control overshoot

nrplyx, 2023-03-01:
Hey!
Trying to control the Sony FX3 camera, however, control seems to overshoot controls for ISO / Iris (Shutterspeed works properly).
When I use gphoto2 --set-config iso=400 command, I can see the ISO value is stepping to value 400, however, once the value is reached, actual ISO jumps immediately to AUTO.

A similar situation is for Iris control, it seems to be stepping correctly to desired value, but then it jumps to maximum aperture f22 or f2.8 depending on the value. If trying to set Aperture f8.0 and more, final value is f22, if I set Aperture f7.1 and below, final value will be f2.8.

Output of "gphoto2 --debug --set-config iso=400":
[https://sharetext.me/raw/fzkrxojvwh](https://sharetext.me/raw/fzkrxojvwh)

Any help will be appreciated.

nrplyx, 2023-03-06:
I played a bit around and If I change camlibs/ptp2/config.c code in "PUT_SONY_VALUE_" macro definition to return GP_ERROR at the moment of "Value matched!" moment (when the code steps to proper value) it seems to not overshoot anymore (it of course returns error, but does the job). So I guess the issue is hidden in the code following this procedure.

#define PUT_SONY_VALUE_(bits,inttype) 								       \
...
		do {									                                                \
			C_PTP_REP (ptp_sony_getalldevicepropdesc (params));		                \
			C_PTP_REP (ptp_generic_getdevicepropdesc (params, prop, &dpd));	\
											                        \
			if (dpd.CurrentValue.bits == value) {				\
				printf ("Value matched!\n");	                                 \
				**return GP_ERROR;**		                                         \

msmeissn, 2023-03-06:
well spotted, this was doing unnecessary work.

nrplyx, 2023-03-06:
I can not confirm this as a solution, I just tried to apply fix [17bc414](https://github.com/gphoto/libgphoto2/commit/17bc4149da9cab810327de36d84506c096197a2a), but FX3 overshooting remains for iso and aperture.

Maybe I should remark in my [comment](https://github.com/gphoto/libgphoto2/issues/896#issuecomment-1455704670) that if I use escape "return GP_OK;", overshoot remains. Only when "return GP_ERROR;" is used as an escape after "Value Matched!" event, overshooting disappears.
I believe the error is not in this macro, but in the code after the macro.

msmeissn, 2023-03-06:
ok, background:

- we do the stepping method and it arrives at the right place.
- and then the code still calls ptp_sony_setdevicecontrolvaluea(PTP_DPC_SONY_ISO) , which usually would be ignored, but here succeeds instead of being ignored and leads to the incorrect result.

msmeissn, 2023-03-06:
(need to add code to handle this)

msmeissn, 2023-03-06:
can you try current git, i add some code to the apis

nrplyx, 2023-03-07:
Thanks!
I just tested [51b3c23](https://github.com/gphoto/libgphoto2/commit/51b3c23cfe5f5ff251ac819ab0ed2af4b6e9cfdb) and the overshooting issue is solved!

closed 2023-03-11 by msmeissn