gphoto/libgphoto2 #898: Support Nikon Coolpix A10

amano-kenji, 2023-03-09:
**Name of the camera**

Nikon Coolpix A10

**USB IDs**

ID 04b0:0368 Nikon Corp. NIKON DSC COOLPIX A10-PTP

**camera summary output**

> *** Error ***
An error occurred in the io-library ('I/O problem'): The supplied vendor or product id (0x0,0x0) is not valid.
*** Error (-7: 'I/O problem') ***

**camera configuration output**

> *** Error ***
An error occurred in the io-library ('I/O problem'): The supplied vendor or product id (0x0,0x0) is not valid.
*** Error (-7: 'I/O problem') ***

**test capture**

  gphoto2 --capture-image-and-download

> *** Error ***
An error occurred in the io-library ('I/O problem'): The supplied vendor or product id (0x0,0x0) is not valid.
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error ***
 
  gphoto2 --capture-preview

> *** Error ***
An error occurred in the io-library ('I/O problem'): The supplied vendor or product id (0x0,0x0) is not valid.
*** Error (-7: 'I/O problem') ***

amano-kenji, 2023-03-10:
gphotofs used to work for Nikon Coolpix A10. But, after a certain point, it stopped working.

msmeissn, 2023-05-09:
can you post 

gphoto2 --summary

output?

amano-kenji, 2023-05-10:
```
*** Error ***
An error occurred in the io-library ('I/O problem'): The supplied vendor or product id (0x0,0x0) is not valid.
*** Error (-7: 'I/O problem') ***

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --summary

Please make sure there is sufficient quoting around the arguments.
```

amano-kenji, 2025-06-27:
[my-logfile.txt](https://github.com/user-attachments/files/20957467/my-logfile.txt)

amano-kenji, 2025-06-27:
my-logfile.txt says I don't have enough permission for USB device. That's probably caused because I don't use logind which automatically attaches appropriate permissions to devices.

amano-kenji, 2025-06-28:
After adding

```
# PTP/MTP devices can be accessed by seat group
SUBSYSTEM=="usb", ENV{ID_USB_INTERFACES}=="*:060101:*", GROUP+="seat", MODE:="660", GOTO="usb_end"
```

to /etc/udev/rules.d/99-local.rules, I can access the camera with gphotofs although the camera seems to drop out for some reasons after a while.

amano-kenji, 2025-06-28:
logind attaches uaccess tag to PTP/MTP devices through

```
# PTP/MTP protocol devices, cameras, portable media players
SUBSYSTEM=="usb", ENV{ID_USB_INTERFACES}=="*:060101:*", TAG+="uaccess"
```

in /usr/lib/udev/rules.d/70-uaccess.rules.

I merely adapted it to my own custom setup. It seems everything assumes systemd-logind nowadays. I replaced logind with seatd.

closed 2025-06-28 by amano-kenji