gphoto/libgphoto2 #899: Support Fujifilm X-T5

jacceko, 2023-03-12:
**Camera**: Fujifilm X-T5  
(camera digitally almost identical to Fujifilm X-H2 which is on merge - the same processor and the same matrix)

**lsusb** (after connect camera in mode "USB tether shooting AUTO")
Bus 004 Device 002: ID 04cb:02fc Fuji Photo Film Co., Ltd USB PTP Camera

**Camera summary:**
Manufacturer: FUJIFILM
Model: X-T5
  Version: 1.00
  Serial Number: 593536383831221101F263104114E3
Vendor Extension ID: 0xe (1.0)
Vendor Extension Description: fujifilm.co.jp: 1.0; 

Capture Formats: JPEG Unknown(3812) Defined Type Unknown(b103)
Display Formats: JPEG, Unknown(3812), Unknown(5000), Defined Type, Unknown(b103), TIFF

Device Capabilities:
	File Download, File Deletion, File Upload
	Generic Image Capture, Open Capture, No vendor specific capture

Storage Devices Summary:
store_10000001:
	StorageDescription: Still
	VolumeLabel: 1593536383831221101F263104114E3
	Storage Type: Builtin RAM
	Filesystemtype: Generic Hierarchical
	Access Capability: Read Only with Object deletion
	Maximum Capability: 18446744073709551615 (17592186044415 MB)
	Free Space (Bytes): 18446744073709551615 (17592186044415 MB)
	Free Space (Images): -1
store_10000002:
	StorageDescription: Live
	VolumeLabel: 2593536383831221101F263104114E3
	Storage Type: Builtin RAM
	Filesystemtype: Generic Hierarchical
	Access Capability: Read Only with Object deletion
	Maximum Capability: 18446744073709551615 (17592186044415 MB)
	Free Space (Bytes): 18446744073709551615 (17592186044415 MB)
	Free Space (Images): -1

Device Property Summary:
Image Size(0x5003):(readwrite) (type=0xffff) Enumeration [
	'3888x2592',
	'3888x2184',
	'2592x2592',
	'5472x3648',
	'5472x3080',
	'3648x3648',
	'7728x5152',
	'7728x4344',
	'5152x5152',
	'3456x2592',
	'3264x2592',
	'6864x5152',
	'6432x5152',
	'4864x3648',
	'4560x3648'
	] value: '7728x5152'
White Balance(0x5005):(readwrite) (type=0x4) Enumeration [2,4,6,8,32769,32770,32771,32774,32775,32776,32777,32778,32800,32801] value: Automatic (2)
F-Number(0x5007):(readwrite) (type=0x4) Enumeration [200] value: f/2 (200)
Focus Mode(0x500a):(readwrite) (type=0x4) Enumeration [1,32769,32770] value: 32769
Exposure Metering Mode(0x500b):(readwrite) (type=0x4) Enumeration [1,2,3,4] value: Average (1)
Exposure Time(0x500d):(readwrite) (type=0x6) Enumeration [7812] value: 0.078 sec (7812)
Exposure Program Mode(0x500e):(readwrite) (type=0x4) Enumeration [1,3,4,6] value: 6
Exposure Index (film speed ISO)(0x500f):(readwrite) (type=0x5) Enumeration [64,80,100,125,160,200,250,320,400,500,640,800,1000,1250,1600,2000,2500,3200,4000,5000,6400,8000,10000,12800,25600,51200,-1,-2,-3] value: ISO -3 (-3)
Exposure Bias Compensation(0x5010):(readwrite) (type=0x3) Enumeration [-5000,-4667,-4333,-4000,-3667,-3333,-3000,-2667,-2333,-2000,-1667,-1333,-1000,-667,-333,0,333,667,1000,1333,1667,2000,2333,2667,3000,3333,3667,4000,4333,4667,5000] value: 0.0 stops (0)
Date & Time(0x5011):(readwrite) (type=0xffff) '20230227T215700'
Pre-Capture Delay(0x5012):(readwrite) (type=0x4) Enumeration [10000,2000,0] value: 0.0s (0)
Sharpness(0x5015):(readwrite) (type=0x3) Range [-40 - 40, step 10] value: 0
[... 2929 lines cut ...]
Readonly: 1
Type: TEXT
Current: 200,220,250,280,320,360,400,450,500,560,640,710,800,900,1000,1100,1300,1400,1600
END
/main/other/d38e
Label: LensFocalLengthList
Readonly: 1
Type: TEXT
Current: 9000,13700
END
/main/other/d17b
Label: PTP Property 0xd17b
Readonly: 0
Type: MENU
Current: 1
Choice: 0 1
END

**test capture**
gphoto2 --capture-image
New file in path /store_10000001/DSCF0001.jpg in camera

**previe capture**
gphoto2 --capture-preview
*** Error (-1: 'Unspecified error') ***

msmeissn, 2023-03-17:
what is the
 gphoto2 --version

capture works already as far as your report shows.

msmeissn, 2023-03-17:
and for preview capture

gphoto2 --capture-preview --debug --debug-logfile=preview.txt  and attach preview.txt here

jacceko, 2023-03-17:
gphoto2 --version
2.5.27

jacceko, 2023-03-17:
gphoto2 --capture-preview --debug --debug-logfile=preview.txt 
[preview.txt](https://github.com/gphoto/libgphoto2/files/11007018/preview.txt)

msmeissn, 2023-08-31:
id is in git now, all looks good

closed 2023-08-31 by msmeissn