gphoto/libgphoto2 #902: Could not claim the USB device

estatistics, 2023-04-09:
gphoto2 --camera "GoPro HERO5 Black" --list-files
                                                                               
*** Error ***              
An error occurred in the io-library ('Could not claim the USB device'): Could not claim interface 0 (Συσκευή ή πόροι είναι απασχολημένοι). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
*** Error (-53: 'Could not claim the USB device') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --camera "GoPro HERO5 Black" --list-files

Please make sure there is sufficient quoting around the arguments.

estatistics, 2023-04-09:
Note that running sudo krusader ... i can access, and copy files from my camera. (other browsers dont recognize GoPro!!!).

estatistics, 2023-04-09:
```
0.000014 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000037 main                        (2): gphoto2 2.5.28
0.000041 main                        (2): gphoto2 has been compiled with the following options:
0.000043 main                        (2):  + gcc (C compiler used)
0.000046 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000048 main                        (2):  + exif (for displaying EXIF information)
0.000051 main                        (2):  + cdk (for accessing configuration options)
0.000053 main                        (2):  + aa (for displaying live previews)
0.000055 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000058 main                        (2):  + readline (for easy navigation in the shell)
0.000062 main                        (2): libgphoto2 2.5.30
0.000065 main                        (2): libgphoto2 has been compiled with the following options:
0.000068 main                        (2):  + standard camlib set (ax203 canon digigr8 dimagev directory docupen jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801)
0.000071 main                        (2):  + gcc (C compiler used)
0.000073 main                        (2):  + no ltdl (for portable loading of camlibs)
0.000075 main                        (2):  + EXIF (for special handling of EXIF files)
0.000079 main                        (2): libgphoto2_port 0.12.1
0.000082 main                        (2): libgphoto2_port has been compiled with the following options:
0.000084 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000087 main                        (2):  + gcc (C compiler used)
0.000089 main                        (2):  + no ltdl (for portable loading of iolibs)
0.000091 main                        (2):  + EXIF (for vusb)
0.000094 main                        (2):  + USB (libusb1, for USB cameras)
0.000096 main                        (2):  + serial (for serial cameras)
0.000098 main                        (2):  + no resmgr (serial port access and locking)
0.000100 main                        (2):  + no ttylock (serial port locking)
0.000103 main                        (2):  + no lockdev (serial port locking)
0.000105 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000108 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000110 main                        (2): invoked with following arguments:
0.000113 main                        (2):   --debug
0.000115 main                        (2):   --debug-logfile=my-logfile.txt
0.000117 main                        (2):   --port
0.000120 main                        (2):   usb:003,022
0.000122 main                        (2):   --list-files
0.000132 main                        (2): Processing 'port' option ('usb:003,022')...
0.000138 gp_port_init_localedir      (2): localedir has been set to /usr/share/locale (compile-time default)
0.000141 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/lib/x86_64-linux-gnu/libgphoto2_port/0.12.1'...
0.000183 foreach_func                (2): Called for filename '/usr/lib/x86_64-linux-gnu/libgphoto2_port/0.12.1/disk'.
0.000285 gp_port_library_list        (2): found fstab fsname UUID=C129-0B33
0.000297 gp_port_library_list        (2): found fstab fsname /swapfile
0.000302 gp_port_library_list        (2): found fstab fsname /dev/disk/by-label/L_System
0.000307 gp_port_library_list        (2): found fstab fsname /dev/disk/by-label/HDD_GAMES
0.000313 gp_port_library_list        (2): found fstab fsname /dev/disk/by-label/HDD_MUSICVIDS
0.000319 gp_port_library_list        (2): found fstab fsname /dev/disk/by-label/DOS_InstGAMES_00
0.000325 gp_port_library_list        (2): found fstab fsname /dev/disk/by-label/Basic_HDD
0.000339 gp_port_info_list_lookup_path(2): Looking for path 'disk:/mnt/Basic_HDD' (0 entries available)...
0.000347 gp_port_info_list_lookup_path(2): Starting regex search for 'disk:/mnt/Basic_HDD'...
0.000360 gp_port_library_list        (2): found fstab fsname /dev/disk/by-label/HDD_Movies
0.000366 gp_port_library_list        (2): found fstab fsname /dev/disk/by-label/HDD_Shows
0.000381 gp_port_library_list        (2): found fstab fsname /dev/disk/by-label/HDD_XXX_00_bckp
0.000386 gp_port_library_list        (2): found fstab fsname /dev/disk/by-label/HDD_XXX_00
0.000391 gp_port_library_list        (2): found fstab fsname /dev/disk/by-partlabel/XXX_VIDS_00_bckp
0.000397 gp_port_library_list        (2): found fstab fsname /dev/disk/by-label/XXX_VIDS_00
0.000402 gp_port_library_list        (2): found fstab fsname /dev/disk/by-label/HD_GRSERIES_00
0.000449 gp_port_library_list        (2): found mtab fsname sysfs
0.000455 gp_port_library_list        (2): found mtab fsname proc
0.000458 gp_port_library_list        (2): found mtab fsname udev
0.000465 gp_port_library_list        (2): found mtab fsname devpts
0.000470 gp_port_library_list        (2): found mtab fsname tmpfs
0.000476 gp_port_library_list        (2): found mtab fsname /dev/sda2
0.000482 gp_port_library_list        (2): found mtab fsname securityfs
0.000485 gp_port_library_list        (2): found mtab fsname tmpfs
0.000492 gp_port_library_list        (2): found mtab fsname tmpfs
[... cut ...]
0.210409 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.210412 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.210424 delete_all_folders          (2): Internally deleting all folders from '/'...
0.210427 lookup_folder               (2): Lookup folder '/'...
0.210430 lookup_folder               (2): Found! / is 0x55786b07c7f0
0.210432 recurse_delete_folder       (2): Recurse delete folder 0x55786b07c7f0//
0.210435 delete_all_files            (2): Delete all files in folder 0x55786b07c7f0//

```

estatistics, 2023-04-09:
Also, plese not that your log files can contain sensitive information that i Deleted them.It is not need to list all devices of your pc.

msmeissn, 2023-05-09:
no worries.

basically there is another tool blocking gphoto2 (as only 1 program can access the USB at a time).

it likely is either GVFS or the KDE device access framework. perhaps you can "eject" the camera from that before trying gphoto2.