gphoto/libgphoto2 #903: Panasonic S5M2 support

tangbl93, 2023-04-10:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*

S5M2

**camera summary output**

shell command run with error

** xml

[S5M2.xml.zip](https://github.com/gphoto/libgphoto2/files/11188433/S5M2.xml.zip)

** wireshark pcapng

> by lumix sync app

[S5M2.pcapng.zip](https://github.com/gphoto/libgphoto2/files/11188430/S5M2.pcapng.zip)

** Info

1. using `/cam.cgi?mode=accctrl&type=req_acc` authorization request will  report error
2. to use the lumix sync app's authorization method, two encryption parameters are required. If the parameters are wrong, the camera will always be stuck in the loading process of the connection
3. the two encrypted parameters of the `req_acc_e` authorization method will be updated every time you connect

```
/cam.cgi?mode=accctrl&type=req_acc_g
/cam.cgi?mode=accctrl&type=req_acc_e&value=...&value2=...
```

examples:

```
/cam.cgi?mode=accctrl&type=req_acc_e&
value=7171bb087476c6087474cf117474d20c7c68cf0c6974cf0c7475cd0c7775cf0c7571b978
value2=2b2daf55bbba9a52

/cam.cgi?mode=accctrl&type=req_acc_e&
value=6b11d81a6e16a51a6e14ac036e14b11e6608ac1e7314ac1e6e15ae1e6d15ac1e6f11da6a
&value2=314dcc47a1daf940
```

tomilov, 2023-04-13:
do you know how to calculate the parameters from the password? What does they mean?

msmeissn, 2023-05-09:
dup of 904, lets proceed there.

closed 2023-05-09 by msmeissn