gphoto/libgphoto2 #904: Panasonic S5M2 support

tangbl93, 2023-04-10:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*

S5M2

**camera summary output**

shell command run with error

** xml

[S5M2.xml.zip](https://github.com/gphoto/libgphoto2/files/11188433/S5M2.xml.zip)

** wireshark pcapng

> by lumix sync app

[S5M2.pcapng.zip](https://github.com/gphoto/libgphoto2/files/11188430/S5M2.pcapng.zip)

** Info

1. using `/cam.cgi?mode=accctrl&type=req_acc` authorization request will  report error
2. to use the lumix sync app's authorization method, two encryption parameters are required. If the parameters are wrong, the camera will always be stuck in the loading process of the connection
3. the two encrypted parameters of the `req_acc_e` authorization method will be updated every time you connect

```
/cam.cgi?mode=accctrl&type=req_acc_g
/cam.cgi?mode=accctrl&type=req_acc_e&value=...&value2=...
```

examples:

```
/cam.cgi?mode=accctrl&type=req_acc_e&
value=7171bb087476c6087474cf117474d20c7c68cf0c6974cf0c7475cd0c7775cf0c7571b978
value2=2b2daf55bbba9a52

/cam.cgi?mode=accctrl&type=req_acc_e&
value=6b11d81a6e16a51a6e14ac036e14b11e6608ac1e7314ac1e6e15ae1e6d15ac1e6f11da6a
&value2=314dcc47a1daf940
```

msmeissn, 2023-05-09:
Interesting. likely be harder to reverse engineer how they generate those.

msmeissn, 2023-05-09:
i am not aware of this new method and how it works

req_acc_g  is also there?

tangbl93, 2023-05-10:
@msmeissn `req_acc_g` don't require parameters。
i guess `req_acc_g` is similar to the activation input, and `req_acc_e` is similar to submit the passcode。

Arcnor, 2023-05-12:
Is there any reason to support this encrypted HTTP based one, instead of PTP?

dyylaaan, 2023-09-08:
`gphoto2 --capture-image` runs sucessfully on my S5M2, but it takes 30s+ to return to the new file location & command prompt (it seems to take much longer if camera SD card has a lot of files on it).

when I run `gphoto2 --shell --debug --debug-loglevel debug <<< capture-image`, it looks like the biggest hang is here:
```0.492092 camera_panasonic_capture    (2): **** GH5: draining old events...
0.492101 camera_panasonic_capture    (2): **** GH5: trigger capture...
0.492125 ptp_usb_sendreq             (2): Sending PTP_OC 0x9404 (Unknown PTP_OC) (0x3000011) request...
0.492227 ptp_usb_getresp             (2): Reading PTP_OC 0x9404 (Unknown PTP_OC) response...
26.500934 camera_panasonic_capture    (2): **** GH5: checking for new object...
26.500983 gp_port_get_timeout         (2): Current port timeout is 20000 milliseconds.
26.501009 gp_port_set_timeout         (2): Setting port timeout to 150 milliseconds.
26.501087 _cb_irq                     (2): 0x126ca90 with status 0
26.501128 _cb_irq                     (2): Requeuing completed transfer 0x126ca90
26.501228 _cb_irq                     (2): 0x127dac0 with status 0
26.501261 _cb_irq                     (2): Requeuing completed transfer 0x127dac0
```

NOTE, the camera is NOT a GH5, but seems to be assuming the GH5 has the most similar driver?
```
0.155509 gp_abilities_list_detect_usb(2): Found 'Panasonic DC-GH5' (0x4da,0x2382)
0.155546 gp_port_free                (2): Freeing port...
0.155555 gp_port_close               (2): Closing port...
0.155987 gp_camera_set_abilities     (2): Setting abilities ('Panasonic DC-GH5')...
0.156011 gp_setting_set              (2): Setting key 'model' to value 'Panasonic DC-GH5' (gphoto2)
0.156024 save_settings               (2): Saving 3 setting(s) to file "/home/pi/.gphoto/settings"
0.156319 gp_port_info_list_lookup_path(2): Looking for path 'usb:001,003' (9 entries available)...
0.156334 gp_port_info_list_get_info  (2): Getting info of entry 3 (9 available)...
0.156345 gp_camera_set_port_info     (2): Setting port info for port 'Universal Serial Bus' at 'usb:001,003'...
0.161995 gp_port_set_timeout         (2): Setting port timeout to 5000 milliseconds.
0.162014 gp_port_set_settings        (2): Setting settings...
0.162028 gp_libusb1_update           (2): (old int=0, conf=-1, alt=-1) port usb:001,003, (new int=0, conf=-1, alt=-1) port usb:001,003
0.162038 gp_libusb1_update           (2): lowlevel libusb1 port not yet opened, no need for libusb changes
0.162049 gp_setting_set              (2): Setting key 'port' to value 'usb:001,003' (gphoto2)
0.162061 save_settings               (2): Saving 3 setting(s) to file "/home/pi/.gphoto/settings"
gphoto2: {/home/pi} /> capture-image
capture-image
0.163402 gp_camera_init              (2): Initializing camera...
0.163453 gp_libusb1_find_device_lib  (2): Looking for USB device (vendor 0x4da, product 0x2382)... found.
0.163572 gp_libusb1_find_device_lib  (2): Detected defaults: config 1, interface 0, altsetting 0, inep 81, outep 01, intep 82, class 06, subclass 01
0.163585 gp_camera_init              (2): Loading '/usr/lib/arm-linux-gnueabihf/libgphoto2/2.5.27/ptp2'...
0.165805 gp_port_open                (2): Opening USB port...
0.165831 gp_libusb1_open             (2): ()
0.165901 gp_libusb1_open             (2): claiming interface 0
0.166164 camera_init                 (2): maxpacketsize 512
0.166197 gp_port_set_timeout         (2): Setting port timeout to 8000 milliseconds.
0.166223 ptp                         (2): PTP: Opening session
0.166333 ptp_usb_sendreq             (2): Sending PTP_OC 0x1002 (Open session) (0x1) request...
0.166496 ptp_usb_getresp             (2): Reading PTP_OC 0x1002 (Open session) response...
0.167562 gp_port_set_timeout         (2): Setting port timeout to 20000 milliseconds.
0.167583 ptp_usb_sendreq             (2): Sending PTP_OC 0x1001 (Get device info) request...
0.167911 ptp_usb_getdata             (2): Reading PTP_OC 0x1001 (Get device info) data...
0.168204 ptp_usb_getresp             (2): Reading PTP_OC 0x1001 (Get device info) response...
0.168803 print_debug_deviceinfo      (2): Device info:
0.168815 print_debug_deviceinfo      (2): Manufacturer: Panasonic
0.168823 print_debug_deviceinfo      (2):   Model: DC-S5M2
0.168830 print_debug_deviceinfo      (2):   device version: 2.0
```

I'm very new to gphoto2 but would love to get this camera running (faster) so please let me know if I can help!