gphoto/libgphoto2 #906: Save To External Drive, phantom drives? pi/linux use

bendanville, 2023-04-28:
Hi I am having an issue when using the capture and download with an external harddrive with a raspberry pi.
What happens is that instead of saving to the harddrive under the media/admin/Passport (name of the drive) a duplicate directory appears under media/admin with the same name. This could be a linux issue though of me not understanding folder permissions?
Important to note this was working well, and having plugged in back into the same port is now giving the above behaviour.

I am using a Canon 80D

msmeissn, 2023-05-09:
I dont know ... currently capture-image-and-download downloads to the "current" directory you run gphoto2 from.
This likely is outside of libgphoto2 / gphoto2 scope.

bendanville, 2023-05-11:
Hi solved it!

For anyone who runs into a problem like this here is my interpretation of the problem and working solution. 
It's worth noting I am new to linux as of a few weeks ago so there is probably some simplistic or wrong assumption here:

In windows you plug a drive in and it will come up with something like "E:/", the way that the PI OS handles it initially seems to match similar logic but will lead you astray.... It's not that linux is wrong, you just can't apply the same assumptions.
When you plug an external drive into a Pi (at least in 2023) a link appears on the desktop and it shows up as /media/drivename. There is a problem with the default utility that is loading the drive where next time it might load to a new location such as /media/drivename1, your old path will still work but will be on the pi therefore slowly filling up your OS SD card.

The solution? From my research there seems to be a few ways to solve this. I went with mounting the drive by it's "device node" name. To find this you use the command in terminal "lsblk"
Which will give you something like this:
`NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0 238.5G  0 disk
├─sda1   8:1    0   512M  0 part /boot/efi
└─sda2   8:2    0   238G  0 part /
sdb      8:16   0   1.8T  0 disk
└─sdb1   8:17   0   1.8T  0 part

In this example sdb1 is the Portable HDD, which I know from the sized of 1.8TB.
Then we make a directory that will be the "mounting point" for the drive, aka define what we want that path to be.
In terminal:
sudo mkdir /mnt/drivename
sudo mount -t ntfs-3g /dev/sdXn /mnt/drivename
sdXn being changed to the what you received from lsblk. For me it was sdb1
You may need to "eject" the drive first in the file explorer window. To make sure it "grabs" the device on start before that problematic utility, I wrote a shell script that calls "sudo mount -t ntfs-3g /dev/sdXn /mnt/drivename" immediately on boot.

To run the shell script on startup I used the ".desktop" method from the below link, as I didn't have success with chrontab.
https://www.tomshardware.com/how-to/run-script-at-boot-raspberry-pi

I can't provide too much more information beyond this as it's all brand new to me, so I'd suggest searching for things from my solution to check you understand what's happening. :)