gphoto/libgphoto2 #907: Can't shoot one more time in one second (Sony ilce-7rm4)

JDidyk19, 2023-04-28:
**Describe the bug**

I recieve gphoto error when i want to shoot in one second again

**libgphoto2 and gphoto2 version**
output of: gphoto 2.5.28, libgphoto 2.5.28

**To Reproduce**
Use this command `gphoto2 --wait-event=3s --capture-image-and-download --port --filename --force-overwrite --quiet`
and in one second again, it occurs gphoto error

vepicurean, 2023-11-27:
I noticed this issue ticket for the Sony A7R4 and -capture-image-and-download is still open. 

It has been a while since we had to post here, as gphoto2 has been working very well for us with the Canon 5DS R. 

I'm still on 2.5.27 for both gphoto2 and libgphoto2 - I stayed on this version as it's been working well for me with the Canon 5DS R camera bodies ( we spent a lot of time on that with you all. Thanks for sharing gphoto as an open source project). 

Finally looking at including the Sony A7R4 and Sony A7R5 in the cameras used for our rapid capture application. 

I've seen a few posts asking about the --wait-event-and-download with the Sony A7R4 not working properly ( we're were hoping the A7R4 would work the way that --wait-event-and-download="FILEADDED" works with Canon EOS camera bodies). 

I can back up the above about --capture-image-and-download. But just as important, no matter what I do --wait-event-and-download may work once, but quickly becomes unstable and stops working with the Sony A7R4. Freezes the port up and have to restart the camera. 

Took me a while, and I don't know what I did in terms of the right setting, but even the simple --capture-image-and-download was giving me trouble for days. I kept changing the settings on the A7R4 and trying --capture-image-and-download until yesterday it started working in a stable manner. 

The issue now is --capture-image-and-download on the Sony A7R4 is much slower to execute in terms of triggering the shutter than --wait-event-and-download="FILEADDED" on the Canon EOS RDS R. I am surprised, I thought the more modern A7R4 would be faster to execute the shutter/trigger no matter command I used. 

In summary, we're tryiing to remote focus ( half press to focus while in continuous autofocus mode), have continuous AI autofocus stay stable, then be able to trigger/capture and download on the Sony A7R4 in the same manner ( as fast or faster ) as the Canon EOS camera bodies do using the following:

gphoto2 --port usb:XXX,YYY   --filename  somefilename   --set-config eosremoterelease="Immediate" --set-config eosremoterelease="Full Release" --wait-event-and-download="FILEADDED"    (something for Sony that does the same in terms of speed as this command) 

Note: we will have multiple cameras that we are triggering rapidly, that is why we specify the usb port number. 

Is there a command you all can think of for the Sony A7R4 that can do things better ( faster and stable) than using --capture-image-and-downlaod ?? 

Also, using a interval -1 tether for --capture-image-and-download with SIGUSR1 on the Sony AR7R did not work for us ( I tried that for days and not stable. Works maybe up to 3 times, and then stops working). 

Thanks for your help as always.

vepicurean, 2023-12-10:
Just following up on this.... The behaviour with gphoto and the Sony A7R4 autofocus challenges during triggering is the same.... 

I have a work around for the Sony, but it requires me to have physical access to the Sony camera system at all times. Which defeats the purpose of the gphoto remote triggering/capture. 

Any suggestions to deal with this issue with the focus challenges and gphoto with the Sony A7R4?