gphoto/libgphoto2 #912: Sony A7iii (fw 4.0.1): "Sorry, your camera does not support generic capture"

mikeev261, 2023-06-17:
**Describe the bug**
I have a Sony A7iii with Firmware 4.01 (latest version as of today). Not sure if the firmware is related. I'm getting the "Sorry, your camera does not support generic capture" error. 

**Name the camera**
USB PTP Class Camera           usb:001,012

**libgphoto2 and gphoto2 version**
gphoto2 2.5.20

Copyright (c) 2000-2018 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.20         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

**To Reproduce**
1. Set Sony A7iii USB to MTP mode
2. Connect to OctoPi via USB
3. ```gphoto2 --wait-event=3s --capture-image-and-download --debug --debug-logfile=debug.log
Waiting for 3 seconds for events from camera. Press Ctrl-C to abort.

*** Error ***
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***```
4. [debug.log](https://github.com/gphoto/libgphoto2/files/11779518/debug.log)

Also: The "PC Remote" option in the camera menu doesn't work... just endlessly says "Connecting..." on the camera screen once I plug in the USB.

hfiguiere, 2023-06-17:
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF

This is a very old version. (4+ years old)

You need to upgrade to a recent version, and you need to set the camera in PC Control mode.

mikeev261, 2023-06-18:
Thanks, I did the following: 

Updated to libgphoto to 2.5.27 and gphoto2 to 2.5.27. I tried downloading and compiling the 2.5.30 and 2.5.28 packages, respectively. libgphoto compiled, and gphoto2 failed with this same error: https://github.com/gphoto/gphoto2/issues/479. So I decided to stick with 2.5.27/2.5.27, as that's the latest managed version pair that I can find via apt-get. 

And now I'm seeing: 
```
gphoto2 --auto-detect
Model                          Port
----------------------------------------------------------
Sony Alpha-A7 III (PC Control) usb:001,00
```
However, the PC control mode still shows "Connecting...", and when I try ```--capture-image``` the console sits for ~10 seconds, and eventually says: 

```
ERROR: Could not capture image.
ERROR: Could not capture.
```

If I do ```--capture-image``` a SECOND time, it then works (and it clears the "Connecting..." screen on the camera).

msmeissn, 2023-06-18:
Newer sony cameras need a bit of waiting in front, so I added this in code now. This would avoid the wait-event.
it is in current GIT.

closed 2023-06-18 by msmeissn