gphoto/libgphoto2 #913: Make Canon Mirror Lockup property writable

anjok, 2023-06-17:
I want to be able to toggle mirror lock for my Canon 6D.

Apparently the customfuncex solution to toggle the lock doesn't work for it.

  https://github.com/gphoto/libgphoto2/blob/master/camlibs/ptp2/canon-eos-customfunc.txt

Instead I found this in the debug logs:

ptp                         (2): event 3: EOS prop d1bf (EOS_MirrorLockupState) info record, datasize is 4
ptp                         (2): event 3: Unknown EOS property d1bf, datasize is 4
ptp                         (2):     0: 01 << currently ON, and this one toggles 
ptp                         (2):     1: 00
ptp                         (2):     2: 00
ptp                         (2):     3: 00

Unfortunately this property is not writable. it's not in the --list-all-configs and thus can't be used.

Is there a way to make this property writable? Or are the writable props determined by the cam?

msmeissn, 2023-06-18:
this is a reporting variable only, setting it will not be possible (although I have not tried).

The customfuncs are specific for each camera and are even less documented than the other codes.

unless we reverse engineer or USB sniff this more we are out of luck.

anjok, 2023-06-18:
Thanks for the reply and the quick PR! I added a comment regarding the label which I think is wrong.

Are there any docs on what makes a property writable? Does the cam report it or how does this work?

And this is not about the customfuncex, which works fine for the 450D, but rather the "EOS prop d1bf (EOS_MirrorLockupState) info record" which seems to have replaced it on the 6D.

For the Canon, apparently it got replaced with that 0xd1bf property. I just saw another property 0xd406 which seems to be setable in the "other" section of the indilib driver, so I thought you "just" need to add a setter for it...

anjok, 2023-06-19:
Here's the 2 log files for ON and OFF state set up in the Canon LCD in the hope they help:

[logs.zip](https://github.com/gphoto/libgphoto2/files/11785187/logs.zip)

I commented here:

 https://github.com/gphoto/libgphoto2/commit/50338bdc279eb2bb3ed6ac81fb8574479c415f93

For both EOS_MirrorLockupState and EOS_MirrorUpSetting I got "01" values when I enable Mirror lock in the UI and do --get-config and "00" when I disable it. So I'm not sure which one it is, but "Mirror Up Status" is not PTP_DPC_CANON_EOS_MirrorLockupState.

msmeissn, 2023-06-20:
ptp                         (2): event 227: EOS prop d13a (EOS_MirrorUpSetting) desc record, datasize 8, propxtype 3

this seems settable.

I hooked it up now as "mirrorup" , you can try to set it

anjok, 2023-06-20:
This works great, thanks a lot! I can toggle the lock flag from indi and that's great, as I really can't access the LCD.

A few things though:

1) I don't know if there are other cams with that sort of setting (probably yes), but something like `mirrorlock` would make more sense?

2) The other new flags appear as writable, see below, but I don't know if that makes sense. They should probably have _put_None? Or even remove them, as they appear to be unclear on what they do:

```pi@astro:~/astro/gphoto/gphoto2-temp-folder/gphoto2 $ gphoto2 --list-config |grep mirror
/main/status/mirrorup -> should be mirrorlock or something
/main/status/mirrorlockstatus -> dunno?
/main/status/mirrordownstatus -> dunno? probably temp status of the mirror
```

3) is there some doc on how to sanely install the dev version on the raspberry pi? I'm using 

`wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh && chmod +x gphoto2-updater.sh && sudo ./gphoto2-updater.sh`

but that puts stuff in /usr/local and I have to `sudo apt-get remove gphoto2` and `sudo rm -Rf /usr/lib/arm-linux-gnueabihf/libgphoto*` for this to work

[image]

msmeissn, 2023-06-21:
i renamed it to mirrorlock.

i add put_None to the others.

msmeissn, 2023-06-21:
commit f418c5d51875981bc5d82ca865d3186573366f11

anjok, 2023-06-21:
Works great, I didn't even have to recompile the indi driver. So thanks again!

[image]

anjok, 2023-06-22:
Closing as fixed

closed 2023-06-22 by msmeissn

anjok, 2023-06-29:
Great work, thanks a lot! Closing as fixed.