gphoto/libgphoto2 #914: Capture time is extremely long (~16 minutes)

mikeev261, 2023-06-18:
**Describe the bug**

Capture time is extremely long (~16 minutes). During this period, the camera appears to be communicating with the computer, as the camera's red "busy" light is solidly on, and the icon representing the memory card in the upper left hand corner is continually pulsing with a number 1 (indicating 1 shot is being buffered). 

If I try running with --debug, I see a continuous output, mostly with raw data. It looks like this: 

```
0000  ad 06 00 00 02 00 09 92-23 00 00 00 3c 00 00 00  ........#...<...
0010  00 00 00 00 04 50 02 00-01 01 02 03 02 07 00 02  .....P..........
0020  03 04 10 13 14 12 05 50-04 00 01 01 02 00 02 00  .......P........
0030  02 0f 00 02 00 04 00 11-80 10 80 06 00 01 80 02  ................
0040  80 03 80 04 80 07 00 30-80 12 80 20 80 21 80 22  .......0... .!."
0050  80 07 50 04 00 00 01 ff-ff 90 01 01 00 00 ff ff  ..P.............
0060  01 00 0a 50 04 00 00 02-01 00 01 00 02 06 00 01  ...P............
0070  00 02 00 04 80 05 80 06-80 09 80 0b 50 04 00 00  ............P...
0080  02 01 00 01 80 02 06 00-01 80 02 80 04 80 05 80  ................
0090  03 80 06 80 0c 50 04 00-00 00 01 00 03 00 02 0b  .....P..........
00a0  00 02 00 01 00 04 00 03-00 05 00 01 80 03 80 31  ...............1
00b0  80 32 80 41 80 42 80 0e-50 04 00 00 02 01 00 01  .2.A.B..P.......
00c0  00 02 14 00 00 80 02 00-03 00 04 00 01 00 50 80  ..............P.
00d0  51 80 52 80 53 80 84 80-85 80 86 80 87 80 07 00  Q.R.S...........
00e0  11 80 15 80 14 80 12 80-13 80 17 80 10 50 03 00  .............P..
00f0  00 02 00 00 00 00 02 29-00 00 00 88 13 5c 12 94  .......).....\..
0100  11 cc 10 a0 0f 74 0e ac-0d e4 0c b8 0b 8c 0a c4  .....t..........
0110  09 fc 08 d0 07 a4 06 dc-05 14 05 e8 03 bc 02 f4  ................
0120  01 2c 01 d4 fe 0c fe 44-fd 18 fc ec fa 24 fa 5c  .,.....D.....$.\
0130  f9 30 f8 04 f7 3c f6 74-f5 48 f4 1c f3 54 f2 8c  .0...<.t.H...T..
0140  f1 60 f0 34 ef 6c ee a4-ed 78 ec 13 50 04 00 01  .`.4.l...x..P...
0150  01 01 00 01 00 02 32 00-01 00 10 80 02 00 15 80  ......2.........
0160  12 80 05 80 03 80 04 80-08 80 09 80 0c 80 0d 80  ................
0170  0e 80 0f 80 37 83 37 85-37 89 57 83 57 85 57 89  ....7.7.7.W.W.W.
0180  77 83 77 85 77 89 11 83-11 85 11 89 21 83 21 85  w.w.w.......!.!.
0190  31 83 31 85 36 83 36 85-36 89 56 83 56 85 56 89  1.1.6.6.6.V.V.V.
01a0  76 83 76 85 76 89 10 83-10 85 10 89 20 83 20 85  v.v.v....... . .
01b0  30 83 30 85 18 80 28 80-19 80 29 80 00 d2 03 00  0.0...(...).....
01c0  00 01 00 00 00 00 02 19-00 00 00 b8 0b 8c 0a c4  ................
01d0  09 fc 08 d0 07 a4 06 dc-05 14 05 e8 03 bc 02 f4  ................
01e0  01 2c 01 d4 fe 0c fe 44-fd 18 fc ec fa 24 fa 5c  .,.....D.....$.\
01f0  f9 30 f8 04 f7 3c f6 74-f5 48 f4 01 d2 02 00 01  .0...<.t.H......
0200  01 01 1f 02 0e 00 01 1f-11 12 13 14 15 20 21 22  ............. !"
0210  23 24 25 26 03 d2 02 00-01 01 04 01 02 03 00 01  #$%&............
0220  02 03 0d d2 06 00 00 01-ff ff ff ff 3c 00 01 00  ............<...
0230  01 00 00 00 00 ff ff ff-ff 01 00 00 00 0e d2 02  ................
0240  00 00 02 01 07 02 10 00-01 02 03 0b 08 09 0a 04  ................
0250  0c 05 0d 06 0e 07 0f 10-0f d2 04 00 01 00 7c 15  ..............|.
0260  00 00 01 c4 09 ac 26 64-00 11 d2 02 00 01 01 01  ......&d........
0270  01 02 02 00 01 02 13 d2-02 00 00 02 01 01 02 06  ................
0280  00 01 02 03 05 06 07 1e-d2 06 00 00 01 ff ff ff  ................
0290  00 20 03 00 00 02 59 00-ff ff ff 00 19 00 00 00  . ....Y.........
02a0  32 00 00 00 40 00 00 00-50 00 00 00 64 00 00 00  2...@...P...d...
02b0  7d 00 00 00 a0 00 00 00-c8 00 00 00 fa 00 00 00  }...............
02c0  40 01 00 00 90 01 00 00-f4 01 00 00 80 02 00 00  @...............
02d0  20 03 00 00 e8 03 00 00-e2 04 00 00 40 06 00 00   ...........@...
02e0  d0 07 00 00 c4 09 00 00-80 0c 00 00 a0 0f 00 00  ................
02f0  88 13 00 00 00 19 00 00-40 1f 00 00 10 27 00 00  ........@....'..
0300  00 32 00 00 80 3e 00 00-20 4e 00 00 00 64 00 00  .2...>.. N...d..
0310  00 7d 00 00 40 9c 00 00-00 c8 00 00 00 fa 00 00  .}..@...........
0320  80 38 01 00 00 90 01 00-00 f4 01 00 00 71 02 00  .8...........q..
0330  00 20 03 00 00 e8 03 00-00 e2 04 00 00 40 06 00  . ...........@..
0340  ff ff ff 01 19 00 00 01-32 00 00 01 40 00 00 01  ........2...@...
0350  50 00 00 01 64 00 00 01-7d 00 00 01 a0 00 00 01  P...d...}.......
0360  c8 00 00 01 fa 00 00 01-40 01 00 01 90 01 00 01  ........@.......
0370  f4 01 00 01 80 02 00 01-20 03 00 01 e8 03 00 01  ........ .......
0380  e2 04 00 01 40 06 00 01-d0 07 00 01 c4 09 00 01  ....@...........
0390  80 0c 00 01 a0 0f 00 01-88 13 00 01 00 19 00 01  ................
03a0  40 1f 00 01 10 27 00 01-00 32 00 01 80 3e 00 01  @....'...2...>..
03b0  00 64 00 01 00 c8 00 01-00 90 01 01 00 20 03 01  .d........... ..
03c0  00 40 06 01 ff ff ff 02-64 00 00 02 c8 00 00 02  .@......d.......
03d0  90 01 00 02 20 03 00 02-40 06 00 02 80 0c 00 02  .... ...@.......
03e0  00 19 00 02 00 32 00 02-00 64 00 02 00 c8 00 02  .....2...d......
[... 133 lines cut ...]
Sony Alpha-A7 III (PC Control) usb:001,004
```

**libgphoto2 and gphoto2 version**
```
:~ $ gphoto2 --version
gphoto2 2.5.28.1

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.30.1       standard camlibs (SKIPPING docupen lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
```

**To Reproduce**
1. Set Sony A7iii to PC Remote mode. 
2. Plug in to raspberry pi Octoprint server (with latest version of Octopi)
3. run ```gphoto2 --capture-image```
4. Observe that the camera takes a picture (which is audible), and then appears to sit there, only to complete eventually (~16 minutes).

msmeissn, 2023-06-19:
I think you have built against libusb 0.1 and not libusb 1.x. 

libusb 0 specifically will not work well for Sonys.

mikeev261, 2023-06-20:
Thanks Marcus. I'll recompile and look for this.

mikeev261, 2023-06-21:
Before compile: 

```
:~ $ dpkg -l | grep usb
ii  libhidapi-libusb0:armhf              0.10.1+dfsg-1                    armhf        Multi-Platform library for communication with HID devices (libusb backend)
ii  libusb-0.1-4:armhf                   2:0.1.12-32                      armhf        userspace USB programming libraryii  libusb-1.0-0:armhf                   2:1.0.24-3                       armhf        userspace USB programming libraryii  usb-modeswitch                       2.6.1-1                          armhf        mode switching tool for controlling "flip flop" USB devices
ii  usb-modeswitch-data                  20191128-3                       all          mode switching data for usb-modeswitch
ii  usbutils                             1:013-3                          armhf        Linux USB utilities
```
So I made sure to remove the 0.1-4 version and install 1.0-0:

```
:~ $ dpkg -l | grep usb
ii  libhidapi-libusb0:armhf              0.10.1+dfsg-1                    armhf        Multi-Platform library for communication with HID devices (libusb backend)
ii  libusb-1.0-0:armhf                   2:1.0.24-3                       armhf        userspace USB programming library
ii  usb-modeswitch                       2.6.1-1                          armhf        mode switching tool for controlling "flip flop" USB devices
ii  usb-modeswitch-data                  20191128-3                       all          mode switching data for usb-modeswitch
ii  usbutils                             1:013-3                          armhf        Linux USB utilities
```

I then installed both packaged, and I unfortunately got the "libgphoto2_port" line showing "no USB". 

I then tried to re-install libusb-dev, and it seems to continue pointing at 0.1.x versions: 
```
 sudo apt-get install libusb-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libhidapi-libusb0 raspinfo
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libusb-0.1-4
The following NEW packages will be installed:
  libusb-0.1-4 libusb-dev
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 57.3 kB of archives.
After this operation, 272 kB of additional disk space will be used.
```
If I try to force libusb-1.0-0 to be installed, I see this: 

```
$ sudo apt install libusb-1.0-0
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libusb-1.0-0 is already the newest version (2:1.0.24-3).
libusb-1.0-0 set to manually installed.
```

So I think I'm kinda stuck here. I can't get USB to work at all without libusb-dev, which seems to install the 0.1 version of libusb alongside the 1.0 version. And then if I do that, libgphoto2 seems to not interact well with my A7iii (unsurprisingly getting the endless single capture symptom again after repeating these steps). 

For another datapoint: I also have an A7iv, and using --trigger-capture I was able to get it to work. So there's something different about the A7iii with this setup that appears problematic.

msmeissn, 2023-06-21:
try to install  libusb-1-dev or libusb-1.0-dev or libusb1.0-0-dev

mikeev261, 2023-06-21:
I got it to this state: 

```
:~ $ dpkg -l | grep usb
ii  libusb-1.0-0:armhf                   2:1.0.24-3                       armhf        userspace USB programming library
ii  libusb-1.0-0-dev:armhf               2:1.0.24-3                       armhf        userspace USB programming library development files
ii  libusb-1.0-doc                       2:1.0.24-3                       all          documentation for userspace USB programming
ii  usb-modeswitch                       2.6.1-1                          armhf        mode switching tool for controlling "flip flop" USB devices
ii  usb-modeswitch-data                  20191128-3                       all          mode switching data for usb-modeswitch
ii  usbutils                             1:013-3                          armhf        Linux USB utilities
```

Then I reconfigured and remade libgphoto2... but unfortunately I'm getting the same result. When I try to take a photo the camera is just endlessly looping on that 1 photo, and it's stuck. I can't recover the camera unless I turn it off and take out the battery. Something about that USB command is really knocking it into the weeds...

Just for sanity: 
```
:~ $ gphoto2 --version
gphoto2 2.5.28.1

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.30.1       standard camlibs (SKIPPING docupen lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
```

Any other ideas? I'm only trying to initiate a capture trigger here with the a7iii (record to memory card). And I'm using: 
```
gphoto2  --trigger-capture
```
Thanks!

msmeissn, 2023-06-21:
Earlier Sony Alpha only could capture to SDRAM. 

Perhaps yours can also capture to card, this needs to be set to be going to the card.

Is it still taking 16 minutes?

mikeev261, 2023-06-22:
The A7iii is still relatively new. And is widely popular, which is why I'm a bit confused as to why I appear to be the only one hitting these issues. Makes me all but convinced that I'm doing something fundamentally wrong. 

But yes, with --trigger-capture it took over an hour before I gave up and unplugged the camera. 

I'm a bit confused by your question though... this is a mirrorless SLR with a very large buffer. Pretty sure the camera always writes to the buffer no matter what (there are no options that I've ever seen) and then this buffer drains to the SD card (FIFO). Is there a certain setting I should be applying in gphoto? 

Again all I'm looking for here is a trigger- I don't need to transfer the image itself. It can stay on the camera.

msmeissn, 2023-06-22:
can you get a fresh debuglogfile from the current build with --trigger-capture  --debug --debug-logfile=capture.log ?

mikeev261, 2023-06-22:
That's a good idea. I have a large print going, but I'll do this as soon as the printer is free again.

mikeev261, 2023-06-23:
Here's a log with ```---trigger-capture```. The console does NOT take 16 minutes (~2-4 seconds instead), but the camera endlessly writes the initial photo from the buffer and can't take a 2nd image. It's almost as if there's a missing "end" command or something that the camera is waiting for. 

```
:~ $ gphoto2 --trigger-capture --debug
0.000432 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000615 main                        (2): gphoto2 2.5.28.1
0.000690 main                        (2): gphoto2 has been compiled with the following options:
0.000755 main                        (2):  + gcc (C compiler used)
0.000816 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000876 main                        (2):  + exif (for displaying EXIF information)
0.000934 main                        (2):  + no cdk (for accessing configuration options)
0.000993 main                        (2):  + no aa (for displaying live previews)
0.001051 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.001110 main                        (2):  + no readline (for easy navigation in the shell)
0.001175 main                        (2): libgphoto2 2.5.30.1
0.001243 main                        (2): libgphoto2 has been compiled with the following options:
0.001327 main                        (2):  + standard camlib set SKIPPING SOME (ax203 canon digigr8 dimagev directory jl2005a jl2005c kodak_dc240 mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801 SKIPPING docupen lumix)
0.001397 main                        (2):  + gcc (C compiler used)
0.001455 main                        (2):  + no ltdl (for portable loading of camlibs)
0.001513 main                        (2):  + EXIF (for special handling of EXIF files)
0.001577 main                        (2): libgphoto2_port 0.12.2
0.001648 main                        (2): libgphoto2_port has been compiled with the following options:
0.001709 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.001766 main                        (2):  + gcc (C compiler used)
0.001822 main                        (2):  + no ltdl (for portable loading of iolibs)
0.001879 main                        (2):  + EXIF (for vusb)
0.001937 main                        (2):  + USB (libusb1, for USB cameras)
0.001995 main                        (2):  + serial (for serial cameras)
0.002053 main                        (2):  + no resmgr (serial port access and locking)
0.002109 main                        (2):  + no ttylock (serial port locking)
0.002165 main                        (2):  + no lockdev (serial port locking)
0.002221 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.002277 main                        (2): IOLIBS env var not set, using compile-time default instead
0.002341 main                        (2): invoked with following arguments:
0.002400 main                        (2):   --trigger-capture
0.002462 main                        (2):   --debug
0.002597 load_settings               (2): Creating gphoto config directory ('/home/mikeev/.gphoto')
0.002830 load_settings               (2): Loading settings from file '/home/mikeev/.gphoto/settings'.
0.003552 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.003645 gp_port_init_localedir      (2): localedir has been set to /usr/local/share/locale (compile-time default)
0.003723 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/local/lib/libgphoto2_port/0.12.2'...
0.004131 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.2/disk'.
0.005106 gp_port_library_list        (2): found fstab fstype proc
0.005208 gp_port_library_list        (2): found fstab fstype vfat
0.005342 gp_port_library_list        (2): found fstab fstype ext4
0.005661 gp_port_library_list        (2): found mtab fstype ext4
0.005766 gp_port_library_list        (2): found mtab fstype devtmpfs
0.005845 gp_port_library_list        (2): found mtab fstype proc
0.005916 gp_port_library_list        (2): found mtab fstype sysfs
0.005997 gp_port_library_list        (2): found mtab fstype securityfs
0.006078 gp_port_library_list        (2): found mtab fstype tmpfs
0.006208 gp_port_library_list        (2): found mtab fstype devpts
0.006287 gp_port_library_list        (2): found mtab fstype tmpfs
0.006396 gp_port_library_list        (2): found mtab fstype tmpfs
0.006513 gp_port_library_list        (2): found mtab fstype cgroup2
0.006593 gp_port_library_list        (2): found mtab fstype pstore
0.006706 gp_port_library_list        (2): found mtab fstype bpf
0.006819 gp_port_library_list        (2): found mtab fstype autofs
0.006896 gp_port_library_list        (2): found mtab fstype mqueue
0.006972 gp_port_library_list        (2): found mtab fstype rpc_pipefs
0.007139 gp_port_library_list        (2): found mtab fstype tracefs
0.007211 gp_port_library_list        (2): found mtab fstype debugfs
0.007277 gp_port_library_list        (2): found mtab fstype configfs
0.007360 gp_port_library_list        (2): found mtab fstype fusectl
0.007398 gp_port_library_list        (2): found mtab fstype vfat
0.007458 gp_port_library_list        (2): found mtab fstype tmpfs
[... cut ...]
Note that I had to copy/paste with --debug above because the --debug-logfile=FILENAME switch doesn't work for me with this latest build (I just get no file and no error message along with it). It definitely worked before (when I was using the apt-get packaged version). 

If I want to just trigger capture and record everything in-camera, ```--trigger-capture``` is the correct option, right? Originally (at the top of this issue) I was using ```--capture-image``` and the camera behaved the same, but the console command didn't end for 16 minutes.

andi34, 2024-09-21:
You could try  
`gphoto2 --set-config output=Off --trigger-capture --wait-event-and-download=FILEADDED`
Or
`gphoto2 --trigger-capture --wait-event-and-download=FILEADDED`

At least seems to work on some Canon cameras.